britkat1980/giv_tcp #22: Inverter Time causes errors on HA 2025.05.x

metalblue, 2023-05-07:
On HA 2023.05.xxx sensors without a device class are assumed to be numeric. This causes errors in the log for the `Inverter Time` sensor.

```
2023-05-07 17:19:36.777 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.givtcp_sd2204g182_invertor_time, topic: 'GivEnergy/xxxxxxxx/Invertor_Details/Invertor_Time' with payload: b'07-05-2023 17:19:24'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 579, in state
    numerical_value = int(value)
ValueError: invalid literal for int() with base 10: '07-05-2023 17:19:24'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 270, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 585, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 649, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 591, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 583, in state
    raise ValueError(
ValueError: Sensor sensor.givtcp_xxxxxxxx_invertor_time has device class None, state class None unit  and suggested precision None thus indicating it has a numeric value; however, it has the non-numeric value: 07-05-2023 17:19:24 (<class 'str'>)
```

metalblue, 2023-05-07:
One solution would be to set the Inverter Time sensor to be a string sensor and use the same solution as has been used for the firmware version, battery type etc sensors.

However under the hood these sensors get reported to HA as an enum sensor, which implies it only has a few unique values, which time clearly doesn't.

rjenx, 2023-05-07:
It was working fine in the previous version so what has changed to cause this error?

metalblue, 2023-05-07:
Something has changed in HomeAssistant core, but I'm not sure what.

metalblue, 2023-05-08:
I think #24 should fix this

rjenx, 2023-05-08:
I had multiple issues with 2023.5.2 so reverted to 2023.4.6 all the issues have gone away including this one. (So far).

springfall2008, 2023-05-13:
I'm having the same issue since the HA update

metalblue, 2023-05-13:
This should now be fixed in the latest version

britkat1980, 2023-06-04:
Fixed

closed 2023-06-04 by britkat1980