britkat1980/giv_tcp #29: Duplicate entities in HA 5.2

JohnKitching, 2023-05-14:
Existing givtcp entities have stopped working, new ones have been created with a suffix of _2 which are working

rjenx, 2023-05-14:
Confirm same issue for me  
HA 2023.4.6

mprowe, 2023-05-14:
Another "me too"....
HA 2023.5.3

stace2023, 2023-05-15:
Same here, this breaks a lot of automation already in place as well, please can entities be swapped around?

CaptSB, 2023-05-15:
Is there a quick fix for this as losing all of the entities really isn’t
helpful. Suddenly I have lost control of my GivEnergy system.

Cheers

Richard
-- 
Sent from Gmail Mobile

CaptSB, 2023-05-15:
Not to mention that, if I switch to the _2 entities, I lose all of my
historical data that is attached to the original entities.

britkat1980, 2023-06-04:
Closed. This was a breaking change.

closed 2023-06-04 by britkat1980