britkat1980/giv_tcp #32: Error disabling scheduled charging

springfall2008, 2023-05-19:
Calling this service:

service: switch.turn_off
data: {}
target:
  entity_id: switch.givtcp_sa2243g277_enable_charge_schedule

Is leading to this error in the log, the same issue happens when I toggle in the GUI

2023-05-19 07:44:08,318 - write - [INFO] - Disabling Charge Schedule
2023-05-19 07:44:13,304 - write - [INFO] - Disabling Charge Schedule
2023-05-19 07:44:27,221 - mqtt - [ERROR] - Error connecting to MQTT Broker: (<class 'TypeError'>, TypeError('payload must be a string, bytearray, int, float or None.'), <traceback object at 0x7fd3f5d2f0c0>)

springfall2008, 2023-05-19:
Rebooting GivTCP seemed to fix the problem, I've not seen that before

springfall2008, 2023-05-27:
So it seems GivTCP gets stuck in this mode and no register writes work.

It would be good to fix the problem or have a way to reboot it automatically on this error

britkat1980, 2023-06-12:
Looking to fix this in 2.1.16. Just soak testing the fix

springfall2008, 2023-06-12:
Great :)

closed 2023-06-28 by britkat1980