britkat1980/giv_tcp #35: Force export doesn't stop when time selected

prototypeuk, 2023-05-28:
Force export works and export starts but doesn't cancel when selected specified time from the drop down list and you have to manually cancel it.  GivTCP version 2.1.13. Gen 2 3.6 Hybrid Inverter

Adrian57, 2023-05-29:
I have similar. Sometimes it words fine, sometimes not. Sometimes it stops exporting at the timeout and sometimes it doesn't. More often than not, it actually works and does stop exporting but in the HA interface the control says "Running" and stays like that for many hours (I'm not sure what clears it in the end - possibly a HA restart or GivTCP restart). I try setting Normal or Cancel but it just sticks at Running.

Ed-M72, 2023-05-29:
> 
A GivTCP restart will clear that, although this shouldn't be necessary as the status is cleared at the end of the export time.  Can you turn on debug log and attach the relevant entries following the "Reverting Force Export settings:" line?

Ed-M72, 2023-05-29:
> Force export works and export starts but doesn't cancel when selected specified time from the drop down list and you have to manually cancel it. GivTCP version 2.1.13. Gen 2 3.6 Hybrid Inverter

Can you turn on debug log and attach the relevant entries following the "Reverting Force Export settings:" line?

prototypeuk, 2023-05-30:
Sorry how to I enable debug mode and where is the log file created saved?

prototypeuk, 2023-06-03:
> 

It never gets to revert export because it doesn't change the state to start with. If you look it shows forcing export for 90 minutes but thereafter it says Not Exporting so never reverts back to Eco mode and stays in Timed Export. Also when forcing export it changes the Reserve Battery charge to 4% but I have mine set to 10% and it changes it to 4% every time and I have to manually change it back again.

[Force Export.txt](https://github.com/britkat1980/giv_tcp/files/11643466/Force.Export.txt)

Ed-M72, 2023-06-04:
Looks like the Inverter time slots are not being written correctly.

@britkat1980 - This is the exception:
 2023-06-03 16:01:52,184 - write - [ERROR] - Force Export failed: (<class 'AssertionError'>, AssertionError('Unexpected response from remote end: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response'), <traceback object at 0x7fb6dd3840>)

I have added extra logging into pull request 42.  I have also fixed the reserve issue so that it only alters the reserve if it has been set to 100% and therefore would affect a force export.

prototypeuk, 2023-06-05:
That's great thank you. If you need me to post any more logs let me know.

Ed-M72, 2023-06-30:
@prototypeuk - could you please retest on 2.2 and let me know the outcome. Both issues should be fixed now.

closed 2024-10-01 by britkat1980