britkat1980/giv_tcp #36: REST API - getCache appears not to work

springfall2008, 2023-05-30:
Calling the getCache endpoint gives an error, in the logs I see:

2023-05-30 20:33:50,165 - app - [ERROR] - Exception on /getCache [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.10/site-packages/flask/app.py", line 1823, in full_dispatch_request
    return self.finalize_request(rv)
  File "/usr/local/lib/python3.10/site-packages/flask/app.py", line 1842, in finalize_request
    response = self.make_response(rv)
  File "/usr/local/lib/python3.10/site-packages/flask/app.py", line 2153, in make_response
    rv = self.json.response(rv)
  File "/usr/local/lib/python3.10/site-packages/flask/json/provider.py", line 309, in response
    f"{self.dumps(obj, **dump_args)}\n", mimetype=mimetype
  File "/usr/local/lib/python3.10/site-packages/flask/json/provider.py", line 230, in dumps
    return json.dumps(obj, **kwargs)
  File "/usr/local/lib/python3.10/json/__init__.py", line 238, in dumps
    **kw).encode(obj)
  File "/usr/local/lib/python3.10/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/local/lib/python3.10/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/usr/local/lib/python3.10/site-packages/flask/json/provider.py", line 122, in _default
    raise TypeError(f"Object of type {type(o).__name__} is not JSON serializable")

britkat1980, 2023-06-04:
getCache is not outputting json data. This is slated to be updated in 2.1.14

springfall2008, 2023-06-06:
Fixed

closed 2023-06-06 by springfall2008