britkat1980/giv_tcp #40: Exception("Garbage or failed Invertor Response: ERROR:-(<class 'KeyError'>, KeyError(HR:013)

stuartornum, 2023-05-31:
Hi,

I'm getting an error (log file at the bottom) when trying to pull data from a GEN1 3.6kW inverter. 

Worthy notes:

- Running latest v2.1.13, was using HA, however, running locally to debug as well
- Also have a GEN2 5kW, this works perfectly fine with GivTCP
- The GEN1 was working before with GivTCP, until a dodgy firmware update bricked the inverter
- A GivEnergy engineer came today to replace the GEN1 control board
- The GEN1 is now back online and reporting back to the GivEnergy Portal just fine
- The GEN1 does not have a battery attached (before, I noticed GivTCP would only work with `NUMBATTERIES_1=1` even though no battery was attached)

GEN1 3.6kW Log: https://gist.github.com/stuartornum/d25677051bb5d6f201a3d8c84b91f1be

Thanks

britkat1980, 2023-06-04:
That is a common error, and at this point has no fix. However it should only occur sporadically. Are you not receiving any data at all for this invertor?

stuartornum, 2023-06-05:
> That is a common error, and at this point has no fix. However it should only occur sporadically. Are you not receiving any data at all for this invertor?

Thanks for the reply.

This happens continually, I've not managed to get any data out of this inverter since it's control board was changed.

Is this down to a specific firmware of the inverter?

Is there any debugging tasks I can do on my side to pinpoint where it's failing?

Thanks again

britkat1980, 2023-06-05:
If you set Log-Level to debug and then send me the log output, that might help point it in the right direction. How are you running this, via HA Addon, or your own docker container?

stuartornum, 2023-06-05:
https://gist.github.com/stuartornum/d25677051bb5d6f201a3d8c84b91f1be

It gets stuck in this loop.
Thanks,

closed 2024-10-01 by britkat1980