britkat1980/giv_tcp #44: Missing Data - Battery Cell voltage or temp mainly

harvybob, 2023-06-06:
Hi, 

Firstly thanks for the great bit of software!

I have a Gen 2 5kw Hybrid(D0.909-A0.909) , with a 9.5kwH battery (3012)

GivTCP (2.1.14) running on Home Assistant 2023.5.4 Supervisor 2023.04.1 Operating System 10.2 Frontend 20230503.3 - latest

Not sure how long this has been happening, but Noticed this in the logs today:

'''
2023-05-31 00:00:50,965 - read - [CRITICAL] - Midnight, so resetting Day/Night stats...
2023-05-31 00:20:13,789 - palm_soc - [CRITICAL] - PALM... PV Automated Load Manager Version: v0.8.3bSoC
2023-05-31 00:20:14,789 - palm_soc - [ERROR] - 404 Client Error: Not Found for url: https://api.solcast.com.au/rooftop_sites//forecasts?format=json&api_key=
2023-05-31 00:20:14,790 - palm_soc - [ERROR] - Error; Problem reading Solcast data, using previous values (if any)
2023-05-31 00:20:14,791 - palm_soc - [ERROR] - Incomplete Solcast data, setting target SoC to 100%
2023-05-31 00:20:14,791 - palm_soc - [ERROR] - Incomplete Solcast data, setting target SoC to 100%
2023-05-31 00:20:14,791 - palm_soc - [ERROR] - Incomplete Solcast data, setting target SoC to 100%
2023-05-31 00:20:14,792 - palm_soc - [ERROR] - Incomplete Solcast data, setting target SoC to 100%
2023-05-31 00:20:15,525 - write - [ERROR] - Setting Charge Target failed: (<class 'AssertionError'>, AssertionError('Unexpected response from remote end: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response'), <traceback object at 0x7fbc04850600>)
2023-05-31 02:26:24,593 - read - [ERROR] - Consecutive failure count= 3 -- (<class 'Exception'>, Exception("Garbage or failed Invertor Response: ERROR:-(<class 'KeyError'>, KeyError(IR:110), <traceback object at 0x7fdebb665280>)"), <traceback object at 0x7f106a73a2c0>)
2023-05-31 05:32:25,725 - read - [ERROR] - Consecutive failure count= 3 -- (<class 'Exception'>, Exception("Garbage or failed Invertor Response: ERROR:-(<class 'KeyError'>, KeyError(IR:183), <traceback object at 0x7fdebb6712c0>)"), <traceback object at 0x7f106a92dc40>)
2023-05-31 05:32:38,488 - read - [ERROR] - Consecutive failure count= 4 -- (<class 'Exception'>, Exception("Garbage or failed Invertor Response: ERROR:-(<class 'KeyError'>, KeyError(IR:000), <traceback object at 0x7fdebb6714c0>)"), <traceback object at 0x7f106a3c6980>)
2023-05-31 05:32:48,746 - read - [ERROR] - Consecutive failure count= 5 -- (<class 'Exception'>, Exception("Garbage or failed Invertor Response: ERROR:-(<class 'KeyError'>, KeyError(HR:113), <traceback object at 0x7fdebb671080>)"), <traceback object at 0x7f106abe5400>)
2023-05-31 15:10:56,725 - read - [CRITICAL] - First time running so saving AC Charge status
2023-05-31 15:10:57,085 - read - [CRITICAL] - Publishing Home Assistant Discovery messages
2023-05-31 15:29:35,478 - read - [CRITICAL] - First time running so saving AC Charge status
2023-05-31 15:29:35,657 - read - [CRITICAL] - Publishing Home Assistant Discovery messages
2023-05-31 16:06:34,712 - read - [CRITICAL] - First time running so saving AC Charge status
2023-05-31 16:06:34,733 - read - [CRITICAL] - Publishing Home Assistant Discovery messages
2023-05-31 16:28:14,669 - read - [ERROR] - Consecutive failure count= 3 -- (<class 'Exception'>, Exception("Garbage or failed Invertor Response: ERROR:-(<class 'KeyError'>, KeyError(HR:120), <traceback object at 0x7fa3b1af8d00>)"), <traceback object at 0x7f940d9e9440>)
2023-05-31 16:28:25,424 - read - [ERROR] - Consecutive failure count= 4 -- (<class 'Exception'>, Exception("Garbage or failed Invertor Response: ERROR:-(<class 'KeyError'>, KeyError(HR:120), <traceback object at 0x7fa3b1add3c0>)"), <traceback object at 0x7f940d3dd940>)
2023-05-31 16:28:38,185 - read - [ERROR] - Consecutive failure count= 5 -- (<class 'Exception'>, Exception("Garbage or failed Invertor Response: ERROR:-(<class 'KeyError'>, KeyError(HR:120), <traceback object at 0x7fa3b1af84c0>)"), <traceback object at 0x7f940cfc01c0>)
2023-05-31 16:29:47,777 - read - [ERROR] - Consecutive failure count= 3 -- (<class 'Exception'>, Exception("Garbage or failed Invertor Response: ERROR:-(<class 'KeyError'>, KeyError(IR:110), <traceback object at 0x7fa3b1af0c40>)"), <traceback object at 0x7f940d5f3dc0>)
2023-05-31 16:51:16,750 - read - [ERROR] - Battery Object empty so skipping
2023-05-31 16:51:16,753 - read - [ERROR] - Battery_Cell_1_Temperature is missing from new data, publishing all other data
2023-05-31 16:51:16,754 - read - [ERROR] - Battery_Temperature is missing from new data, publishing all other data
2023-05-31 16:51:16,754 - read - [ERROR] - Battery_SOC is missing from new data, publishing all other data
2023-05-31 16:51:16,755 - read - [ERROR] - Battery_Cell_6_Voltage is missing from new data, publishing all other data
2023-05-31 16:51:16,755 - read - [ERROR] - Battery_Cell_4_Voltage is missing from new data, publishing all other data
2023-05-31 16:51:16,755 - read - [ERROR] - Battery_Design_Capacity is missing from new data, publishing all other data
2023-05-31 16:51:16,756 - read - [ERROR] - Battery_Cell_5_Voltage is missing from new data, publishing all other data
2023-05-31 16:51:16,756 - read - [ERROR] - Battery_Cycles is missing from new data, publishing all other data
2023-05-31 16:51:16,756 - read - [ERROR] - Battery_Cell_8_Voltage is missing from new data, publishing all other data
2023-05-31 16:51:16,757 - read - [ERROR] - Battery_Voltage is missing from new data, publishing all other data
2023-05-31 16:51:16,757 - read - [ERROR] - Battery_Cell_3_Temperature is missing from new data, publishing all other data
2023-05-31 16:51:16,757 - read - [ERROR] - Battery_Cell_15_Voltage is missing from new data, publishing all other data
2023-05-31 16:51:16,758 - read - [ERROR] - Battery_Cell_14_Voltage is missing from new data, publishing all other data
2023-05-31 16:51:16,758 - read - [ERROR] - Battery_Cell_2_Voltage is missing from new data, publishing all other data
2023-05-31 16:51:16,758 - read - [ERROR] - Battery_Cell_13_Voltage is missing from new data, publishing all other data
2023-05-31 16:51:16,759 - read - [ERROR] - Battery_Cell_11_Voltage is missing from new data, publishing all other data
2023-05-31 16:51:16,759 - read - [ERROR] - Battery_Serial_Number is missing from new data, publishing all other data
2023-05-31 16:51:16,759 - read - [ERROR] - Battery_Cell_2_Temperature is missing from new data, publishing all other data
2023-05-31 16:51:16,760 - read - [ERROR] - Battery_Cell_12_Voltage is missing from new data, publishing all other data
2023-05-31 16:51:16,760 - read - [ERROR] - Battery_Remaining_Capacity is missing from new data, publishing all other data
2023-05-31 16:51:16,760 - read - [ERROR] - DX2243G390 is missing from new data, publishing all other data
2023-05-31 16:51:16,761 - read - [ERROR] - Battery_Cell_7_Voltage is missing from new data, publishing all other data
2023-05-31 16:51:16,761 - read - [ERROR] - Battery_Cell_4_Temperature is missing from new data, publishing all other data
2023-05-31 16:51:16,761 - read - [ERROR] - Battery_Cell_16_Voltage is missing from new data, publishing all other data
2023-05-31 16:51:16,762 - read - [ERROR] - Battery_Firmware_Version is missing from new data, publishing all other data
2023-05-31 16:51:16,762 - read - [ERROR] - Battery_Capacity is missing from new data, publishing all other data
2023-05-31 16:51:16,762 - read - [ERROR] - Battery_Cell_3_Voltage is missing from new data, publishing all other data
2023-05-31 16:51:16,763 - read - [ERROR] - Battery_Cells is missing from new data, publishing all other data
2023-05-31 16:51:16,763 - read - [ERROR] - Battery_Cell_1_Voltage is missing from new data, publishing all other data
2023-05-31 16:51:16,763 - read - [ERROR] - Battery_USB_present is missing from new data, publishing all other data
2023-05-31 16:51:16,764 - read - [ERROR] - Battery_Cell_10_Voltage is missing from new data, publishing all other data
2023-05-31 16:51:16,764 - read - [ERROR] - Battery_Cell_9_Voltage is missing from new data, publishing all other data'''

The garbage ones I am aware is already raised, and the solcast I dont use GIVTCP as I dont have enough API calls for 2 and hour, but the issue is the other errors. Any ideas where to understand where this error may be? Seems that GivTCP cant read it from the inverter?

78davidcross, 2023-06-13:
Just to confirm, I get the Battery errors listed in the screenshot as well.

andrzej-r, 2023-06-25:
Not sure if this is the same issue. I can see the temperature and voltage entities but they do not provide HA's min/mean/max long-term statistics, I would expect such sensors to have. That is, these entities cannot be selected in e.g. Statistics Graph Card.

In practice, this means I can only see/plot data at most a couple of weeks old. This is particularly problematic for battery temperatures, which I would like to track over the whole lifetime of the battery.

britkat1980, 2023-06-28:
This is a case of inverter not returning battery data. I have tidied up the logging to make it a bit less verbose

closed 2023-06-28 by britkat1980