britkat1980/giv_tcp #56: Feature request - ability to read registers directly via REST API

springfall2008, 2023-06-09:
Currently via REST you can only force a entire refresh of the register set (runAll), but you can't read a particular register

Could we have it so that you can read a register directly?

Being able to write directly would also be nice, but I understand the risk of breaking stuff maybe too high.

britkat1980, 2023-06-28:
The agreement with GE is not to allow unfettered register writing in GivTCP, its just too dangerous!

If there's anything you'd like to control, then I can look at adding it, if appropriate

springfall2008, 2023-06-28:
I’d like to be able to read back a register value to check a write was successful. Right now I force a full refresh and the check the JSON data. I don’t think direct writes are required as such.

springfall2008, 2023-06-28:
In terms of controls, I think the REST API has most stuff now, although there is something a bit odd with Eco mode where tries to change the reserve to a saved value. I'd prefer if the interface change one register at once and didn't depend at all on the GivTCP cache.

closed 2024-10-01 by britkat1980