britkat1980/giv_tcp #64: Eco(Paused) not working

Gary-Manuel, 2023-06-22:
I believe that Eco(Paused) just sets Battery Power Reserve to 100% in order to disable discharge. It would appear though, that 100% is not a valid setting for Battery Power Reserve, either from HA or from the portal. 99% is however a valid setting and it is suggested that GivTCP could use this instead of 100%.

britkat1980, 2023-06-28:
I am able to set battery power reserve to 100% on my HY5.0, what model do you have and what f/w?

Gary-Manuel, 2023-06-28:
Hi. Just checked again and it's the same. Can set to 99% but not 100% from both HA and the portal. Same on both inverters. 2 x HY3.6 Gen2 on v909 with a single 9.5kWh battery on each running v3012.

britkat1980, 2023-06-28:
Ok, perhaps this is a Gen2 thing? I’ll check with the GE team

> Hi. Just checked again and it's the same. Can set to 99% but not 100% from both HA and the portal. Same on both inverters. 2 x HY3.6 Gen2 on v909 with a single 9.5kWh battery on each running v3012.
________________________________
From: britkat1980 ***@***.***>
Sent: 28 June 2023 18:49
To: britkat1980/giv_tcp ***@***.***>
Cc: Gary-Manuel ***@***.***>; Author ***@***.***>
Subject: Re: [britkat1980/giv_tcp] Eco(Paused) not working (Issue #64)
I am able to set battery power reserve to 100% on my HY5.0, what model do you have and what f/w?

closed 2024-10-01 by britkat1980