britkat1980/giv_tcp #71: Battery Power missing battery to grid

adamwylde, 2023-06-30:
Perhaps I'm misunderstanding what the various outputs in GivTcp mean, but the 'battery power' seems to reflect a positive/negative flow of power into/out of the battery (regardless of where its going)? However I find that it does not include exporting to the grid?

I'm trying to use that (perhaps incorrectly) as a guage in Home Assistant but when exporting it sits relatively idle whilst 'battery to grid' is registering the export:

[image]

adamwylde, 2023-06-30:
Restarting GivTCP seems to have resolved this for now and the figures now make sense, although this is not the first time I've seen it happen.

[image]

britkat1980, 2023-07-10:
The power flow stats (XX to XX) are derived values and as such are not always consistent, depending on sampling values. I tend not to rely too heavily on them

closed 2023-08-11 by britkat1980