britkat1980/giv_tcp #75: Second battery details not reporting in Home Assistant

yavinmaster, 2023-07-05:
I'm running 2.2.3 but this issue appears to have been present since I started using the system in March.  I have two batteries connected to my Hybrid Invertor (Gen 2).  Home Assistant MQTT reports one device for the inverter, but only one battery device (the master).  
My docker compose file contains:

`NUMBATTERIES_1=2`

The second battery isn't shown in Home Assistant and the following errors are reported (I'm guessing these "duplicates" are the second battery:

`2023-07-05 02:19:39.942 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Serial_Number is already used by sensor.givtcp_battery_details_battery_serial_number - ignoring sensor.givtcp_battery_details_battery_serial_number
2023-07-05 02:19:39.944 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_SOC is already used by sensor.givtcp_battery_details_battery_soc - ignoring sensor.givtcp_battery_details_battery_soc
2023-07-05 02:19:39.946 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Capacity is already used by sensor.givtcp_battery_details_battery_capacity - ignoring sensor.givtcp_battery_details_battery_capacity
2023-07-05 02:19:39.950 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Design_Capacity is already used by sensor.givtcp_battery_details_battery_design_capacity - ignoring sensor.givtcp_battery_details_battery_design_capacity
2023-07-05 02:19:39.954 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Remaining_Capacity is already used by sensor.givtcp_battery_details_battery_remaining_capacity - ignoring sensor.givtcp_battery_details_battery_remaining_capacity
2023-07-05 02:19:39.956 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Firmware_Version is already used by sensor.givtcp_battery_details_battery_firmware_version - ignoring sensor.givtcp_battery_details_battery_firmware_version
2023-07-05 02:19:39.958 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cells is already used by sensor.givtcp_battery_details_battery_cells - ignoring sensor.givtcp_battery_details_battery_cells
2023-07-05 02:19:39.960 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cycles is already used by sensor.givtcp_battery_details_battery_cycles - ignoring sensor.givtcp_battery_details_battery_cycles
2023-07-05 02:19:39.962 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Temperature is already used by sensor.givtcp_battery_details_battery_temperature - ignoring sensor.givtcp_battery_details_battery_temperature
2023-07-05 02:19:39.964 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Voltage is already used by sensor.givtcp_battery_details_battery_voltage - ignoring sensor.givtcp_battery_details_battery_voltage
2023-07-05 02:19:39.966 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_1_Voltage is already used by sensor.givtcp_battery_details_battery_cell_1_voltage - ignoring sensor.givtcp_battery_details_battery_cell_1_voltage
2023-07-05 02:19:39.968 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_2_Voltage is already used by sensor.givtcp_battery_details_battery_cell_2_voltage - ignoring sensor.givtcp_battery_details_battery_cell_2_voltage
2023-07-05 02:19:39.970 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_3_Voltage is already used by sensor.givtcp_battery_details_battery_cell_3_voltage - ignoring sensor.givtcp_battery_details_battery_cell_3_voltage
2023-07-05 02:19:39.972 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_4_Voltage is already used by sensor.givtcp_battery_details_battery_cell_4_voltage - ignoring sensor.givtcp_battery_details_battery_cell_4_voltage
2023-07-05 02:19:39.974 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_5_Voltage is already used by sensor.givtcp_battery_details_battery_cell_5_voltage - ignoring sensor.givtcp_battery_details_battery_cell_5_voltage
2023-07-05 02:19:39.976 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_6_Voltage is already used by sensor.givtcp_battery_details_battery_cell_6_voltage - ignoring sensor.givtcp_battery_details_battery_cell_6_voltage
2023-07-05 02:19:39.978 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_7_Voltage is already used by sensor.givtcp_battery_details_battery_cell_7_voltage - ignoring sensor.givtcp_battery_details_battery_cell_7_voltage
2023-07-05 02:19:39.980 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_8_Voltage is already used by sensor.givtcp_battery_details_battery_cell_8_voltage - ignoring sensor.givtcp_battery_details_battery_cell_8_voltage
2023-07-05 02:19:39.982 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_9_Voltage is already used by sensor.givtcp_battery_details_battery_cell_9_voltage - ignoring sensor.givtcp_battery_details_battery_cell_9_voltage
2023-07-05 02:19:39.984 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_10_Voltage is already used by sensor.givtcp_battery_details_battery_cell_10_voltage - ignoring sensor.givtcp_battery_details_battery_cell_10_voltage
2023-07-05 02:19:39.986 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_11_Voltage is already used by sensor.givtcp_battery_details_battery_cell_11_voltage - ignoring sensor.givtcp_battery_details_battery_cell_11_voltage
2023-07-05 02:19:39.988 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_12_Voltage is already used by sensor.givtcp_battery_details_battery_cell_12_voltage - ignoring sensor.givtcp_battery_details_battery_cell_12_voltage
2023-07-05 02:19:39.991 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_13_Voltage is already used by sensor.givtcp_battery_details_battery_cell_13_voltage - ignoring sensor.givtcp_battery_details_battery_cell_13_voltage
2023-07-05 02:19:40.029 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_14_Voltage is already used by sensor.givtcp_battery_details_battery_cell_14_voltage - ignoring sensor.givtcp_battery_details_battery_cell_14_voltage
2023-07-05 02:19:40.031 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_15_Voltage is already used by sensor.givtcp_battery_details_battery_cell_15_voltage - ignoring sensor.givtcp_battery_details_battery_cell_15_voltage
2023-07-05 02:19:40.033 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_16_Voltage is already used by sensor.givtcp_battery_details_battery_cell_16_voltage - ignoring sensor.givtcp_battery_details_battery_cell_16_voltage
[... cut ...]
2023-07-05 02:19:40.131 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_3_Temperature is already used by sensor.givtcp_battery_details_battery_cell_3_temperature - ignoring sensor.givtcp_battery_details_battery_cell_3_temperature
2023-07-05 02:19:40.133 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID GivTCP_Battery_Details_Battery_Cell_4_Temperature is already used by sensor.givtcp_battery_details_battery_cell_4_temperature - ignoring sensor.givtcp_battery_details_battery_cell_4_temperature`

yavinmaster, 2023-07-06:
To give further information, both batteries are showing in MQTT, the second one just isn't showing in Home Assistant.

yavinmaster, 2023-07-10:
My setup is GivTCP, Mosquito MQTT and Home Assistant all in docker containers controlled by a compose file.  Let me know if you need more details.  It looks like the MQTT data is correctly showing two battery serial numbers with their data below these, it is just that Home Assistant isn't bringing them both over:
[image]
[image]

britkat1980, 2023-07-10:
OK, that all looks very normal. Could you show me the homeassistant discover topics in the MQTT broker? That must be where the problem is

britkat1980, 2023-07-10:
Actually did you set your mqtt topic to "GivEnergy\Data", rather than the default "GivEnergy". If so, that is probably messing with the formatting of the mqtt data structure and possibly causing this issue. The MQTT Topic should only be a single topic with no subtopics.

yavinmaster, 2023-07-10:
Yes - the mqtt topic is "GivEnergy\Data".  It looks like this was the default topic in the original GivEnergy version of giv_tcp.
If I remove this, will this lose all of my history in HA or cause a load of duplicates?  I know I had to follow your instructions to remove the last set of duplicates but your process meant no history was lost.

Thanks.

yavinmaster, 2023-07-10:
Just in case you still need it.  I think these are the discovery topics you wanted.  They appear OK.  What is interesting is in HA, the battery which is showing, says it's serial number is that of the inverter not either of the batteries.  I'm guessing this adds weight to your theory that the topics are messed up with the subtopic being added to the config?
[image]

britkat1980, 2023-07-10:
This is definitely the problem. But why has it only just arisen now? It hasn’t used that default topic in a really long time, so I don’t know why it would suddenly make a difference.
I suspect that it will cause duplicates if you change the topic, but presumably it was working previously and would have been with the right batteries?

yavinmaster, 2023-07-11:
Thanks.  I made the changes to the topic, stopped Giv_tcp, deleted the entities in MQTT and Home Assistant and restarted giv_tcp.  I now have both sets of battery details.  The other thing that has changes is that I previously only had one invertor mqtt device and one battery device, as well as having two battery devices, the invertor device is now "split" into control, energy, details,last update, power, status, time slots....I'm guessing this is how it was meant to be shown but didn't have this until now.

closed 2023-07-11 by yavinmaster