britkat1980/giv_tcp #77: GivTCP Charge/Discharge Time Remaining appear wrong

veehexx, 2023-07-07:
I've recently moved to givTCP from givenergy_local HA addon, and migrating my dashboards over.
I've noticed that the 'GivTCP Charge Time Remaining' & 'GivTCP Discharge Time Remaining' entity values appear to be wrong.

Specifically these entities in question:
sensor.givtcp_ea########_charge_time_remaining
sensor.givtcp_ea########_discharge_time_remaining

Around 8:20 my system went from discharge to charging the battery. battery was 15% full, and battery charging rate around 1.3kw, so that should mean approximately 4hours to 100%.
[image]
Based on how the graph reports historic values, it appears there's an incorrect calculation somewhere rather than no data.

the related timestamp entities also seem wrong. 
sensor.givtcp_ea########_charge_completion_time = 2023-07-07T08:08:43+00:00
sensor.givtcp_ea########_discharge_completion_time = 2023-07-07T08:27:24+00:00

The current real time is 09:08 (UK timezone), so it puts the charge time as NOW, and discharge in about 15mins time, even battery is in a charging state.

I should be on the latest & greatest versions of software:
Home Assistant 2023.7.1
Supervisor 2023.07.1
Operating System 10.3
Frontend 20230705.1 - latest 

giv_tcp 2.2.3

britkat1980, 2023-07-10:
Do you have the timezone set correctly? I'm in UK and as its BST the timestamp should have +01:00 at the end. Perhaps this is part of the problem?
Secondly, what is your target SOC set to? It calculates time to charge from current SOC up to target.

veehexx, 2023-07-11:
it looks like discharge was working as expected last night.
however charging sensor still seems wrong. 
5.2kw battery, so charge time should somewhere around +3.8hours.

[image]

since you mention the timestamps should account for BST, any more tips on where to confirm thats setup right? if i've read HA wiki correctly, the `time_zone` should account for this so nothing else i need to define.

veehexx, 2023-07-18:
still having the issue so figured i'd take a look at code to see if i could figure out how it was calculated.

https://github.com/britkat1980/giv_tcp/blob/a9d357cd739d5e3a8d564031f4ceb4afbbac733a/GivTCP/read.py#L425

my `Target_SOC` was wrong (`number.givtcp_ea########_target_soc`); was set to 20%. Changed to 100 and the charge_completion_time is now in line with my own calculations!

Looks like it's been set like this since 8th July which, if memory serves correctly, is when i moved to givtcp; Target_SOC has been 20% since the start.

closed 2023-07-30 by britkat1980