britkat1980/giv_tcp #91: V2.3.0

pnd14sh, 2023-07-30:
Repeated exceptions in log when connecting to Gen1 HY3.6. Claiming no serial number found, then loops thru Garbage or failed inverter exception over and over.

Only data that seems to be missing from invertor read is  GivTCP Import Energy Today kWh from what I can see

Temporarily set Inverter_Ac_x to true to "kid" it an older inverter  as a workaround and it seems all inverter data comes through 

[image]

From replies on GE Community, this seems to be specific for Gen 1 Hybrids on this givtcp release as inverter firmware update is not available to Gen 1s but available to Gen2s

Aryankids, 2023-07-31:
@pnd14sh what is _Inverter_Ac_x_  , could you please elaborate?

CaptSB, 2023-07-31:
The dates in the release notes have gone to pot now as well. Apparently
2.3.0 and 2.3.1 were released on 29/6/23 and 30/6/23… whoops.

2.2.4 has now stopped working for me after a battery update to rev 2013.
All entities are returning unknown. Somehow my last restore of 2.2.4 undid
itself so I returned this afternoon to find 2.3.1 installed again.

I will try installing 2.3.1 again but, at the moment, I am left with a
non-working HA which I rely on to run my automations for Octopus Flux. I
hope this is fixed now!!

Cheers

CaptSB, 2023-07-31:
Now on 2.3.2 but getting this:

-  [ERROR   ] - Consecutive failure count= 4 -- (<class 'Exception'>,
Exception("Garbage or failed inverter Response: ERROR:-(<class
'KeyError'>, KeyError(HR:243), <traceback object at 0x7f8d961580>)"),
<traceback object at 0x7f7d8e1040>)
2023-07-31 21:12:25,287 - Inv1 - read

All entities returning Unknown

Is this a start again from scratch job?

pnd14sh, 2023-07-31:
For those on older hybrid inverter firmware versions, by going to the givtcp configuration within HA->HACs->Add-ons you should be able to have a workaround by setting the "Is it on old firmware ? too true for your respective inverter. It may be referred to as something different on AC inverters ( but I don't have one to confirm), but I was pointed at the workaround by a AC inverter owner and it  has work as a workaround hopefully until all the GE inverters are brought up to the same level of capacities/version again in the future.

CaptSB, 2023-07-31:
Nope. I can’t get it to work at all. I have uninstalled Giv_tcp and started
from scratch. All entities are coming back Unknown no matter how I have it
set up.

I have reverted to 2.2.4 and had it working for a couple of hours but it
then went back to 2.3.1 and now I have 2.3.2 and it is just dead. Battery
firmware also updated to 2013 today so I don’t know if this has affected
anything.

At this point HA is just a little bunch of circuits doing not a lot!!

pnd14sh, 2023-07-31:
@CaptSB if you had it working on 2.2.4 for a couple of hours, then why don't you revert back to that and disable auto-update on the info panel so you don't get 2.3.x automatically install over your working system ?. To me better to have something working that is out of date, then having something that doesn't work at all.

CaptSB, 2023-07-31:
Thanks for that. I hadn’t realised the Auto Update was turned on until this
evening - it is off now. Fortunately a complete reinstall from scratch
fixed things once I had typed the IP correctly in the config!! Now running
2.3.2 and can get some sleep tonight!

Cheers

closed 2024-10-01 by britkat1980