britkat1980/giv_tcp #93: Issue with 2.3.1

ProximusAl, 2023-07-31:
Sorry if I'm not doing this correctly, never made a bug report on GitHub before.

Docker 2.3.0 works perfectly but since updating to 2.3.1 I now get this:

2023-07-31 07:35:35,694 - startup - [CRITICAL] - Recreating settings.py for invertor 1
Traceback (most recent call last):
File "/app/startup.py", line 256, in <module>
{1: '192.168.76.138', 2: '192.168.76.150'}
if SuperTimezone: outp.write(" timezone=\""+str(SuperTimezone)+"\"\n")
NameError: name 'SuperTimezone' is not defined

CaptSB, 2023-07-31:
For me, 2.3.0 stopped working at approx 01:30 on 30 Jul. 2.3.1 has either not fixed the issue or made it worse.

ProximusAl, 2023-07-31:
> For me, 2.3.0 stopped working at approx 01:30 on 30 Jul. 2.3.1 has either not fixed the issue or made it worse.

Your issue may be this....different to mine:

New ENV/Config item Inverter_AC_X - Only turn this on if you have an AC inverter on "old" firmware and you get errors on startup

CaptSB, 2023-07-31:
I have tried returning to 2.2.4 with some success - Most Entities returned to a working state.

2.3.1 simply doesn’t work with my setup and some entities in 2.2.4 seem not to work correctly either… I will carry on and see if I can understand what is happening.

CaptSB, 2023-07-31:
The 2.2.4 issue was my mistake - I had incorrectly re-named an entity that is a Template sensor.

So… 2.2.4 is all working; 2.3.0 was working until 01:30 yesterday morning; 2.3.1 doesn’t work - all numerical entities are being reported as non-numeric (probably “Unknown”)

I have a Gen 1 Hybrid inverter with a single battery.

I am running the latest HA on a Raspberry Pi and using MQTT to connect to giv_tcp.

I will stick with 2.2.4 until I can be assured that a new version works without issue - happy to test things though if a “fixed” version is issued - Not an expert but I have generally good technical knowledge.

CaptSB, 2023-07-31:
What a great dat for a GivEnergy Battery Firmware Update!!

ProximusAl, 2023-07-31:
So the issue as far as I can see is this line:

if SuperTimezone: outp.write("    timezone=\""+str(SuperTimezone)+"\"\n")

This doesn't take into account if IsAddon being false therefore crapping out because SuperTimeZone is only ever defined when True.

78davidcross, 2023-07-31:
I am also receiving the "NameError: name 'SuperTimezone' is not defined" error when trying to start 2.3.1.

I have single Hybrid Inverter running firmware 909, single battery.  I run GivTCP on a Synology in Docker using the giv_tcp-ma "latest" image.

I have added "Inverter_AIO_X" as an ENV variable and set it to "False".

78davidcross, 2023-07-31:
To confirm, version 2.3.2 has now resolved this for me, thanks.

ProximusAl, 2023-07-31:
And for me. Thanks. 

Will close the issue.

closed 2023-07-31 by ProximusAl

CaptSB, 2023-07-31:
Sorry for my earlier rants. I seem to have fixed it now. Simple typo in my
reinstalled config in the end but the issues just build up and I started to
panic!

Keep up the great work!

PR #94 (2023-08-05): Define SuperTimezone and set to false if not running as Add On.
Should fix #93. SuperTimezone is not defined unless the isAddon if statement is run. This causes a NameError Exception when the code is not running under Supervisor.

While it would be possible to put "if SuperTimezone: outp.write(" timezone=""+str(SuperTimezone)+""\n")" in a try->catch block or under another if statement, it's better to have a predefined variable to avoid the exception altogether. Setting it to false should allow your if statement to work as intended.