britkat1980/giv_tcp #95: IndexError: list index out of range

JamesMonkey, 2023-08-01:
Hi there,

First-time user setting up on HA. Ten minutes before the battery level is set, this event was fired and gave the following error:

INFO: Accepting connections at http://localhost:3042.
2023-08-01 02:12:46,291 - Inv1 - read        -  [CRITICAL] - First time running so saving AC Charge status
2023-08-01 02:12:46,300 - Inv1 - read        -  [CRITICAL] - Publishing Home Assistant Discovery messages
2023-08-01 02:12:54,469 - Inv1 - read        -  [CRITICAL] - Publishing Home Assistant Discovery messages
2023-08-01 02:12:58,644 - Inv1 - read        -  [CRITICAL] - Publishing Home Assistant Discovery messages
2023-08-01 02:13:05,872 - Inv1 - read        -  [CRITICAL] - Publishing Home Assistant Discovery messages
2023-08-01 02:15:37,451 - Inv1 - palm_soc    -  [INFO    ] - PALM... PV Automated Load Manager Version:v1.0.0SoC
2023-08-01 02:15:38,843 - Inv1 - palm_soc    -  [INFO    ] - No second array
2023-08-01 02:15:38,843 - Inv1 - palm_soc    -  [INFO    ] - Successful Solcast download.
2023-08-01 02:15:38,845 - Inv1 - palm_soc    -  [INFO    ] - Applying BST offset to Solcast data
Traceback (most recent call last):
  File "/app/GivTCP_1/palm_soc.py", line 704, in <module>
    solcast.update()
  File "/app/GivTCP_1/palm_soc.py", line 619, in update
    pv_est10[i] = int(solcast_data_1['forecasts'][cntr]['pv_estimate10'] * 1000)
IndexError: list index out of range

The Solcast API key is correct as the GivEnergy dashboard uses the same key without issue, showing three days historical and three days predicted. As I am new to this, any additional info you wish for me to provide I'll gladly add.

Thanks!

dandwhelan, 2023-08-02:
I've the same issue. Over the past 3 nights my battery has charged to 100%.
I looked on MQTT explorer and the GivEnergy/ED2248G563/Energy/Today/xx are all active and visible.

I'm on a Gen 2 Hybrid 5kw inverter.

2023-08-02 00:00:09,118 - Inv1 - read        -  [CRITICAL] - Midnight, so resetting Day/Night stats...
2023-08-02 00:00:10,366 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/ED2248G563/Energy/Today/PV_Energy_Today_kWh
2023-08-02 00:00:10,367 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/ED2248G563/Energy/Today/Import_Energy_Today_kWh
2023-08-02 00:00:10,368 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/ED2248G563/Energy/Today/Export_Energy_Today_kWh
2023-08-02 00:00:10,370 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/ED2248G563/Energy/Today/AC_Charge_Energy_Today_kWh
2023-08-02 00:00:10,371 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/ED2248G563/Energy/Today/Invertor_Energy_Today_kWh
2023-08-02 00:00:10,372 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/ED2248G563/Energy/Today/Self_Consumption_Energy_Today_kWh
2023-08-02 00:00:10,373 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/ED2248G563/Energy/Today/Load_Energy_Today_kWh
2023-08-02 00:00:10,374 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/ED2248G563/Energy/Today/Battery_Charge_Energy_Today_kWh
2023-08-02 00:00:10,375 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/ED2248G563/Energy/Today/Battery_Discharge_Energy_Today_kWh
2023-08-02 00:00:10,376 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/ED2248G563/Energy/Today/Battery_Throughput_Today_kWh
2023-08-02 00:00:10,378 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/ED2248G563/Energy/Rates/Import_ppkwh_Today
**2023-08-02 00:00:22,133 - Inv1 - read        -  [CRITICAL] - Midnight, so resetting Day/Night stats...
2023-08-02 00:00:35,136 - Inv1 - read        -  [CRITICAL] - Midnight, so resetting Day/Night stats...
2023-08-02 00:20:34,064 - Inv1 - palm_soc    -  [INFO    ] - PALM... PV Automated Load Manager Version:v1.0.0SoC
2023-08-02 00:20:35,511 - Inv1 - palm_soc    -  [INFO    ] - url_sw = 'https://api.solcast.com.au/rooftop_sites/clearedthisinfonotnormalwords'
2023-08-02 00:20:36,961 - Inv1 - palm_soc    -  [INFO    ] - Successful Solcast download.
2023-08-02 00:20:36,963 - Inv1 - palm_soc    -  [INFO    ] - Applying BST offset to Solcast data
Traceback (most recent call last):
  File "/app/GivTCP_1/palm_soc.py", line 704, in <module>
    solcast.update()
  File "/app/GivTCP_1/palm_soc.py", line 612, in update
    pv_est10[i] = (int(solcast_data_1['forecasts'][cntr]['pv_estimate10'] * 1000) +
IndexError: list index out of range**

cpaius, 2023-08-02:
Having the same issue on GivTCP 2.3.2

cpaius, 2023-08-03:
#Duplicate of #99 . A PR is already on the way #97

closed 2023-08-06 by britkat1980