britkat1980/giv_tcp #111: Version 2.3.3 throws errors when trying to connect to new AIO or Gateway

lh-reg, 2023-08-28:
I have a brand new 13.5kWh GivEnergy AIO with the Giv-Gateway. However, when I try to connect GivTCP I just get errors all the time and no data. I'm using the following:

GivTCP: 2.3.3
AIO: D0.607-A0.607 (latest firmware)
Gateway: D0.0-A0.5 (latest firmware)

I have tried every combination of the AIO_1 and AC_1 flags and all cause the same type of error (just a slightly different error number). I have also tried to sync the Inverter time several times, with no effect. I'm not trying to connect to the Gateway as all the data comes through the battery. The following is the log output from starting the plugin:

2023-08-28 19:21:23,088 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2023-08-28 19:21:23,091 - startup - [INFO] - Supervisor Timezone: Europe/London
2023-08-28 19:21:25,096 - startup - [INFO] - Networks available for scanning are: {0: '192.168.xxx.1'}
2023-08-28 19:21:25,097 - startup - [CRITICAL] - Scanning network for inverters...
2023-08-28 19:21:25,097 - startup - [INFO] - Scanning network (1):192.168.xxx.1
2023-08-28 19:21:26,306 - startup - [INFO] - 2 Inverters found on 192.168.xxx.1 - {1: '192.168.xxx.yyy', 2: '192.168.xxx.zzz'}
2023-08-28 19:21:26,520 - startup - [CRITICAL] - Inverter CHxxxxxxxx which is a Gen 1 - All in One has been found at: 192.168.xxx.yyy
2023-08-28 19:21:26,816 - startup - [CRITICAL] - Inverter GWxxxxxxxx which is a Gen 1 - Gateway has been found at: 192.168.xxx.zzz
2023-08-28 19:21:26,817 - startup - [CRITICAL] - GivTCP isAddon: True
2023-08-28 19:21:26,817 - startup - [CRITICAL] - Config directory already exists
2023-08-28 19:21:26,818 - startup - [CRITICAL] - Running Redis
2023-08-28 19:21:26,818 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2023-08-28 19:21:26,903 - startup - [CRITICAL] - Recreating settings.py for invertor 1
2023-08-28 19:21:26,904 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2023-08-28 19:21:26,904 - startup - [CRITICAL] - Running Invertor read loop every 5s
2023-08-28 19:21:26,905 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2023-08-28 19:21:26,911 - startup - [CRITICAL] - Starting Gunicorn on port 6345
[2023-08-28 19:21:27 +0100] [35] [INFO] Starting gunicorn 21.2.0
[2023-08-28 19:21:27 +0100] [35] [INFO] Listening at: http://0.0.0.0:6345 (35)
[2023-08-28 19:21:27 +0100] [35] [INFO] Using worker: sync
[2023-08-28 19:21:27 +0100] [36] [INFO] Booting worker with pid: 36
[2023-08-28 19:21:27 +0100] [39] [INFO] Booting worker with pid: 39
[2023-08-28 19:21:27 +0100] [40] [INFO] Booting worker with pid: 40
2023-08-28 19:21:27,887 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2023-08-28 19:21:36,164 - Inv1 - read        -  [ERROR   ] - Consecutive failure count= 4 -- (<class 'Exception'>, Exception("Garbage or failed inverter Response: ERROR:-(<class 'KeyError'>, KeyError(IR:110), <traceback object at 0x7fe256fd7c80>)"), <traceback object at 0x7ffaf2d5fa80>)
...
2023-08-28 19:23:03,529 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2023-08-28 19:23:07,985 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 44, in <module>
    logger.debug("Serial Number retrieved: "+GiV_Settings.serial_number)
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
2023-08-28 19:23:54,899 - Inv1 - read        -  [ERROR   ] - Consecutive failure count= 4 -- (<class 'Exception'>, Exception("Garbage or failed inverter Response: ERROR:-(<class 'KeyError'>, KeyError(IR:110), <traceback object at 0x7fe256fe0e80>)"), <traceback object at 0x7ffaf2d85d00>)

Any help would be appreciated.

mauriceatkinson, 2023-09-04:
I have the same issue.

`2023-09-03 18:14:47,119 - Inv1 - read        -  [ERROR   ] - Consecutive failure count= 7 -- (<class 'Exception'>, Exception("Garbage or failed inverter Response: ERROR:-(<class 'KeyError'>, KeyError(HR:243), <traceback object at 0x7f2fb1443380>)"), <traceback object at 0x7faaeefdb940>)
2023-09-03 18:14:48,210 - Inv2 - read        -  [ERROR   ] - Consecutive failure count= 10 -- (<class 'Exception'>, Exception("Garbage or failed inverter Response: ERROR:-(<class 'KeyError'>, KeyError(IR:000), <traceback object at 0x7f4b176db600>)"), <traceback object at 0x7fdb59d52c80>)
2023-09-03 18:14:53,478 - Inv1 - read        -  [ERROR   ] - Consecutive failure count= 8 -- (<class 'Exception'>, Exception("Garbage or failed inverter Response: ERROR:-(<class 'KeyError'>, KeyError(HR:243), <traceback object at 0x7f2fb1444900>)"), <traceback object at 0x7faaeefe5e40>)
2023-09-03 18:14:55,683 - Inv2 - read        -  [ERROR   ] - Consecutive failure count= 11 -- (<class 'Exception'>, Exception("Garbage or failed inverter Response: ERROR:-(<class 'KeyError'>, KeyError(IR:000), <traceback object at 0x7f4b176e8ac0>)"), <traceback object at 0x7fdb5923db00>)
2023-09-03 18:14:55,683 - Inv2 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...`

mimikm, 2023-09-11:
`Traceback (most recent call last):
  File "/app/GivTCP_1/read.py", line 64, in getData
    raise Exception ("Garbage or failed inverter Response: "+ str(plant.result))
Exception: Garbage or failed inverter Response: ERROR:-(<class 'KeyError'>, KeyError(HR:243), <traceback object at 0x7efddf1374c0>)
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/GivTCP_1/read.py", line 1161, in <module>
    globals()[sys.argv[1]]()
  File "/app/GivTCP_1/read.py", line 787, in self_run2
    runAll("True")
  File "/app/GivTCP_1/read.py", line 749, in runAll
    result=getData(full_refresh)
  File "/app/GivTCP_1/read.py", line 77, in getData
    consecFails(e)
  File "/app/GivTCP_1/read.py", line 725, in consecFails
    oldDataCount= pickle.load(inp)
EOFError: Ran out of input`

Same for me

britkat1980, 2023-11-04:
To get GivTCP to work with an AIO, you must set the isAIO config setting to True

closed 2023-11-04 by britkat1980