britkat1980/giv_tcp #113: AM/PM display is incorrect in Smart Home Display

gmckeown, 2023-09-08:
In the smart home display, 12 noon shows as 12am -- this should be 12 pm. The incorrect logic is in updateTimeStamp() in app.js.

A potential fix is to replace:
```javascript
let suffix = 'am';

if (hours > 12) {
  hours -= 12;
  suffix = 'pm';
} else if (hours === 0) {
  hours = 12;
}
```

With:
```javascript
const suffix = (hours < 12) ? 'am': 'pm';
hours = hours % 12;
hours = hours ? hours : 12;
minutes = minutes < 10 ? '0' + minutes : minutes;
```

Another way to pad the minutes:
```javascript
minutes = minutes.toString().padStart(2, '0');
```

Haven't checked whether you've got any JS version constraints or style guidelines, so apologies if any of this doesn't fit in!

Also note that an alternative to the whole function might be to use date.toLocaleTimeString with hour12 = true.

gmckeown, 2023-09-08:
Your repo came up first in a Google search for "givtcp github", so not sure if it was best to raise the issue here or the main givtcp repo... have raised against both.

britkat1980, 2023-11-04:
fixed in 2.4.0

closed 2023-11-04 by britkat1980