britkat1980/giv_tcp #122: Battery SOC changes for no reason

jasont247, 2023-10-06:
GivTCP Target SOC changes for no reason. The battery will be charing from the grid and the Target SOC changes to a random value.

For a workaround I have created an automation so when the GivTCP Target SOC is less than 100% it sets that value back to 100% Set value for Control GivTCP Target SOC

andrzej-r, 2023-10-07:
Is it really random? I noticed occasional sudden drops in SoC when switching from discharging to charging (PV or AC) without any battery power output or temperature rise recorded. See the drop around 9am on 28/9:
[image]

I don't know if this is expected (some kind of SoC calibration?) or if it is linked to battery health. I have a newest inverter/battery firmware installed (v909/3013). Please let me know if this matches what you see.

jasont247, 2023-10-07:
I’m not talking about the battery SOC. I meant the AC charge upper % limit
/ Control GivTCP Target SOC.

andrzej-r, 2023-10-07:
Sorry, I missed the "target" part. No idea why would that change, in my case Target SoC always stays at a value I set it to.

britkat1980, 2023-11-04:
Check tat you have no automations running either in HA, or GivTCP Smart Target, or via the GE portal (Octopus R&D etc...)

andrzej-r, 2023-11-25:
> Is it really random? I noticed occasional sudden drops in SoC when switching from discharging to charging (PV or AC) without any battery power output or temperature rise recorded. See the drop around 9am on 28/9

This is apparently a known firmware issue and GE is working on resolving it.

closed 2024-10-01 by britkat1980