britkat1980/giv_tcp #123: How to unselect time slot values?

andrzej-r, 2023-10-07:
Rather stupid but annoying issue:

When experimenting with inverter settings in Home Assistant I set a charging time slot 2 to certain start/stop times and now I am finding it impossible to unselect them, I can only specify different times (move the charging slot to a time it does not cause problems).

How to get rid of unneeded charging/discharging slots?

gcoan, 2023-10-19:
Set both the start and stop time to 00:00:00?  That’s what 5hey are set to when not set.

Or presumably setting start to the same value as the end time will have the same effect

britkat1980, 2023-11-04:
Timeslots cannot be unset. But you can turn off the schedules for both Charge and Discharge.

closed 2023-11-04 by britkat1980