britkat1980/giv_tcp #125: Can't connect to new system

stefan-west1, 2023-10-13:
Hi all,

Got my system installed today. Trying to run in docker on unraid. Can't seem to get it to connect at all. Below is my docker run command and the resulting logs. I've tried running older versions etc. after reading some other errors but nothing. Ran telnet from unraid and confirmed it can connect to port 8899. I also tried running the container as host and letting it auto find the inverter, and can confirm the same errors and same IP to be sure!

```
docker run
  -d
  --name='GivTCP'
  --net='bridge'
  -e TZ="Europe/London"
  -e HOST_OS="Unraid"
  -e HOST_HOSTNAME="MantleCore"
  -e HOST_CONTAINERNAME="GivTCP"
  -e 'NUMINVERTORS'='1'
  -e 'INVERTOR_IP_1'='192.168.86.12'
  -e 'NUMBATTERIES_1'='1'
  -e 'MQTT_OUTPUT'='True'
  -e 'MQTT_ADDRESS'='192.168.86.31'
  -e 'LOG_LEVEL'='Error'
  -e 'HA_AUTO_D'='True'
  -l net.unraid.docker.managed=dockerman 'britkat/giv_tcp-ma:2.2.4'
62a65a240bf7bfd9b52d67aad2bed687f8b3644f5990fcd90c874a02009dc294

The command finished successfully!
```

```
2023-10-13 21:51:43,014 - startup - [CRITICAL] - SUPERVISOR TOKEN does not exist
2023-10-13 21:51:43,015 - startup - [CRITICAL] - GivTCP isAddon: False
2023-10-13 21:51:43,015 - startup - [CRITICAL] - No config directory exists, so creating it...
2023-10-13 21:51:43,016 - startup - [CRITICAL] - Running Redis
2023-10-13 21:51:43,017 - startup - [CRITICAL] - Running RQ Dashboard on port 9181
2023-10-13 21:51:43,017 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2023-10-13 21:51:43,051 - startup - [CRITICAL] - Recreating settings.py for invertor 1
2023-10-13 21:51:43,053 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2023-10-13 21:51:43,053 - startup - [CRITICAL] - Running Invertor read loop every 5s
2023-10-13 21:51:43,054 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2023-10-13 21:51:43,055 - startup - [CRITICAL] - Starting Gunicorn on port 6345
[2023-10-13 21:51:43 +0100] [15] [INFO] Starting gunicorn 20.1.0
[2023-10-13 21:51:43 +0100] [15] [INFO] Listening at: http://0.0.0.0:6345 (15)
[2023-10-13 21:51:43 +0100] [15] [INFO] Using worker: sync
[2023-10-13 21:51:43 +0100] [18] [INFO] Booting worker with pid: 18
[2023-10-13 21:51:43 +0100] [19] [INFO] Booting worker with pid: 19
[2023-10-13 21:51:43 +0100] [20] [INFO] Booting worker with pid: 20
 * Running on 0.0.0.0:9181
2023-10-13 21:51:43,921 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- 192.168.86.31
2023-10-13 21:52:56,003 - Inv1 - read        -  [ERROR   ] - Consecutive failure count= 4 -- (<class 'Exception'>, Exception("Garbage or failed inverter Response: ERROR:-(<class 'KeyError'>, KeyError(HR:013), <traceback object at 0x1472834b67c0>)"), <traceback object at 0x14d19af92880>)
2023-10-13 21:53:16,581 - Inv1 - read        -  [ERROR   ] - Consecutive failure count= 5 -- (<class 'Exception'>, Exception("Garbage or failed inverter Response: ERROR:-(<class 'KeyError'>, KeyError(HR:013), <traceback object at 0x1472834b6680>)"), <traceback object at 0x14d19af92d40>)
2023-10-13 21:53:37,159 - Inv1 - read        -  [ERROR   ] - Consecutive failure count= 6 -- (<class 'Exception'>, Exception("Garbage or failed inverter Response: ERROR:-(<class 'KeyError'>, KeyError(HR:013), <traceback object at 0x1472834b6700>)"), <traceback object at 0x14d19af934c0>)
2023-10-13 21:53:48,193 - Inv1 - read        -  [ERROR   ] - Consecutive failure count= 7 -- (<class 'Exception'>, Exception("Garbage or failed inverter Response: ERROR:-(<class 'KeyError'>, KeyError(HR:013), <traceback object at 0x1472834b6b80>)"), <traceback object at 0x14d19af93780>)
2023-10-13 21:54:08,769 - Inv1 - read        -  [ERROR   ] - Consecutive failure count= 8 -- (<class 'Exception'>, Exception("Garbage or failed inverter Response: ERROR:-(<class 'KeyError'>, KeyError(HR:013), <traceback object at 0x1472834b6580>)"), <traceback object at 0x14d19af93b40>)
2023-10-13 21:54:29,347 - Inv1 - read        -  [ERROR   ] - Consecutive failure count= 9 -- (<class 'Exception'>, Exception("Garbage or failed inverter Response: ERROR:-(<class 'KeyError'>, KeyError(HR:013), <traceback object at 0x1472834b6740>)"), <traceback object at 0x14d19afc8240>)
```

jasonwfrancis1, 2023-10-15:
I have the same issue, with it installed as a HA add on.

jasonwfrancis1, 2023-10-15:
Found the issue, for the AIO you need to set the number of batteries to 0.

stefan-west1, 2023-10-15:
I really hoped that fixed it but unfortunately not for me. Although it is an all in one I have. I’m sure it’s probably something really simple I’m missing

stefan-west1, 2023-10-15:
Have also tried adding 

`INVERTOR_AIO_1=True `

Along with batteries set to 0 but still no luck unfortunately

sheenweb, 2023-10-22:
I too am having the same issue with it installed on home assistant.
I have tried what has been mentioned above, other posts online suggested setting the time on the inverter which i tried. 

Mine maybe a bit different as my inverter and battery have been swapped over recently, same dongle with no other changes, So my thinking is is the firmware on the GIv kit. Battery is latest, nothing available for the inverter. I reverted to older GivTCP backups, setting it up from scratch. I reverted the core-mosquitto addon too which it is setup for (I think - had a few beers). Updated again to the latest just cannot seem to get it going.

> 2023-10-22 19:58:12,965 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
> 2023-10-22 19:58:12,976 - startup - [INFO] - Supervisor Timezone: Europe/London
> 2023-10-22 19:58:14,987 - startup - [INFO] - Networks available for scanning are: {0: '192.168.0.1'}
> 2023-10-22 19:58:14,987 - startup - [CRITICAL] - Scanning network for inverters...
> 2023-10-22 19:58:14,987 - startup - [INFO] - Scanning network (1):192.168.0.1
> 2023-10-22 19:58:16,211 - startup - [INFO] - Scanning network (2):192.168.0.1
> 2023-10-22 19:58:17,520 - startup - [CRITICAL] - No inverters found...
> 2023-10-22 19:58:17,520 - startup - [CRITICAL] - GivTCP isAddon: True
> 2023-10-22 19:58:17,520 - startup - [CRITICAL] - Config directory already exists
> 2023-10-22 19:58:17,522 - startup - [CRITICAL] - Running Redis
> 2023-10-22 19:58:17,523 - startup - [CRITICAL] - Setting up invertor: 1 of 1
> 2023-10-22 19:58:17,550 - startup - [CRITICAL] - Recreating settings.py for invertor 1
> 2023-10-22 19:58:17,551 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
> 2023-10-22 19:58:17,552 - startup - [CRITICAL] - Running Invertor read loop every 5s
> 2023-10-22 19:58:17,553 - startup - [CRITICAL] - Subscribing MQTT Broker for control
> 2023-10-22 19:58:17,559 - startup - [CRITICAL] - Starting Gunicorn on port 6345
> 2023-10-22 19:58:17,582 - startup - [CRITICAL] - Setting daily charge target forecast job to run at: 00:20
> [2023-10-22 19:58:17 +0100] [55] [INFO] Starting gunicorn 21.2.0
> [2023-10-22 19:58:17 +0100] [55] [INFO] Listening at: http://0.0.0.0:6345 (55)
> [2023-10-22 19:58:17 +0100] [55] [INFO] Using worker: sync
> [2023-10-22 19:58:17 +0100] [57] [INFO] Booting worker with pid: 57
> [2023-10-22 19:58:17 +0100] [56] [INFO] Booting worker with pid: 56
> [2023-10-22 19:58:17 +0100] [58] [INFO] Booting worker with pid: 58
> 2023-10-22 19:58:18,799 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
> 2023-10-22 19:58:23,372 - Inv1 - read        -  [ERROR   ] - Consecutive failure count= 6 -- (<class 'Exception'>, Exception("Garbage or failed inverter Response: ERROR:-(<class 'ValueError'>, ValueError('hour must be in 0..23'), <traceback object at 0x7ff8ba14f600>)"), <traceback object at 0x7f0ee4603980>)

stefan-west1, 2023-10-30:
I think the issue might just be givtcp doesn't support the latest firmware maybe? I can't see any other reason why it wouldn't work

sheenweb, 2023-10-30:
I got it to work, although I changed too much at once so I'm not sure what it was.... which included firewall , VLAN on the home network itself and then home assistant (removed givTCP, removed files manually, done updates to HA and so on). I have my inverter and battery being changed again to new kit in the coming weeks / month or two... if i have the same problem again later on I'll try and troubleshoot it in bitesize chunks and report back if it provides any leads

britkat1980, 2023-11-04:
The ValueError is because one of the time based entities has an invalid time, so using the GE portal check the times for each and update. This should kick GivTCP into life.

stefan-west1, 2023-11-04:
Turns out I had some weird software glitch. I did eventually get it working but it would randomly still give the "garbage or failed invertor" error, which led to gaps in the data (completely screwed up batpred for me too). It also caused random SOC drop-off and eventually stopped me from being able to charge or discharge at all. GivEnergy connected remotely and sorted it all, and since then seems to be working fine

closed 2023-11-04 by stefan-west1