britkat1980/giv_tcp #127: MQTT connection dies and doesn't reconnect

mchugh19, 2023-10-22:
Thanks for givtcp, it's been generally fantastic! I have noticed an issue where every few weeks there's an issue where homeassistant doesn't seem to have any data from givtcp. It looks like givtcp is still talking to the inverter as the `Last_Updated_Time` value is properly updated, and I noticed an error in the log from this time: 

> 2023-10-22 00:14:13,189 - Inv1 - mqtt        -  [ERROR   ] - Error connecting to MQTT Broker: (<class 'ConnectionRefusedError'>, ConnectionRefusedError(111, 'Connection refused'), <traceback object at 0x7f9dd58a>)
2023-10-22 00:14:24,309 - Inv1 - mqtt        -  [ERROR   ] - Error connecting to MQTT Broker: (<class 'ConnectionRefusedError'>, ConnectionRefusedError(111, 'Connection refused'), <traceback object at 0x7f9dd58a>)

I was able to restore connectivity by accessing the `RunAll` url, and saw the 
> 2023-10-22 08:15:24,954 - Inv1 - read        -  [CRITICAL] - Publishing Home Assistant Discovery messages
log message. This seemed to force the mqtt reconnect, and it's been fine since. 

Might it be possible to add additional retry logic in case the mqtt connection drops out?

gcoan, 2023-10-22:
I had the same issue, I found in the morning that the overnight charge hadn’t occurred despite the instructions being sent to the inverters ok. 
found the same connection refused errors in the logs.
In my case the last update time was updated most of the time through the failing period before it too stopped being updated after a couple of hours. I could see the connection refused errors in the home assistant console log but it wasn’t written to the server log file. 
in my case I restarted the givtcp add-on and that restored connectivity ok

PianSom, 2023-10-22:
On a related note, it seems the `online` status in MQTT does not update when comms are lost or givtcp goes down. Might it be possible to implement a LWT-type solution to keep track of problems?

britkat1980, 2023-11-04:
latest version 2.4.0 includes a PR for MQTT will capability, so online topic should update when the connection dies.

closed 2024-10-01 by britkat1980

mchugh19, 2024-10-06:
Hey @britkat1980 

Now running with 3.0.2, I'm seeing a similar unable to connect issue. Logs include
```
2024-10-06 11:44:13,844 - Inv1 - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'ValueError'>, ValueError("time data '2024-10-06T12:44:13.844073+01:00' does not match format '%Y-%m-%dT%H:%M:%S%z'"), <traceback object at 0x7f6c5d02ba80>), 764)
2024-10-06 11:44:13,844 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f6c5d029f40>), 1633)
2024-10-06 11:44:13,844 - Inv1 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1809)
2024-10-06 11:44:46,588 - Inv1 - GivLUT      -  [ERROR   ] - Timed out waiting for regcache
```

gcoan, 2024-10-06:
> Hey @britkat1980
>
>
>
> Now running with 3.0.2, I'm seeing a similar unable to connect issue. Logs include
>
> ```
>
> 2024-10-06 11:44:13,844 - Inv1 - read        -  [ERROR   ] - Error processing Inverter [...]

You do get occasionally communications errors such as this with GivTCP.  Sometimes it's the underlying wifi/LAN network, sometimes it's just the inverter playing up.

As long as they are not too frequent, it's not a major issue, but it is worth adding an automation to detect givtcp stopping working entirely.  Example of the one I wrote is here: https://springfall2008.github.io/batpred/output-data/#givtcp-activity-monitor