britkat1980/giv_tcp #130: Battery voltage for AIO

TimClement, 2023-11-01:
The AIO does not appear to have a battery details topic, but does provide the battery voltage in the raw JSON (and it also shows in the GE app as a battery statistic). Would it be possible to feed this to HA as a sensor if raw input is enabled? It might be of interest to people seeing the "sudden drop in SOC on discharge" issue.

britkat1980, 2023-11-04:
Battery details for AIO are a WIP. Its a completely different data structure, so I need to modify the underlying (un-maintained) library. I'm planning on getting this working before 2.5.0

DJBenson, 2023-11-28:
Came here to ask about this, after discussing with Paul on the groups that the registers are there it's great to hear that you're looking into this @britkat1980.

pbyrne-ms, 2023-12-04:
You can add manually to HA, via MQTT, in the meantime

```
mqtt:
  sensor:
    - name: AIO#1 Battery Voltage
	    unique_id: 6b27bc8d-0b08-4131-85d1-d2304c79226d
	    force_update: true
	    icon: mdi:flash-triangle-outline
	    device_class: voltage
	    state_class: measurement
	    qos: 0
	    unit_of_measurement: "V"
	    state_topic: "aio1/CH<serial>/raw/invertor/v_battery"
    - name: AIO#1 Battery Current
	    unique_id: 10c410c5-1bd4-4e88-8577-666b4e575818
	    force_update: true
	    icon: mdi:current-dc
	    device_class: current
	    state_class: measurement
	    qos: 0
	    unit_of_measurement: "A"
	    state_topic: "aio1/CH<serial>//raw/invertor/i_battery"
```
I've done this and its workign fine for about a month now..

closed 2024-10-01 by britkat1980