britkat1980/giv_tcp #131: cli.py dump_registers missing required argument isAIO

divenal, 2023-11-02:
dump_registers calls refresh_plant(), but that has had a new argument isAIO added to it.
I assume the fix would be to add a new cmd-line flag --aio  via @click ?

Incidentally, the client stuff seems to depend on old versions of pydactic and pymodbus (?) : they've had incompatible api changes in the latest versions, so I had to force a downgrade. But I guess you don't want to upgrade to the latest APIs since that could break existing users.

britkat1980, 2023-11-04:
The underlying library is unmaintained so I've had to modify it, and as I don't use the cli.py I haven't looked at it. If you want to propose a change that'd be great

divenal, 2023-11-06:
I have a change that seems to work. Just need to figure out the security so I can actually push it up to github.

Another minor annoyance is that cli.py assumes that imports are relative to givenergy_modbus, whereas everything else fully-qualifies the imports. But I'm just using cli.py as a check that givenergy_modbus is mostly working. (The only maintained/developed version of this modbus module seems to be in this repository...)

closed 2024-10-01 by britkat1980