britkat1980/giv_tcp #137: Upgrade from 22.3.3 to 2.4.1

ZakalweCheradanine, 2023-11-10:
I'm running HA on a Raspberry Pi 400. Upgraded to 2.4.1 and GivTCP now no longer works (previous version was 2.3.3 which worked fine).
MQTT Explorer verifies that MQTT is working correctly, but it does not see the Givenergy inverter.  The inverter IP address is correct in the GivTCP configuration page. It looks to me that GivTCP is not finding the inverter on the network?

GivTCP log:
2023-11-10 11:17:04,533 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2023-11-10 11:17:04,544 - startup - [INFO] - Supervisor Timezone: Europe/London
2023-11-10 11:17:06,560 - startup - [CRITICAL] - Scanning network for inverters...
2023-11-10 11:17:09,187 - startup - [CRITICAL] - No inverters found...
2023-11-10 11:17:09,202 - startup - [CRITICAL] - Running Redis
2023-11-10 11:17:09,212 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2023/11/10 11:17:09 [notice] 49#49: using the "epoll" event method
2023/11/10 11:17:09 [notice] 49#49: nginx/1.20.2
2023/11/10 11:17:09 [notice] 49#49: OS: Linux 6.1.21-v8
2023/11/10 11:17:09 [notice] 49#49: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/11/10 11:17:09 [notice] 49#49: start worker processes
2023/11/10 11:17:09 [notice] 49#49: start worker process 54
2023/11/10 11:17:09 [notice] 49#49: start worker process 55
2023/11/10 11:17:09 [notice] 49#49: start worker process 56
2023/11/10 11:17:09 [notice] 49#49: start worker process 57
2023-11-10 11:17:09,285 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2023-11-10 11:17:09,286 - startup - [CRITICAL] - Running Invertor read loop every 15s
2023-11-10 11:17:09,294 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2023-11-10 11:17:09,303 - startup - [CRITICAL] - Starting Gunicorn on port 6345
[2023-11-10 11:17:09 +0000] [61] [INFO] Starting gunicorn 21.2.0
[2023-11-10 11:17:09 +0000] [61] [INFO] Listening at: http://0.0.0.0:6345 (61)
[2023-11-10 11:17:09 +0000] [61] [INFO] Using worker: sync
[2023-11-10 11:17:09 +0000] [62] [INFO] Booting worker with pid: 62
[2023-11-10 11:17:10 +0000] [63] [INFO] Booting worker with pid: 63
[2023-11-10 11:17:10 +0000] [64] [INFO] Booting worker with pid: 64
2023-11-10 11:17:11,486 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2023-11-10 11:17:22,308 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...

Thanks in advance

britkat1980, 2023-11-21:
Are you running as and Addon in HA or as a standalone docker container?

pewtertankard, 2023-11-27:
> Are you running as and Addon in HA or as a standalone docker container?

I also have this issue. I’m running it as an add on in HA. Which is the recommended configuration?

ZakalweCheradanine, 2023-11-27:
Its running on a Raspberry Pi 400 Home Assistant Operating System as an add-on.
I ended up reverting to an older version. The latest 2.4.3 works OK. The MQTT broker update to 6.4.0 buggered it up again, which forced a re installation of MQTT. Now working OK.

Idark02, 2023-11-28:
I think I have a similar issue. I have recently installed Givtcp for the first time (version 2.4.2 and then updated to 2.4.3) running as an add on in HA but I have not been able to get my inverter to communicate properly with Givtcp. I have some functionality, I can make manual changes to settings within Givtcp and see my data but it is not working as it should and automations do not pass through. I have tried uninstalling and reinstalling, I have checked my configuration (IP address etc.), tried ticking the option in configuration for old firmware on/off, but on boot up I keep getting a message in the log saying ‘unable to interrogate inverter to get base details’.
Does anyone have any idea how I am able to get around this, or is this a glitch that needs to be ironed out? I have

2023-11-28 02:13:30,788 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2023-11-28 02:13:30,806 - startup - [INFO] - Supervisor Timezone: Europe/London
2023-11-28 02:13:32,828 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2023-11-28 02:13:36,953 - startup - [ERROR] - Gathering inverter details for 192.168.1.222 failed.
2023-11-28 02:13:36,953 - startup - [ERROR] - Unable to interrogate inverter to get base details
2023-11-28 02:13:37,155 - startup - [ERROR] - Gathering inverter details for 192.168.1.222 failed.
2023-11-28 02:13:37,155 - startup - [ERROR] - Unable to interrogate inverter to get base details
2023-11-28 02:13:37,156 - startup - [INFO] - Searching for Inverters again
2023-11-28 02:13:37,157 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2023-11-28 02:13:41,284 - startup - [ERROR] - Gathering inverter details for 192.168.1.222 failed.
2023-11-28 02:13:41,284 - startup - [ERROR] - Unable to interrogate inverter to get base details
2023-11-28 02:13:41,487 - startup - [ERROR] - Gathering inverter details for 192.168.1.222 failed.
2023-11-28 02:13:41,487 - startup - [ERROR] - Unable to interrogate inverter to get base details
2023-11-28 02:13:41,489 - startup - [INFO] - Searching for Inverters again
2023-11-28 02:13:41,489 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2023-11-28 02:13:45,624 - startup - [ERROR] - Gathering inverter details for 192.168.1.222 failed.
2023-11-28 02:13:45,625 - startup - [ERROR] - Unable to interrogate inverter to get base details
2023-11-28 02:13:45,824 - startup - [ERROR] - Gathering inverter details for 192.168.1.222 failed.
2023-11-28 02:13:45,825 - startup - [ERROR] - Unable to interrogate inverter to get base details
2023-11-28 02:13:45,834 - startup - [CRITICAL] - Running Redis
2023-11-28 02:13:45,844 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2023/11/28 02:13:45 [notice] 190#190: using the "epoll" event method
2023/11/28 02:13:45 [notice] 190#190: nginx/1.20.2
2023/11/28 02:13:45 [notice] 190#190: OS: Linux 6.1.59
2023/11/28 02:13:45 [notice] 190#190: getrlimit(RLIMIT_NOFILE): [1073741816](tel:1073741816):[1073741816](tel:1073741816)
2023/11/28 02:13:45 [notice] 190#190: start worker processes
2023/11/28 02:13:45 [notice] 190#190: start worker process 195
2023/11/28 02:13:45 [notice] 190#190: start worker process 196
2023/11/28 02:13:45 [notice] 190#190: start worker process 197
2023/11/28 02:13:45 [notice] 190#190: start worker process 198
2023-11-28 02:13:45,957 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2023-11-28 02:13:45,958 - startup - [CRITICAL] - Running Invertor (192.168.1.222) read loop every 5s
2023-11-28 02:13:45,967 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2023-11-28 02:13:45,978 - startup - [CRITICAL] - Starting Gunicorn on port 6345
[2023-11-28 02:13:46 +0000] [202] [INFO] Starting gunicorn 21.2.0
[2023-11-28 02:13:46 +0000] [202] [INFO] Listening at: http://0.0.0.0:6345/ (202)
[2023-11-28 02:13:46 +0000] [202] [INFO] Using worker: sync
[2023-11-28 02:13:46 +0000] [203] [INFO] Booting worker with pid: 203
[2023-11-28 02:13:47 +0000] [204] [INFO] Booting worker with pid: 204
[2023-11-28 02:13:47 +0000] [205] [INFO] Booting worker with pid: 205
2023-11-28 02:13:49,777 - Inv1 - mqtt_client - [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2023-11-28 02:13:56,521 - Inv1 - read - [CRITICAL] - First time running so saving AC Charge status
2023-11-28 02:13:56,534 - Inv1 - read - [CRITICAL] - Publishing Home Assistant Discovery messages
2023-11-28 05:22:34,931 - Inv1 - read - [ERROR ] - Battery Object empty so skipping
2023-11-28 14:41:28,396 - Inv1 - read - [ERROR ] - Battery Object empty so skipping

ChrisMontford, 2024-02-20:
I think I have also got the same problem, and it took me a day or two to realise that even if the log entry said "Unable to interrogate inverter to get base details" it was working quite happily.

I have got it connected to Predbat, and its working fine, and I can see all the attributes from the inverter updating in Home Assistant.

I have all of this running on a Home Assistant Green.

My logfile looks like this:

 2024-02-20 13:53:40,604 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-02-20 13:53:40,624 - startup - [INFO] - Supervisor Timezone: Europe/London
2024-02-20 13:53:42,651 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-02-20 13:53:43,879 - startup - [INFO] - (<class 'AttributeError'>, AttributeError("'ModbusIOException' object has no attribute 'registers'"), <traceback object at 0xffff86098bc0>)
2024-02-20 13:53:43,879 - startup - [INFO] - 192.168.86.211 is not an EVC
2024-02-20 13:53:45,315 - startup - [ERROR] - Gathering inverter details for 192.168.86.210 failed.
2024-02-20 13:53:45,316 - startup - [ERROR] - Unable to interrogate inverter to get base details
2024-02-20 13:53:45,525 - startup - [ERROR] - Gathering inverter details for 192.168.86.210 failed.
2024-02-20 13:53:45,525 - startup - [ERROR] - Unable to interrogate inverter to get base details
2024-02-20 13:53:45,527 - startup - [INFO] - Searching for Inverters again

Idark02, 2024-02-20:
Have you tried running any automations?

Mine does the same, even though I get the ‘unable to interrogate inverter’ message, I can see/monitor all of my GivEnergy data in HA.

The trouble I have is when I set up automations/make changes virtually nothing works. From memory, there are a few settings I am able to change but most do not.

It seems to be able to read the data but not make changes - I put this down to the unable to interrogate inverter issue.

I have stopped using GivTCP in the hope of a fix/update but there haven’t been any updates in relation to this for a while

Ian

rjenx, 2024-03-29:
I too have this issue, or at least a similar one. I have an automation that sets the target soc based on solar forecast which failed to set the value. When I investigated there were no errors from the automation. Manually trying to set the target_soc value using the Home Assistant service number.set_value did not result in any change. I restarted the addon, HA with no change. target_soc is set at 100% and cannot change it. 

Errors in the Addon when restarting it are:
```
2024-03-29 09:23:07,836 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-03-29 09:23:09,870 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-03-29 09:23:13,741 - startup - [CRITICAL] - Inverter CE2225G256 which is a Gen 1 - AC has been found at: 192.168.1.154
2024-03-29 09:23:13,746 - startup - [CRITICAL] - Running Redis
2024-03-29 09:23:13,757 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024/03/29 09:23:13 [notice] 69#69: using the "epoll" event method
2024/03/29 09:23:13 [notice] 69#69: nginx/1.20.2
2024/03/29 09:23:13 [notice] 69#69: OS: Linux 6.6.20-haos
2024/03/29 09:23:13 [notice] 69#69: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/03/29 09:23:13 [notice] 69#69: start worker processes
2024/03/29 09:23:13 [notice] 69#69: start worker process 73
2024-03-29 09:23:14,116 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-03-29 09:23:14,119 - startup - [CRITICAL] - Running Invertor (192.168.1.154) read loop every 5s
2024-03-29 09:23:14,124 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-03-29 09:23:14,130 - startup - [CRITICAL] - Starting Gunicorn on port 6345
2024-03-29 09:23:14,148 - startup - [CRITICAL] - Serving Web Dashboard from port 3042
2024-03-29 09:23:14,788 - startup - [CRITICAL] - Setting daily charge target forecast job to run at: 23:51
[2024-03-29 09:23:22 +0000] [77] [INFO] Starting gunicorn 21.2.0
[2024-03-29 09:23:22 +0000] [77] [INFO] Listening at: http://0.0.0.0:6345 (77)
[2024-03-29 09:23:22 +0000] [77] [INFO] Using worker: sync
[2024-03-29 09:23:23 +0000] [89] [INFO] Booting worker with pid: 89
[2024-03-29 09:23:23 +0000] [90] [INFO] Booting worker with pid: 90
[2024-03-29 09:23:24 +0000] [91] [INFO] Booting worker with pid: 91
2024-03-29 09:23:40,671 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
 INFO  Accepting connections at http://localhost:3042
2024-03-29 09:24:10,187 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force 
```

rjenx, 2024-03-29:
[givtcp debug file.txt](https://github.com/britkat1980/giv_tcp/files/14802315/givtcp.debug.file.txt)

Log file set to debug.

gcoan, 2024-03-29:
That looks like comms failure to the inverter.  Try restarting mosquito mqtt broker and a reset to defaults on the inverter

rjenx, 2024-03-30:
> That looks like comms failure to the inverter. Try restarting mosquito mqtt broker and a reset to defaults on the inverter

Thanks Gcoan, that seems to have fixed it. I will monitor for a couple of days to make sure.

closed 2024-10-01 by britkat1980