britkat1980/giv_tcp #139: Inverter Time

SheriffOfNotts, 2023-11-12:
Hello, I keep getting the following message in Predbat "Invertor time is 2023-11-12 05:25:57+00:00 AppDeamon time 2023-11-
12 09:50:00.019620+00:00 this is -264.07 minutes skewed, Predbat
tus
may not function correctly, please fix this by updating your inverter or
fixing AppDeamon time zone"

I think it's due to the GivTCP not updating the inverter time for 4 hours. The AIO is hard wired and has had a solid connection during this time.

[image]

Restarting GivTCP resolves the issue. Any ideas?

PianSom, 2023-11-12:
What version of GivTCP are you using with your AIO? I have not managed to get 2.4.1 to work. The last working version on mine is 2.2.4

SheriffOfNotts, 2023-11-12:
2.4.1 - works perfectly apart from this.. Have you set battery to 0?

PianSom, 2023-11-12:
Yes, I have. 

Here's an extract from my docker-compose. I'd be very interested to see what env's you have set, if you wouldn't mind sharing?

```
      - NUMINVERTORS=1   
      - INVERTOR_IP_1=###my AIO IP###
      - INVERTOR_AIO_1=True          
      - NUMBATTERIES_1=0     
      - ```

SheriffOfNotts, 2023-11-14:
I don't use a docker, but the config is:

[image]
[image]

Have you configured MQTT?

PianSom, 2023-11-14:
Thanks for sharing. Seems like maybe the HA add-on is working ok, but not the standalone Docker version. Odd.

Yes, I have a fully working MQTT (works with many other devices, and indeed v2.2.4 of GivTCP).

Looks like I will have to wait for @britkat1980 to have time to take a look. 

PS Sorry for hijacking your issue!

britkat1980, 2023-11-21:
Is this a repeatable error or a one off. I’ve not seen the system time not updating when everything else is working…

closed 2024-10-01 by britkat1980