britkat1980/giv_tcp #141: Serial Number not found AIO

dochtie, 2023-11-17:
Hi,

I have GivTCP running as an add on in HA. For some reason I can’t get it to detect the serial number of my AIO. I’ve tried the combinations of setting my flag to false for AIO and also old firmware.

2023-11-17 07:40:58,540 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2023-11-17 07:40:58,559 - startup - [INFO] - Supervisor Timezone: Europe/London
2023-11-17 07:41:00,583 - startup - [CRITICAL] - Scanning network for inverters...
2023-11-17 07:41:03,048 - startup - [CRITICAL] - No inverters found...
2023-11-17 07:41:03,058 - startup - [CRITICAL] - Running Redis
2023-11-17 07:41:03,069 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2023/11/17 07:41:03 [notice] 49#49: using the "epoll" event method
2023/11/17 07:41:03 [notice] 49#49: nginx/1.20.2
2023/11/17 07:41:03 [notice] 49#49: OS: Linux 6.1.59
2023/11/17 07:41:03 [notice] 49#49: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/11/17 07:41:03 [notice] 49#49: start worker processes
2023/11/17 07:41:03 [notice] 49#49: start worker process 54
2023/11/17 07:41:03 [notice] 49#49: start worker process 55
2023/11/17 07:41:03 [notice] 49#49: start worker process 56
2023/11/17 07:41:03 [notice] 49#49: start worker process 57
2023-11-17 07:41:03,216 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2023-11-17 07:41:03,217 - startup - [CRITICAL] - Running Invertor read loop every 15s
2023-11-17 07:41:03,227 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2023-11-17 07:41:03,238 - startup - [CRITICAL] - Starting Gunicorn on port 6345
[2023-11-17 07:41:04 +0000] [61] [INFO] Starting gunicorn 21.2.0
[2023-11-17 07:41:04 +0000] [61] [INFO] Listening at: http://0.0.0.0:6345 (61)
[2023-11-17 07:41:04 +0000] [61] [INFO] Using worker: sync
[2023-11-17 07:41:04 +0000] [62] [INFO] Booting worker with pid: 62
[2023-11-17 07:41:04 +0000] [63] [INFO] Booting worker with pid: 63
[2023-11-17 07:41:04 +0000] [64] [INFO] Booting worker with pid: 64
2023-11-17 07:41:07,376 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2023-11-17 07:42:47,501 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'

closed 2023-11-17 by dochtie

PsychoV3, 2023-11-17:
I have the exact same issue as a new user of GivTCP

dochtie, 2023-11-17:
Set number of batteries to 0. I overlooked it this time.

PsychoV3, 2023-11-17:
Brill worked first time, was pulling what little hair I have out over this lol

britkat1980, 2023-12-11:
I'm testing an update to ignore NUMBATTERIES for AIO devices, so this problem is masked from user until I can get AIO batteries working.