britkat1980/giv_tcp #142: Force Charge service

sjs-1956, 2023-11-17:
At the moment the sections maximum is 180 mins. Can this be increased to 360 as my batteries take nearly 6 hours to charge if I want to go to 100%?

sjs-1956, 2023-12-26:
Any thoughts/comments on this request?

anthonyangel, 2024-02-02:
The force charge service has an option for `Running`, which I assumed would trigger an indefinite force charge, but it appears to do nothing.

[image]

gcoan, 2024-02-02:
> The force charge service has an option for `Running`, which I assumed would trigger an indefinite force charge, but it appears to do nothing.
> 
> 
> 
> [image]
> 
> 

I set the entity to Running and it works fine for me, but I set charge start time to 00:00:00 and charge end time to 23:59:00 beforehand

CaptSB, 2024-02-03:
Indeed, you need to be within the period of Time Slot 1 for it to work by
just setting it to Running.

Sent from Gmail Mobile

closed 2024-10-01 by britkat1980