britkat1980/giv_tcp #144: Interpreting SOC on AIO battery

TimClement, 2023-11-21:
This may be a question rather than an issue.
The actual capacity of the AIO battery is 15.96Kwh, which corresponds to 100% charge according to GivTCP, but only 13.5Kwh is supposed to be usable, leaving a residual charge of about 2.5Kwh. However, if the battery power cut-off and reserve are left at their default 4% values, GivTCP reports about 0.7Kwh remaining at 4%. SOC diminishes rapidly from some way above this*, and the power output appears to be throttled. Is GivTCP allowing more discharge than GivEnergy expect, or is this an artefact of the reporting?

*About 30% on my installation, giving just under 11Kwh peak to usable lower limit according to GivTCP (and consistent with the integrated discharge power). I'm assuming this is the widely reported calibration issue.

britkat1980, 2023-11-21:
GivTCP reports the Soc % directly from the inverter, but kWh is calculated from the inverter reported Ah capacity and a nominal voltage level. Then the % is applied.
There is nothing that GovTCP does which would allow it to discharge lower than the 4%, it’s the “Normal” behaviour of the inverter as widely seen

TimClement, 2023-11-22:
I have added a couple of integrating utility meters to my HA to track energy charged and discharged in each cycle from charge to charge, based on power in and out. The results from the first cycle strongly support the hypothesis that while the AIO reports its battery capacity as 16Kwh, the BMS restricts the cycles to 13.5Kwh, consistent with their documentation, and the SOC percentage is reported against this. I think the SOC Kwh calculation should reflect this - perhaps by reporting on a 0 to 13.5Kwh scale.

britkat1980, 2023-11-23:
OK thanks, let me take a look and see how best to incorporate the practical upper limit rather than the design limit

PianSom, 2023-11-24:
> The results from the first cycle strongly support the hypothesis that while the AIO reports its battery capacity as 16Kwh, the BMS restricts the cycles to 13.5Kwh, consistent with their [...]

As a counter, on Wed this week I started the day at a reported 5% charge, charged fully, and discharged monotonically back to 5% several hours before midnight (any PV input was eaten by house load). My AIO meters reported battery charge of 14.6kWh and discharge of 14.0kWh.  (This is using latest beta BMS - 24301)

TimClement, 2023-11-24:
Interesting. I haven't ben able to find a BMS version, just the inverter firmware (607), but I'm assuming I'm on a recent official release, with attendant errors in charge management. (For me, the reported SOC collapses if I discharge beyond about 30%.) Having looked a bit further, my HA measurements of charge and discharge are consistent with those logged in the portal. I'm running two cycles per day on Octopus Cosy. This morning, 70% was 9.93KwH discharged. Currently at 5.37Kwh or 36% discharged. If I could get there, these imply full discharge of 14.1Kwh and 14.92Kwh. Yesterday, when I raised this, two full cycles of 9.85 and 9.31Kwh ("full" discharge 14.1Kwh and 13.3Kwh) respectively. On the GivEnergy forum, GE have implied that the reported SOC is a linear interpolation between the maximum and minimum battery voltage from calibration, which would not be a reliable measure except near the limits.

PianSom, 2023-11-24:
BMS version should be on the Software tab of My Inverter on the web portal.

Yes, it was frustration with the random falls in reported charge that drove to the beta program. The current beta version (which I have been on a week, tomorrow) is the first I have used which doesn't have the jumps. My sense is that the reported SOC is more manipulated than a linear interpolation, since I haven't calibrated since the last version (so 0% and 100% should be unchanged) and the jumps have disappeared. I may be completely wrong about that though!

Quite why a 95% charge might imply 14.6/14kWh is beyond me. Like you, I'd thought that 100% would either 13.5 or 15.96 kWh.

TimClement, 2023-11-25:
I found my BMS version, which seems to be 7. 
I also found all the invertor data on the GivEnergy site that is available for download. Plotting battery SOC against charge/discharge energy for today's up and down cycles gives a convincingly straight line (as it should), except at almost full charge on the way up, when it flattens out a bit, and when I hit the discharge limit around 30%, where SOC drops rapidly for very little discharge. (The BMS seems to throttle output at this stage too, so the grid gradually takes over).
Battery voltage is almost independent of SOC on the way up (a bit lower at the start and higher at the end, as expected). A noisy plot on the way down (demand, temperature?) but again dropping rapidly at the bottom.
How did you get on the beta programme? It would be good to have the full battery capacity now the weather is getting cold!

PianSom, 2023-11-25:
Pleading on the community forum and/or an email to support should do the trick. Paul L aka TheDragon is the one person I have interacted with, and he is extremely responsive usually. I asked via the forum.

As I said, I have been on this beta BMS for a week, and it seems good. He was talking about expanding thereat program about now if that was the case - so good timing!

TimClement, 2023-11-25:
Thanks for the advice. Thanks too for all the work you have put into this integration - it's a key part of my energy management automation.

PianSom, 2023-11-25:
Thanks due to @britkat1980 , not me!!!

TimClement, 2023-11-25:
Whoops -  thanks are due to both of you, for patience as well as your other help, but @britkat1980 of course for the integration itself.

closed 2024-10-01 by britkat1980