britkat1980/giv_tcp #145: "No attribute 'serial_number'

JRascagneres, 2023-11-21:
```2023-11-21 15:36:56,329 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2023-11-21 15:36:56,347 - startup - [INFO] - Supervisor Timezone: Europe/London
2023-11-21 15:36:58,366 - startup - [CRITICAL] - Scanning network for inverters...
2023-11-21 15:37:00,802 - startup - [CRITICAL] - No inverters found...
2023-11-21 15:37:00,810 - startup - [CRITICAL] - Running Redis
2023-11-21 15:37:00,821 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2023/11/21 15:37:00 [notice] 49#49: using the "epoll" event method
2023/11/21 15:37:00 [notice] 49#49: nginx/1.20.2
2023/11/21 15:37:00 [notice] 49#49: OS: Linux 6.1.21-v8
2023/11/21 15:37:00 [notice] 49#49: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/11/21 15:37:00 [notice] 49#49: start worker processes
2023/11/21 15:37:00 [notice] 49#49: start worker process 54
2023/11/21 15:37:00 [notice] 49#49: start worker process 55
2023/11/21 15:37:00 [notice] 49#49: start worker process 56
2023/11/21 15:37:00 [notice] 49#49: start worker process 57
2023-11-21 15:37:00,990 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2023-11-21 15:37:00,992 - startup - [CRITICAL] - Running Invertor read loop every 15s
2023-11-21 15:37:01,007 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2023-11-21 15:37:01,053 - startup - [CRITICAL] - Starting Gunicorn on port 6345
2023-11-21 15:37:01,107 - startup - [CRITICAL] - Serving Web Dashboard from port 3000
[2023-11-21 15:37:03 +0000] [61] [INFO] Starting gunicorn 21.2.0
[2023-11-21 15:37:03 +0000] [61] [INFO] Listening at: http://0.0.0.0:6345 (61)
[2023-11-21 15:37:03 +0000] [61] [INFO] Using worker: sync
[2023-11-21 15:37:03 +0000] [76] [INFO] Booting worker with pid: 76
[2023-11-21 15:37:03 +0000] [75] [INFO] Booting worker with pid: 75
[2023-11-21 15:37:03 +0000] [77] [INFO] Booting worker with pid: 77
2023-11-21 15:37:05,282 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
 INFO  Accepting connections at http://localhost:3000
2023-11-21 15:38:44,775 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2023-11-21 15:38:45,390 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
```

Just tried setup and am getting this. Running as HA add-on.

JRascagneres, 2023-11-21:
client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#") AttributeError: type object 'GiV_Settings' has no attribute 'serial_number' 2023-11-21 19:18:46,819 - Inv1 - read - [ERROR ] - 10 failed inverter reads in a row so removing regCache to force update... 2023-11-21 19:22:42,885 - Inv1 - read - [ERROR ] - 10 failed inverter reads in a row so removing regCache to force update... 2023-11-21 19:26:40,587 - Inv1 - read - [ERROR ] - 10 failed inverter reads in a row so removing regCache to force update... 2023-11-21 19:30:38,302 - Inv1 - read - [ERROR ] - 10 failed inverter reads in a row so removing regCache to force update... 2023-11-21 19:34:35,467 - Inv1 - read - [ERROR ] - 10 failed inverter reads in a row so removing regCache to force update... 2023-11-21 19:38:28,838 - Inv1 - read - [ERROR ] - 10 failed inverter reads in a row so removing regCache to force update...

JRascagneres, 2023-11-21:
```
client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#") AttributeError: type object 'GiV_Settings' has no attribute 'serial_number' 2023-11-21 19:18:46,819 - Inv1 - read - [ERROR ] - 10 failed inverter reads in a row so removing regCache to force update... 2023-11-21 19:22:42,885 - Inv1 - read - [ERROR ] - 10 failed inverter reads in a row so removing regCache to force update... 2023-11-21 19:26:40,587 - Inv1 - read - [ERROR ] - 10 failed inverter reads in a row so removing regCache to force update... 2023-11-21 19:30:38,302 - Inv1 - read - [ERROR ] - 10 failed inverter reads in a row so removing regCache to force update... 2023-11-21 19:34:35,467 - Inv1 - read - [ERROR ] - 10 failed inverter reads in a row so removing regCache to force update... 2023-11-21 19:38:28,838 - Inv1 - read - [ERROR ] - 10 failed inverter reads in a row so removing regCache to force update...
```

britkat1980, 2023-11-21:
Can you confirm the ip address you are using from in the config?

JRascagneres, 2023-11-21:
@britkat1980 The IP I'm using is correct. I can visit it and I reach the network controls. Additionally this is the IP that shows in the GivEnergy App

JRascagneres, 2023-11-21:
Is there any additional info that'd help? Debug logs?

JRascagneres, 2023-11-22:
Just does that and then spams failed inverter reads
```
2023-11-21 19:14:54,767 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2023-11-21 19:15:00,820 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2023-11-21 19:16:34,876 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
2023-11-21 19:18:46,819 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2023-11-21 19:22:42,885 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2023-11-21 19:26:40,587 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
```

JRascagneres, 2023-11-22:
There IS some data coming out - Tried to write some code to read it - 
`WH2323XXXX       �FA2323XXXX`
They're my serial numbers.

JRascagneres, 2023-11-22:
@britkat1980 Its a Gen 3 inverter.
If I can message you somehow I can give you TCP dumps of the data? It DOES talk and send data but unsure how to decode it. Did a simply ASCII decode on the packet and can read the serial numbers but thats about as far as I've gotten. I'm not comfortable with publically sharing the the dump

britkat1980, 2023-11-23:
Fixed in 2.4.2

closed 2023-11-23 by britkat1980

PaulHalloran, 2024-02-07:
Thanks so much for creating GivTCP. 

I'm having what looks like the same issue with an All In One and GivTCP v 2.4.3. Seems to be failing because it does not have anything in serial_number:

2024-02-07 22:16:16,143 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-02-07 22:16:16,173 - startup - [INFO] - Supervisor Timezone: Europe/London
2024-02-07 22:16:18,216 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-02-07 22:16:23,243 - startup - [CRITICAL] - No inverters found...
2024-02-07 22:16:23,245 - startup - [INFO] - Searching for Inverters again
2024-02-07 22:16:23,246 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-02-07 22:16:28,298 - startup - [CRITICAL] - No inverters found...
2024-02-07 22:16:28,299 - startup - [INFO] - Searching for Inverters again
2024-02-07 22:16:28,300 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-02-07 22:16:33,401 - startup - [CRITICAL] - No inverters found...
2024-02-07 22:16:33,480 - startup - [CRITICAL] - Running Redis
2024-02-07 22:16:33,581 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024/02/07 22:16:33 [notice] 250#250: using the "epoll" event method
2024/02/07 22:16:33 [notice] 250#250: nginx/1.20.2
2024/02/07 22:16:33 [notice] 250#250: OS: Linux 6.1.63-haos-raspi
2024/02/07 22:16:33 [notice] 250#250: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/02/07 22:16:33 [notice] 250#250: start worker processes
2024/02/07 22:16:33 [notice] 250#250: start worker process 255
2024/02/07 22:16:33 [notice] 250#250: start worker process 256
2024/02/07 22:16:33 [notice] 250#250: start worker process 257
2024/02/07 22:16:33 [notice] 250#250: start worker process 258
2024-02-07 22:16:34,165 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-02-07 22:16:34,168 - startup - [CRITICAL] - Running Invertor (192.168.5.0) read loop every 30s
2024-02-07 22:16:34,199 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-02-07 22:16:34,232 - startup - [CRITICAL] - Starting Gunicorn on port 6345
2024-02-07 22:16:34,279 - startup - [CRITICAL] - Serving Web Dashboard from port 3042
2024-02-07 22:16:34,681 - startup - [CRITICAL] - Setting daily charge target forecast job to run at: 11:20
[2024-02-07 22:16:46 +0000] [262] [INFO] Starting gunicorn 21.2.0
[2024-02-07 22:16:46 +0000] [262] [INFO] Listening at: http://0.0.0.0:6345 (262)
[2024-02-07 22:16:46 +0000] [262] [INFO] Using worker: sync
[2024-02-07 22:16:47 +0000] [275] [INFO] Booting worker with pid: 275
[2024-02-07 22:16:47 +0000] [274] [INFO] Booting worker with pid: 274
[2024-02-07 22:16:47 +0000] [276] [INFO] Booting worker with pid: 276
2024-02-07 22:16:54,878 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
 INFO  Accepting connections at http://localhost:3042
2024-02-07 22:18:35,156 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
2024-02-07 22:19:01,383 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...

JRascagneres, 2024-02-07:
@PaulHalloran Have you set its an AIO in the config and said 0 batteries?

jbm64, 2024-05-23:
I also have the same issue, also with GivTCP V2.4.3, no attribute 'serial_number'
Giv is an AIO, IP set to 192.168.99.42 as found in log, 'Old firmware' is on, tried with number of batteries = 0 and =1, similar result.
Some log info:-

2024-05-23 07:49:03,916 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-05-23 07:49:03,919 - startup - [INFO] - Supervisor Timezone: Europe/London
2024-05-23 07:49:05,925 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-05-23 07:49:10,405 - startup - [CRITICAL] - Inverter CH2316Gxxx which is a Gen 1 - All in One has been found at: 192.168.99.42
2024-05-23 07:49:10,568 - startup - [CRITICAL] - Inverter GW2316Gxxx which is a Gen 1 - Gateway has been found at: 192.168.99.196
2024-05-23 07:49:10,569 - startup - [INFO] - Searching for Inverters again
2024-05-23 07:49:10,572 - startup - [CRITICAL] - Running Redis
2024-05-23 07:49:10,575 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024/05/23 07:49:10 [notice] 69#69: using the "epoll" event method
2024/05/23 07:49:10 [notice] 69#69: nginx/1.20.2
2024/05/23 07:49:10 [notice] 69#69: OS: Linux 6.6.29-haos
2024/05/23 07:49:10 [notice] 69#69: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/05/23 07:49:10 [notice] 69#69: start worker processes
2024/05/23 07:49:10 [notice] 69#69: start worker process 73
2024/05/23 07:49:10 [notice] 69#69: start worker process 74
2024-05-23 07:49:10,598 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-05-23 07:49:10,598 - startup - [CRITICAL] - Running Invertor (192.168.99.42) read loop every 15s
2024-05-23 07:49:10,599 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-05-23 07:49:10,601 - startup - [CRITICAL] - Starting Gunicorn on port 6345
2024-05-23 07:49:10,602 - startup - [CRITICAL] - Serving Web Dashboard from port 3042
[2024-05-23 07:49:10 +0100] [78] [INFO] Starting gunicorn 21.2.0
[2024-05-23 07:49:10 +0100] [78] [INFO] Listening at: http://0.0.0.0:6345 (78)
[2024-05-23 07:49:10 +0100] [78] [INFO] Using worker: sync
[2024-05-23 07:49:10 +0100] [90] [INFO] Booting worker with pid: 90
[2024-05-23 07:49:11 +0100] [92] [INFO] Booting worker with pid: 92
[2024-05-23 07:49:11 +0100] [94] [INFO] Booting worker with pid: 94
2024-05-23 07:49:11,571 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
 UPDATE  The latest version of `serve` is 14.2.3
 INFO  Accepting connections at http://localhost:3042
2024-05-23 07:49:17,656 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2024-05-23 07:50:51,690 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'

also some mqtt log info:-
2024-05-23 07:50:51: New connection from 172.30.32.1:36235 on port 1883.
2024-05-23 07:50:51: New client connected from 172.30.32.1:36235 as GivEnergy_GivTCP_1_Control (p2, c1, k60, u'addons').
2024-05-23 07:50:51: Client GivEnergy_GivTCP_1_Control closed its connection.

PaulHalloran, 2024-05-23:
Hi @jbm64, I'm sure someone will come back with a better answer, but this does not look like the same issue I had, because your inverter is found. Just a guess, but I wonder if your IP address could be wrong. Maybe you have a Gateway as well and it is pointing to the IP address for that (not sure if this is a problem or not)?

jbm64, 2024-05-23:
Hi @PaulHalloran,
IP address appears to be correct as it finds both inverter and gateway...

> 2024-05-23 07:49:10,405 - startup - [CRITICAL] - Inverter CH2316Gxxx which is a Gen 1 - All in One has been found at: 192.168.99.42
> 2024-05-23 07:49:10,568 - startup - [CRITICAL] - Inverter GW2316Gxxx which is a Gen 1 - Gateway has been found at: 192.168.99.196

britkat1980, 2024-05-23:
Hi all, the current prod version of GivTCP doesn't support the gateway, but the latest beta version does. Could you try installing that and seeing if it fixes your problem.
There's been loads of changes and improvements and it will now detect what batteries you have and what firmware etc... so it should (almost) self configure.
You can install them by adding: https://github.com/britkat1980/ha-addons to the HA addons Store repository list.

jbm64, 2024-05-23:
@britkat1980,
Have installed the beta addon as above but with no success:-

2024-05-23 11:36:23,593 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-05-23 11:36:23,595 - startup - [INFO] - Supervisor Timezone: Europe/London
2024-05-23 11:36:25,601 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-05-23 11:36:31,258 - startup - [INFO] - (<class 'AttributeError'>, AttributeError("'ModbusIOException' object has no attribute 'registers'"), <traceback object at 0x7f742e29ae80>)
2024-05-23 11:36:31,259 - startup - [INFO] - 192.168.99.244 is not an EVC
2024-05-23 11:36:32,739 - startup - [CRITICAL] - Inverter CH2316Gxxx which is a Gen 1 - All in One has been found at: 192.168.99.42
2024-05-23 11:36:32,920 - startup - [CRITICAL] - Inverter GW2316Gxxx which is a Gen 1 - Gateway has been found at: 192.168.99.196
2024-05-23 11:36:32,920 - startup - [INFO] - Searching for Inverters again
2024-05-23 11:36:32,922 - startup - [CRITICAL] - Running Redis
2024-05-23 11:36:32,924 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024/05/23 11:36:32 [notice] 69#69: using the "epoll" event method
2024/05/23 11:36:32 [notice] 69#69: nginx/1.20.2
2024/05/23 11:36:32 [notice] 69#69: OS: Linux 6.6.29-haos
2024/05/23 11:36:32 [notice] 69#69: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/05/23 11:36:32 [notice] 69#69: start worker processes
2024/05/23 11:36:32 [notice] 69#69: start worker process 73
2024/05/23 11:36:32 [notice] 69#69: start worker process 74
2024-05-23 11:36:32,948 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-05-23 11:36:32,948 - startup - [CRITICAL] - Running Invertor (192.168.99.42) read loop every 30s
2024-05-23 11:36:32,950 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-05-23 11:36:32,950 - startup - [CRITICAL] - Starting Gunicorn on port 6345
[2024-05-23 11:36:33 +0100] [78] [INFO] Starting gunicorn 21.2.0
[2024-05-23 11:36:33 +0100] [78] [INFO] Listening at: http://0.0.0.0:6345 (78)
[2024-05-23 11:36:33 +0100] [78] [INFO] Using worker: sync
[2024-05-23 11:36:33 +0100] [79] [INFO] Booting worker with pid: 79
[2024-05-23 11:36:33 +0100] [82] [INFO] Booting worker with pid: 82
[2024-05-23 11:36:33 +0100] [83] [INFO] Booting worker with pid: 83
2024-05-23 11:36:33,868 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2024-05-23 11:38:13,996 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
2024-05-23 11:38:21,695 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...

and mqtt:-
2024-05-23 11:35:10: New connection from 172.30.32.1:37317 on port 1883.
2024-05-23 11:35:10: New client connected from 172.30.32.1:37317 as GivEnergy_GivTCP_1_Control (p2, c1, k60, u'addons').
2024-05-23 11:35:10: Client GivEnergy_GivTCP_1_Control closed its connection.

tried with batteries = 0 and =1, no difference

britkat1980, 2024-05-23:
That’s really odd. Can you turn the log level up to Debug and send me the logs

jbm64, 2024-05-23:
@britkat1980 
debug lines are limited, this is what's in there. Where is the option to log more lines?

2024-05-23 13:02:16,389 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
2024-05-23 13:02:16,391 - Inv1 - mqtt_client -  [DEBUG   ] - Connecting to broker(sub): core-mosquitto
2024-05-23 13:02:16,465 - Inv1 - mqtt_client -  [DEBUG   ] - connected OK Returned code=0
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
2024-05-23 13:02:20,348 - Inv1 - read        -  [DEBUG   ] - ----------------------------Starting----------------------------
2024-05-23 13:02:20,349 - Inv1 - read        -  [DEBUG   ] - Getting All Registers
2024-05-23 13:02:20,350 - Inv1 - read        -  [DEBUG   ] - Connecting to: 192.168.99.42
2024-05-23 13:02:20,352 - Inv1 - queue       -  [DEBUG   ] - Pushed job aa2bc9c8-9b5d-4159-814e-b5268c8d86e9 into GivTCP_1
2024-05-23 13:02:20,514 - Inv1 - sync        -  [DEBUG   ] - Connection to Modbus server established. Socket ('192.168.99.103', 57779)
2024-05-23 13:02:20,514 - Inv1 - transaction -  [DEBUG   ] - Current transaction state - IDLE
2024-05-23 13:02:20,514 - Inv1 - transaction -  [DEBUG   ] - Running transaction 1
2024-05-23 13:02:20,515 - Inv1 - sync        -  [DEBUG   ] - New Transaction state 'SENDING'
2024-05-23 13:02:20,674 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2024-05-23 13:02:20,675 - Inv1 - payload     -  [DEBUG   ] - [b'\x00\x00', b'\x00\x00', b'\x00\x00', b'\x00\x8a']
2024-05-23 13:02:20,675 - Inv1 - transaction -  [DEBUG   ] - Adding transaction 1
2024-05-23 13:02:20,676 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2024-05-23 13:02:21,177 - Inv1 - transaction -  [DEBUG   ] - Current transaction state - TRANSACTION_COMPLETE
2024-05-23 13:02:21,178 - Inv1 - transaction -  [DEBUG   ] - Running transaction 2
2024-05-23 13:02:21,180 - Inv1 - sync        -  [DEBUG   ] - New Transaction state 'SENDING'
2024-05-23 13:02:21,336 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2024-05-23 13:02:21,337 - Inv1 - payload     -  [DEBUG   ] - [b'\x00\x00', b'\x00\x00', b'\x00\x00', b'\x00\x8a']
2024-05-23 13:02:21,338 - Inv1 - transaction -  [DEBUG   ] - Adding transaction 2
2024-05-23 13:02:21,339 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2024-05-23 13:02:21,840 - Inv1 - transaction -  [DEBUG   ] - Current transaction state - TRANSACTION_COMPLETE
2024-05-23 13:02:21,841 - Inv1 - transaction -  [DEBUG   ] - Running transaction 3
2024-05-23 13:02:21,841 - Inv1 - sync        -  [DEBUG   ] - New Transaction state 'SENDING'
2024-05-23 13:02:21,996 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2024-05-23 13:02:21,997 - Inv1 - payload     -  [DEBUG   ] - [b'\x00\x00', b'\x00\x00', b'\x00\x00', b'\x00\x8a']
2024-05-23 13:02:21,998 - Inv1 - transaction -  [DEBUG   ] - Adding transaction 3
2024-05-23 13:02:21,999 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2024-05-23 13:02:22,501 - Inv1 - transaction -  [DEBUG   ] - Current transaction state - TRANSACTION_COMPLETE
2024-05-23 13:02:22,502 - Inv1 - transaction -  [DEBUG   ] - Running transaction 4
2024-05-23 13:02:22,503 - Inv1 - sync        -  [DEBUG   ] - New Transaction state 'SENDING'
2024-05-23 13:02:22,657 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2024-05-23 13:02:22,658 - Inv1 - payload     -  [DEBUG   ] - [b'\x00\x00', b'\x00\x00', b'\x00\x00', b'\x00\x8a']
2024-05-23 13:02:22,658 - Inv1 - transaction -  [DEBUG   ] - Adding transaction 4
2024-05-23 13:02:22,658 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2024-05-23 13:02:23,160 - Inv1 - transaction -  [DEBUG   ] - Current transaction state - TRANSACTION_COMPLETE
2024-05-23 13:02:23,160 - Inv1 - transaction -  [DEBUG   ] - Running transaction 5
2024-05-23 13:02:23,161 - Inv1 - sync        -  [DEBUG   ] - New Transaction state 'SENDING'
2024-05-23 13:02:23,318 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
[... cut ...]
2024-05-23 13:02:56,721 - Inv1 - sync        -  [DEBUG   ] - New Transaction state 'SENDING'
2024-05-23 13:02:56,875 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2024-05-23 13:02:56,876 - Inv1 - payload     -  [DEBUG   ] - [b'\x00\x00', b'\x00\x00', b'\x00\x00', b'\x00\x8a']
2024-05-23 13:02:56,877 - Inv1 - transaction -  [DEBUG   ] - Adding transaction 5
2024-05-23 13:02:56,878 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'

jbm64, 2024-06-03:
@britkat1980 
All up and working, uninstalled & reinstalled, my error, didn't install beta first time round, thank you.

chris031, 2024-07-12:
This Beta also worked for me and removes my previous error: 👍
- 10 failed inverter reads in a row so removing regCache to force update...

I only now have this error: 🧐 any idea how to resolve:
_inverter Update failed so using last known good data from cache: ('KeyError', 'read.py', 1420)
2024-07-12 11:22:41,426 - Inv1 - read        -  [DEBUG   ] - processData result: processData Error processing registers: ('KeyError', 'read.py', 1420)_

A larger piece of the log

2024-07-12 11:22:41,098 - Inv1 - read        -  [DEBUG   ] - Data cleansed for: Battery_Details
2024-07-12 11:22:41,340 - Inv1 - read        -  [DEBUG   ] - Data cleansed for: Power
2024-07-12 11:22:41,340 - Inv1 - read        -  [DEBUG   ] - Data cleansed for: Power
2024-07-12 11:22:41,424 - Inv1 - read        -  [ERROR   ] - **inverter Update failed so using last known good data from cache: ('KeyError', 'read.py', 1420)
2024-07-12 11:22:41,426 - Inv1 - read        -  [DEBUG   ] - processData result: processData Error processing registers: ('KeyError', 'read.py', 1420)**
2024-07-12 11:22:41,427 - Inv1 - read        -  [DEBUG   ] - Running pubFromPickle
2024-07-12 11:22:41,430 - Inv1 - read        -  [DEBUG   ] - Converting Model to publish safe string
2024-07-12 11:22:41,431 - Inv1 - read        -  [DEBUG   ] - Converting datetime to publish safe string
2024-07-12 11:22:41,433 - Inv1 - read        -  [DEBUG   ] - Dealt with invertor
2024-07-12 11:22:41,434 - Inv1 - read        -  [DEBUG   ] - Dealt with HV_Battery_Stacks
2024-07-12 11:22:41,435 - Inv1 - read        -  [DEBUG   ] - Dealt with Meter_ID_1
2024-07-12 11:22:41,435 - Inv1 - read        -  [DEBUG   ] - Dealt with Meter_ID_2
2024-07-12 11:22:41,436 - Inv1 - read        -  [DEBUG   ] - Dealt with meters
2024-07-12 11:22:41,437 - Inv1 - read        -  [DEBUG   ] - Dealt with raw
2024-07-12 11:22:41,437 - Inv1 - read        -  [DEBUG   ] - Dealt with Battery_Details
2024-07-12 11:22:41,438 - Inv1 - read        -  [DEBUG   ] - Dealt with Power
2024-07-12 11:22:41,439 - Inv1 - read        -  [DEBUG   ] - Dealt with Power
2024-07-12 11:22:41,440 - Inv1 - read        -  [DEBUG   ] - Dealt with TC2337G258
2024-07-12 11:22:41,440 - Inv1 - read        -  [DEBUG   ] - Dealt with Meter_ID1
2024-07-12 11:22:41,441 - Inv1 - read        -  [DEBUG   ] - Dealt with Meter_ID2
2024-07-12 11:22:41,442 - Inv1 - read        -  [DEBUG   ] - Dealt with Meter_Details
2024-07-12 11:22:41,442 - Inv1 - read        -  [DEBUG   ] - Dealt with Today
2024-07-12 11:22:41,443 - Inv1 - read        -  [DEBUG   ] - Dealt with Total
2024-07-12 11:22:41,444 - Inv1 - read        -  [DEBUG   ] - Dealt with Energy
2024-07-12 11:22:41,444 - Inv1 - read        -  [DEBUG   ] - Dealt with Timeslots
2024-07-12 11:22:41,445 - Inv1 - read        -  [DEBUG   ] - Dealt with Control
2024-07-12 11:22:41,446 - Inv1 - read        -  [DEBUG   ] - Dealt with Stats

britkat1980, 2024-07-12:
When moving to Beta/Dev you will need to delete all pkl and touch files in the config/GivTCP folder. There is a breaking change there. I think it’s specifically the ratedata file

>
> This Beta also worked for me and removes my previous error: 👍
>
> 10 failed inverter reads in a row so removing regCache to force update...
>
>
> I only now have this error: 🧐 any idea how to resolve:
> inverter Update failed so using last known good data from cache: ('KeyError', 'read.py', 1420)
2024-07-12 11:22:41,426 - Inv1 - read - [DEBUG ] - processData result: processData Error processing registers: ('KeyError', 'read.py', 1420)
>
>
> A larger piece of the log
>
>
> 2024-07-12 11:22:41,098 - Inv1 - read - [DEBUG ] - Data cleansed for: Battery_Details
2024-07-12 11:22:41,340 - Inv1 - read - [DEBUG ] - Data cleansed for: Power
2024-07-12 11:22:41,340 - Inv1 - read - [DEBUG ] - Data cleansed for: Power
2024-07-12 11:22:41,424 - Inv1 - read - [ERROR ] -  inverter Update failed so using last known good data from cache: ('KeyError', 'read.py', 1420)
2024-07-12 11:22:41,426 - Inv1 - read - [DEBUG ] - processData result: processData Error processing registers: ('KeyError', 'read.py', 1420)
2024-07-12 11:22:41,427 - Inv1 - read - [DEBUG ] - Running pubFromPickle
2024-07-12 11:22:41,430 - Inv1 - read - [DEBUG ] - Converting Model to publish safe string
2024-07-12 11:22:41,431 - Inv1 - read - [DEBUG ] - Converting datetime to publish safe string
2024-07-12 11:22:41,433 - Inv1 - read - [DEBUG ] - Dealt with invertor
2024-07-12 11:22:41,434 - Inv1 - read - [DEBUG ] - Dealt with HV_Battery_Stacks
2024-07-12 11:22:41,435 - Inv1 - read - [DEBUG ] - Dealt with Meter_ID_1
2024-07-12 11:22:41,435 - Inv1 - read - [DEBUG ] - Dealt with Meter_ID_2
2024-07-12 11:22:41,436 - Inv1 - read - [DEBUG ] - Dealt with meters
2024-07-12 11:22:41,437 - Inv1 - read - [DEBUG ] - Dealt with raw
2024-07-12 11:22:41,437 - Inv1 - read - [DEBUG ] - Dealt with Battery_Details
2024-07-12 11:22:41,438 - Inv1 - read - [DEBUG ] - Dealt with Power
2024-07-12 11:22:41,439 - Inv1 - read - [DEBUG ] - Dealt with Power
2024-07-12 11:22:41,440 - Inv1 - read - [DEBUG ] - Dealt with TC2337G258
2024-07-12 11:22:41,440 - Inv1 - read - [DEBUG ] - Dealt with Meter_ID1
2024-07-12 11:22:41,441 - Inv1 - read - [DEBUG ] - Dealt with Meter_ID2
2024-07-12 11:22:41,442 - Inv1 - read - [DEBUG ] - Dealt with Meter_Details
2024-07-12 11:22:41,442 - Inv1 - read - [DEBUG ] - Dealt with Today
2024-07-12 11:22:41,443 - Inv1 - read - [DEBUG ] - Dealt with Total
2024-07-12 11:22:41,444 - Inv1 - read - [DEBUG ] - Dealt with Energy
2024-07-12 11:22:41,444 - Inv1 - read - [DEBUG ] - Dealt with Timeslots
2024-07-12 11:22:41,445 - Inv1 - read - [DEBUG ] - Dealt with Control
2024-07-12 11:22:41,446 - Inv1 - read - [DEBUG ] - Dealt with Stats

brianainger, 2024-07-23:
hi,
many thanks i installed this as i was getting the 10 attempts message with givtcp
this version worked first time.
just a quicky i have an aio and gateway
i see the aio but not the gateway
di i need to put the ip somewhere ?
thanks again

zedr32, 2024-12-31:
First of all, thank you for creating this integration—it’s been working flawlessly for the past few months. Although it took some effort to set up initially, it has been rock-solid until now.

I’m running an AIO, and unfortunately, at around 12:30 AM this morning (while I was asleep), I lost all telemetry from my AIO.

When restarting GivTCP, I encounter the following error described in the first post:

AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'

Notes and steps I've Tried So Far:
Running version: 2.4.9
Rebooted Home Assistant.
Rebooted the AIO by cutting power to the house, turning off the battery, and flipping all breakers on the switchboard.
Confirmed communication with the inverter at 192.168.1.192 and the battery at 192.168.1.54.
Reinstalled GivTCP.
Confirmed the battery is set to 0.
Rebuilt the configuration.
Unfortunately, none of these steps have resolved the issue, and I’m still encountering the same error.

I’m completely stumped and can’t pinpoint what might have changed. The only possibility I can think of is a potential firmware update on the GivEnergy side, but I’m not sure how to check for this.

Does anyone have any suggestions or insights on what might be causing this issue? Any help would be greatly appreciated!

```
Dump of the log as follows
2024-12-31 11:34:54,867 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-12-31 11:34:54,869 - startup - [INFO] - Supervisor Timezone: Australia/Adelaide
2024-12-31 11:34:56,875 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-12-31 11:35:00,671 - startup - [CRITICAL] - Inverter GW2418G087 which is a Gen 1 - Gateway has been found at: 192.168.1.54
2024-12-31 11:35:00,860 - startup - [CRITICAL] - Inverter CG2340G305 which is a Gen 1 - All in One has been found at: 192.168.1.192
2024-12-31 11:35:00,860 - startup - [INFO] - Searching for Inverters again
2024-12-31 11:35:00,861 - startup - [CRITICAL] - Running Redis
2024-12-31 11:35:00,862 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024/12/31 11:35:00 [notice] 69#69: using the "epoll" event method
2024/12/31 11:35:00 [notice] 69#69: nginx/1.20.2
2024/12/31 11:35:00 [notice] 69#69: OS: Linux 6.6.54-haos
2024/12/31 11:35:00 [notice] 69#69: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/12/31 11:35:00 [notice] 69#69: start worker processes
2024/12/31 11:35:00 [notice] 69#69: start worker process 73
2024/12/31 11:35:00 [notice] 69#69: start worker process 74
2024/12/31 11:35:00 [notice] 69#69: start worker process 75
2024/12/31 11:35:00 [notice] 69#69: start worker process 76
2024-12-31 11:35:00,877 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-12-31 11:35:00,877 - startup - [CRITICAL] - Running Invertor (192.168.1.192) read loop every 15s
2024-12-31 11:35:00,878 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-12-31 11:35:00,878 - startup - [CRITICAL] - Starting Gunicorn on port 6345
[2024-12-31 11:35:01 +1030] [80] [INFO] Starting gunicorn 23.0.0
[2024-12-31 11:35:01 +1030] [80] [INFO] Listening at: http://0.0.0.0:6345 (80)
[2024-12-31 11:35:01 +1030] [80] [INFO] Using worker: sync
[2024-12-31 11:35:01 +1030] [81] [INFO] Booting worker with pid: 81
[2024-12-31 11:35:01 +1030] [84] [INFO] Booting worker with pid: 84
[2024-12-31 11:35:01 +1030] [85] [INFO] Booting worker with pid: 85
2024-12-31 11:35:01,752 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2024-12-31 11:36:41,879 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2297, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1686, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2100, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3142, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3814, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3934, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
```

KingDigweed, 2024-12-31:
No way! I have the exact same issue and symptoms as @zedr32, although I've got a Gen3 Hybrid inverter and 2x 5.2kWh batteries. Just a load of errors and a crash spat out at the turn of midnight.

[Link to Pastebin of logs because it's really long](https://pastebin.com/jmFgP6gU)

I then thought I ought to move to V3 of the addon, but I just get stuck waiting for an inverter to be discovered and I can't access the webgui to progress or configure as a result. :( Any ideas @britkat1980? Thank you for your work on the addon and happy new year to you.

```2024-12-31 01:18:23,148 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2024-12-31 01:18:23,158 - startup     -  [INFO    ] - Network Found: 192.168.0.1/16
2024-12-31 01:18:25,158 - startup     -  [INFO    ] - Searching for Inverters
2024-12-31 01:18:25,159 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...

JRascagneres, 2024-12-31:
Everyone is hitting it. Looking at it now. Tried givtcp v3 and see this more obvious error:
```
2024-12-31T01:46:40.215504+00:00' does not match format '%Y-%m-%dT%H:%M:%S%z'
``

Seems to be an issue with the date

zedr32, 2024-12-31:
Well, it's a shame but also positive that I'm not the only one experiencing this issue 😆 

Weirdly, with V3 it would always throw errors that no inverters are found. However, further down in the logs it would attempt to communicate with the gateway on the IP assigned as Inverter_IP_1 and all was well afterwards.

So I'm not entirely sure how that initial auto discovery works but i never worried about it. 🤷

KingDigweed, 2024-12-31:
Ah OK, that's handy to know at least. I wonder if my subnet being /16 instead of /24 is making the scanning process take a massive amount of time, hence never progressing... Which is annoying because my inverter is only at 192.168.0.66. Might have to open a new issue as I'm guessing I should really be on V3 anyway!

gcoan, 2024-12-31:
I too had GivTCP fail at 00:10, although with different errors 

```
2024-12-31 02:21:42,241 - G - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'ValueError'>, ValueError("time data '2024-12-31T02:21:42.233111+00:00' does not match format '%Y-%m-%dT%H:%M:%S%z'"), <traceback object at 0x7f22dbdcbf80>), 1687)
2024-12-31 02:21:42,242 - G - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825)
```

After lots of failed attempts to work out the root cause, trying both the dev and beta versions of GivTCP 3 (I was already on 3.0.4), I figured that the problem was the inverter date/time.

Re-syncing the inverter time from the givenergy portal to each of my inverters in turn fixed it - see https://github.com/britkat1980/giv_tcp/issues/318

JRascagneres, 2024-12-31:
> I too had GivTCP fail at 00:10, although with different errors
> 
> ```
> 2024-12-31 02:21:42,241 - G - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'ValueError'>, ValueError("time data '2024-12-31T02:21:42.233111+00:00' does not match format '%Y-%m-%dT%H:%M:%S%z'"), <traceback object at 0x7f22dbdcbf80>), 1687)
> 2024-12-31 02:21:42,242 - G - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825)
> ```
> 
> After lots of failed attempts to work out the root cause, trying both the dev and beta versions of GivTCP 3 (I was already on 3.0.4), I figured that the problem was the inverter date/time.
> 
> Re-syncing the inverter time from the givenergy portal to each of my inverters in turn fixed it - see #318
> 
> @JRascagneres I think you've got the same issue I had

Oh my god thank you! Thats fixed... This fixed v2 too...

KingDigweed, 2024-12-31:
Yep, confirmed this fixed for me too. Nice one, thank you.

gcoan, 2024-12-31:
Two hours of my life spent trying to debug it

Some kind of inverter Y2K bug if several of us had the same issue?

JRascagneres, 2024-12-31:
Yeah, seems like the date went wrong. See my comment on this issue - https://github.com/britkat1980/giv_tcp/issues/318 - A bit of TLDR of my investigation