britkat1980/giv_tcp #150: 2.4.2 and 2.4.3 issues...

SheriffOfNotts, 2023-11-24:
I've just upgraded to 2.4.2 and then 2.4.3 and getting the following. I've uninstalled, reinstalled and no change..Any ideas?

2023-11-24 16:37:11,380 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2023-11-24 16:37:11,390 - startup - [INFO] - Supervisor Timezone: Europe/London
2023-11-24 16:37:13,405 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2023-11-24 16:37:16,991 - startup - [CRITICAL] - Inverter GW2315G079 which is a Gen 1 - Gateway has been found at: 192.168.0.46
2023-11-24 16:37:17,186 - startup - [CRITICAL] - Inverter CH2314G037 which is a Gen 1 - All in One has been found at: 192.168.0.119
2023-11-24 16:37:17,187 - startup - [INFO] - Searching for Inverters again
2023-11-24 16:37:17,189 - startup - [CRITICAL] - Running Redis
2023-11-24 16:37:17,193 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2023/11/24 16:37:17 [notice] 69#69: using the "epoll" event method
2023/11/24 16:37:17 [notice] 69#69: nginx/1.20.2
2023/11/24 16:37:17 [notice] 69#69: OS: Linux 6.1.59
2023/11/24 16:37:17 [notice] 69#69: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/11/24 16:37:17 [notice] 69#69: start worker processes
2023/11/24 16:37:17 [notice] 69#69: start worker process 73
2023/11/24 16:37:17 [notice] 69#69: start worker process 74
2023-11-24 16:37:17,307 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2023-11-24 16:37:17,308 - startup - [CRITICAL] - Running Invertor (192.168.0.119) read loop every 15s
2023-11-24 16:37:17,311 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2023-11-24 16:37:17,319 - startup - [CRITICAL] - Starting Gunicorn on port 6345
2023-11-24 16:37:17,328 - startup - [CRITICAL] - Serving Web Dashboard from port 3042
[2023-11-24 16:37:18 +0000] [78] [INFO] Starting gunicorn 21.2.0
[2023-11-24 16:37:18 +0000] [78] [INFO] Listening at: http://0.0.0.0:6345 (78)
[2023-11-24 16:37:18 +0000] [78] [INFO] Using worker: sync
[2023-11-24 16:37:18 +0000] [90] [INFO] Booting worker with pid: 90
[2023-11-24 16:37:18 +0000] [91] [INFO] Booting worker with pid: 91
[2023-11-24 16:37:19 +0000] [92] [INFO] Booting worker with pid: 92
2023-11-24 16:37:24,237 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
 INFO  Accepting connections at http://localhost:3042
2023-11-24 16:37:30,640 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache
2023-11-24 16:37:36,568 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache
2023-11-24 16:37:57,519 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache
2023-11-24 16:38:18,485 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache
2023-11-24 16:38:39,367 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache
2023-11-24 16:39:00,333 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache
2023-11-24 16:39:04,358 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.

SheriffOfNotts, 2023-11-28:
Restored from prior backup and holding off upgrading.

Idark02, 2023-11-28:
I think I have a similar issue. I have recently installed Givtcp for the first time (version 2.4.2 and then updated to 2.4.3) but I have not been able to get my inverter to communicate properly with Givtcp. I have some functionality, I can make manual changes to settings within Givtcp and see my data but it is not working as it should and automations do not pass through. I have tried uninstalling and reinstalling, I have checked my configuration (IP address etc.), tried ticking the option in configuration for old firmware on/off, but on boot up I keep getting a message in the log saying ‘unable to interrogate inverter to get base details’.
Does anyone have any idea how I am able to get around this, or is this a glitch that needs to be ironed out?

2023-11-28 02:13:30,788 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2023-11-28 02:13:30,806 - startup - [INFO] - Supervisor Timezone: Europe/London
2023-11-28 02:13:32,828 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2023-11-28 02:13:36,953 - startup - [ERROR] - Gathering inverter details for 192.168.1.222 failed.
2023-11-28 02:13:36,953 - startup - [ERROR] - Unable to interrogate inverter to get base details
2023-11-28 02:13:37,155 - startup - [ERROR] - Gathering inverter details for 192.168.1.222 failed.
2023-11-28 02:13:37,155 - startup - [ERROR] - Unable to interrogate inverter to get base details
2023-11-28 02:13:37,156 - startup - [INFO] - Searching for Inverters again
2023-11-28 02:13:37,157 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2023-11-28 02:13:41,284 - startup - [ERROR] - Gathering inverter details for 192.168.1.222 failed.
2023-11-28 02:13:41,284 - startup - [ERROR] - Unable to interrogate inverter to get base details
2023-11-28 02:13:41,487 - startup - [ERROR] - Gathering inverter details for 192.168.1.222 failed.
2023-11-28 02:13:41,487 - startup - [ERROR] - Unable to interrogate inverter to get base details
2023-11-28 02:13:41,489 - startup - [INFO] - Searching for Inverters again
2023-11-28 02:13:41,489 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2023-11-28 02:13:45,624 - startup - [ERROR] - Gathering inverter details for 192.168.1.222 failed.
2023-11-28 02:13:45,625 - startup - [ERROR] - Unable to interrogate inverter to get base details
2023-11-28 02:13:45,824 - startup - [ERROR] - Gathering inverter details for 192.168.1.222 failed.
2023-11-28 02:13:45,825 - startup - [ERROR] - Unable to interrogate inverter to get base details
2023-11-28 02:13:45,834 - startup - [CRITICAL] - Running Redis
2023-11-28 02:13:45,844 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2023/11/28 02:13:45 [notice] 190#190: using the "epoll" event method
2023/11/28 02:13:45 [notice] 190#190: nginx/1.20.2
2023/11/28 02:13:45 [notice] 190#190: OS: Linux 6.1.59
2023/11/28 02:13:45 [notice] 190#190: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/11/28 02:13:45 [notice] 190#190: start worker processes
2023/11/28 02:13:45 [notice] 190#190: start worker process 195
2023/11/28 02:13:45 [notice] 190#190: start worker process 196
2023/11/28 02:13:45 [notice] 190#190: start worker process 197
2023/11/28 02:13:45 [notice] 190#190: start worker process 198
2023-11-28 02:13:45,957 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2023-11-28 02:13:45,958 - startup - [CRITICAL] - Running Invertor (192.168.1.222) read loop every 5s
2023-11-28 02:13:45,967 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2023-11-28 02:13:45,978 - startup - [CRITICAL] - Starting Gunicorn on port 6345
[2023-11-28 02:13:46 +0000] [202] [INFO] Starting gunicorn 21.2.0
[2023-11-28 02:13:46 +0000] [202] [INFO] Listening at: http://0.0.0.0:6345 (202)
[2023-11-28 02:13:46 +0000] [202] [INFO] Using worker: sync
[2023-11-28 02:13:46 +0000] [203] [INFO] Booting worker with pid: 203
[2023-11-28 02:13:47 +0000] [204] [INFO] Booting worker with pid: 204
[2023-11-28 02:13:47 +0000] [205] [INFO] Booting worker with pid: 205
2023-11-28 02:13:49,777 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2023-11-28 02:13:56,521 - Inv1 - read        -  [CRITICAL] - First time running so saving AC Charge status
2023-11-28 02:13:56,534 - Inv1 - read        -  [CRITICAL] - Publishing Home Assistant Discovery messages
2023-11-28 05:22:34,931 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2023-11-28 14:41:28,396 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping

SheriffOfNotts, 2023-11-28:
It had worked flawlessly to the point of upgrading. I tried everything, uninstalled MQTT, GivTCP etc and reinstalled. Nothing worked. Restored to earlier version and it works perfectly. Out of interested do you also have an AIO?

Idark02, 2023-11-28:
I am new to this, so I am still learning and picking things up but don’t feel that there is much else I can try. I don’t have an AIO, I have a gen2 Hybrid but still seem to be getting the same issue as you.
I have been trying to look into it and it seems like others have also gotten around the issue by restoring to an earlier version or from previous data. I am not sure how I would do this or if it is even possible given mine is a new install and I have no previous versions or data.
I have noticed that this issue is now marked as closed, I still had this page open from last night (it hadn’t updated) and it was showing as open when I sent my first comment. I have made the same comment in another issue thread as I wasn’t sure this would still be live, is there any way of re-opening this issue?

Gazbert67, 2023-11-29:
I am getting the same;
[ERROR] - Gathering inverter details for 192.168.1.XXX failed
[ERROR] - Unable to interrogate inverter to get base details

My hybrid invertor has a static ip address set and hardwired network connection.

I get some data showing, but its not up to date.

britkat1980, 2023-12-11:
Can someone turn on debugging and post the logs, there's not much to go on here. Its odd that GivTCP sees the inverter but can't get data out of it. Has there been a recent firmware update? If so perhaps toggling the "new firmware" config setting might help?

britkat1980, 2023-12-11:
Also, if anyone can test a dev build with debug on it might help diagnose the problem. Let me know if I get any takers

guinius, 2023-12-12:
> Also, if anyone can test a dev build with debug on it might help diagnose the problem. Let me know if I get any takers

@britkat1980 I have a Gen3 running latest GE SW, happy to try out a dev build with debug.

guinius, 2023-12-15:
> > Also, if anyone can test a dev build with debug on it might help diagnose the problem. Let me know if I get any takers
> 
> @britkat1980 I have a Gen3 running latest GE SW, happy to try out a dev build with debug.

Had a play with Debug on and noticed with my Gen3, the debug response was different when selected AIO. So ended up getting it to work with AIO on and battery qty as 2. This resolved it for me.

britkat1980, 2023-12-15:
Thanks that's useful to know. Gen 3 obviously use the same (new) Modbus implementation, that the AIO use. I'll modify the note on that setting to make it more obvious

General-Meow, 2023-12-18:
I'm having the same issues with getting the base details. I'm running 2 Hybrid inverters, giv_tcp is running on a ubuntu x84 linux box, within docker compose. 

Here's the logs I get if that helps

```~/givenergy$ docker compose up
[+] Running 2/0
 ✔ Container givenergy-GivTCP-1                                      Created                                                                                                                                                                                                                              0.1s
 ! GivTCP Published ports are discarded when using host network mode                                                                                                                                                                                                                                      0.0s
Attaching to givenergy-GivTCP-1
givenergy-GivTCP-1  | 2023-12-18 14:41:57,871 - startup - [CRITICAL] - SUPERVISOR TOKEN does not exist
givenergy-GivTCP-1  | 2023-12-18 14:41:59,873 - startup - [DEBUG] - Networks available for scanning are: {0: '192.168.68.74'}
givenergy-GivTCP-1  | 2023-12-18 14:41:59,874 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
givenergy-GivTCP-1  | 2023-12-18 14:41:59,874 - startup - [DEBUG] - EVC- Scanning network (1):192.168.68.74
givenergy-GivTCP-1  | 2023-12-18 14:42:01,182 - startup - [DEBUG] - EVC- Scanning network (2):192.168.68.74
givenergy-GivTCP-1  | 2023-12-18 14:42:02,492 - startup - [DEBUG] - INV- Scanning network (1):192.168.68.74
givenergy-GivTCP-1  | 2023-12-18 14:42:03,799 - startup - [DEBUG] - 2 Inverters found on 192.168.68.74 - {1: '192.168.68.60', 2: '192.168.68.61'}
givenergy-GivTCP-1  | 2023-12-18 14:42:03,799 - startup - [DEBUG] - Getting inverter stats for: 192.168.68.60
givenergy-GivTCP-1  | 2023-12-18 14:42:04,030 - startup - [ERROR] - Gathering inverter details for 192.168.68.60 failed.
givenergy-GivTCP-1  | 2023-12-18 14:42:04,030 - startup - [ERROR] - Unable to interrogate inverter to get base details
givenergy-GivTCP-1  | 2023-12-18 14:42:04,253 - startup - [ERROR] - Gathering inverter details for 192.168.68.60 failed.
givenergy-GivTCP-1  | 2023-12-18 14:42:04,253 - startup - [ERROR] - Unable to interrogate inverter to get base details
givenergy-GivTCP-1  | 2023-12-18 14:42:04,253 - startup - [DEBUG] - Getting inverter stats for: 192.168.68.61
givenergy-GivTCP-1  | 2023-12-18 14:42:04,484 - startup - [ERROR] - Gathering inverter details for 192.168.68.61 failed.
givenergy-GivTCP-1  | 2023-12-18 14:42:04,484 - startup - [ERROR] - Unable to interrogate inverter to get base details
givenergy-GivTCP-1  | 2023-12-18 14:42:04,713 - startup - [ERROR] - Gathering inverter details for 192.168.68.61 failed.
givenergy-GivTCP-1  | 2023-12-18 14:42:04,713 - startup - [ERROR] - Unable to interrogate inverter to get base details
givenergy-GivTCP-1  | 2023-12-18 14:42:04,713 - startup - [INFO] - Searching for Inverters again
givenergy-GivTCP-1  | 2023-12-18 14:42:04,713 - startup - [DEBUG] - Networks available for scanning are: {0: '192.168.68.74'}
givenergy-GivTCP-1  | 2023-12-18 14:42:04,713 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
givenergy-GivTCP-1  | 2023-12-18 14:42:04,714 - startup - [DEBUG] - EVC- Scanning network (1):192.168.68.74
givenergy-GivTCP-1  | 2023-12-18 14:42:06,022 - startup - [DEBUG] - EVC- Scanning network (2):192.168.68.74
givenergy-GivTCP-1  | 2023-12-18 14:42:07,332 - startup - [DEBUG] - INV- Scanning network (1):192.168.68.74
givenergy-GivTCP-1  | 2023-12-18 14:42:08,639 - startup - [DEBUG] - 2 Inverters found on 192.168.68.74 - {1: '192.168.68.60', 2: '192.168.68.61'}
givenergy-GivTCP-1  | 2023-12-18 14:42:08,639 - startup - [DEBUG] - Getting inverter stats for: 192.168.68.60
givenergy-GivTCP-1  | 2023-12-18 14:42:08,842 - startup - [ERROR] - Gathering inverter details for 192.168.68.60 failed.
givenergy-GivTCP-1  | 2023-12-18 14:42:08,842 - startup - [ERROR] - Unable to interrogate inverter to get base details
givenergy-GivTCP-1  | 2023-12-18 14:42:09,051 - startup - [ERROR] - Gathering inverter details for 192.168.68.60 failed.
givenergy-GivTCP-1  | 2023-12-18 14:42:09,051 - startup - [ERROR] - Unable to interrogate inverter to get base details
givenergy-GivTCP-1  | 2023-12-18 14:42:09,052 - startup - [DEBUG] - Getting inverter stats for: 192.168.68.61
givenergy-GivTCP-1  | 2023-12-18 14:42:09,285 - startup - [ERROR] - Gathering inverter details for 192.168.68.61 failed.
givenergy-GivTCP-1  | 2023-12-18 14:42:09,285 - startup - [ERROR] - Unable to interrogate inverter to get base details
givenergy-GivTCP-1  | 2023-12-18 14:42:09,527 - startup - [ERROR] - Gathering inverter details for 192.168.68.61 failed.
givenergy-GivTCP-1  | 2023-12-18 14:42:09,527 - startup - [ERROR] - Unable to interrogate inverter to get base details
givenergy-GivTCP-1  | 2023-12-18 14:42:09,528 - startup - [INFO] - Searching for Inverters again
givenergy-GivTCP-1  | 2023-12-18 14:42:09,528 - startup - [DEBUG] - Networks available for scanning are: {0: '192.168.68.74'}
givenergy-GivTCP-1  | 2023-12-18 14:42:09,528 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
givenergy-GivTCP-1  | 2023-12-18 14:42:09,528 - startup - [DEBUG] - EVC- Scanning network (1):192.168.68.74
givenergy-GivTCP-1  | 2023-12-18 14:42:10,837 - startup - [DEBUG] - EVC- Scanning network (2):192.168.68.74
[... cut ...]
givenergy-GivTCP-1  | [2023-12-18 14:42:14 +0000] [197] [INFO] Listening at: http://0.0.0.0:6345 (197)
givenergy-GivTCP-1  | [2023-12-18 14:42:14 +0000] [197] [INFO] Using worker: sync
givenergy-GivTCP-1  | [2023-12-18 14:42:14 +0000] [209] [INFO] Booting worker with pid: 209
givenergy-GivTCP-1  | [2023-12-18 14:42:14 +0000] [210] [INFO] Booting worker with pid: 210
givenergy-GivTCP-1  | [2023-12-18 14:42:14 +0000] [211] [INFO] Booting worker with pid: 211
givenergy-GivTCP-1  |  INFO  Accepting connections at http://localhost:4000```

General-Meow, 2023-12-18:
> I'm having the same issues with getting the base details. I'm running 2 Hybrid inverters, giv_tcp is running on a ubuntu x84 linux box, within docker compose.
> ...

Ok So I've got it working now, works absolutely fine. One thing I forgot to mention was that I was using the latest docker image build from docker hub rather than clone the repo and using the docker compose from there. 

For those that are having issues with it not being able to get the base details from the inverter, clone the repo, make changes to the docker-compose.yml file to contain the correct details for your setup (I just changed the IP addresses and web dashboard settings) and run `docker-compose up`

closed 2024-10-01 by britkat1980