britkat1980/giv_tcp #157: New install - docker container. Can't connect to inverter

mjburton, 2023-12-12:
Hi, hoping someone can help me. I've created a new docker stack and the container is up and running, however it won't connect to the inverter. I've connected to the docker container and confirmed it can ping the Giv IP address so this doesn't look to be a comms issue. I've tried different versions of the image in case a more recent one is broken, but to no avail.  Am I missing something obvious?  Thank you!

Config:

`version: "3.9"
services:
  GivTCP:
    image: britkat/giv_tcp-ma:2.4.2
    ports:
      - "1883:1883"                   # MQTT
      - "6345:6345"                   # REST (add increments for each invertor)
      - "3000:3000"                   # This should match the WEB_DASH_PORT ENV below
      - "6379:6379"                   # Redis for job queueing
      - "9181:9181"                   # RQ Dashboard to view job scheduling
    environment:
      - INVERTOR_IP_1=192.168.11.7   # Set this to the IP address of your first Invertor on your local network
      - NUMINVERTORS=1                # Number of inverters
      - NUMBATTERIES_1=1              # Number of battery modules installed and connected to the above invertor
      - MQTT_OUTPUT=True              # "True" if you want to publish your data to MQTT, "False" otherwise
      - MQTT_ADDRESS=127.0.0.1    # IP address of an existing MQTT broker, or leave as "127.0.0.1" to use the internal broker
      - MQTT_TOPIC=GivEnergy/Data     # Root topic to publish data to. If left blank it will default to GivEnergy/<invertor_serial_number>
      - MQTT_PORT=1883                # Port of your existing broker, leave as "1883" for internal broker
      - LOG_LEVEL=Error               # Level of logs to be reported: "Error", "Info" or "Debug"
      - DEBUG_FILE_LOCATION=          # Location of log file stored inside the container, defalt location is /app/GivTCP
      - PRINT_RAW=True                # If True this will publish all inverotr data unprocessed as well as standard data
      - SELF_RUN=False                # If True the container will self-run and connect and publish data. If "False" the you will need to trigger externally via REST
      - SELF_RUN_LOOP_TIMER=10        # Wait time between every read command to the invertor
      - INFLUX_OUTPUT=False           # "True" if you want to publish your data to InfluxDB, "False" otherwise
      - INFLUX_URL=                   # URL of an external Influx instance
      - INFLUX_TOKEN=                 # Access Token for your Influx instance
      - INFLUX_BUCKET=                # Data Bucket of your Influx instance you want data sent to
      - JSON_OUTPUT=True              # Outputs the data in JSON format
      - INFLUX_ORG=                   # Influx instance Organisation
      - HA_AUTO_D=False               # If True (and if MQTT_OUTPUT is True) this will publish Home Assistant Auto Discovery messages to the broker
      - HADEVICEPREFIX=GivTCP         # Defines the preficfor all entities in Home Assistant
      - PYTHONPATH=/app
      - DAYRATESTART=05:30            # Time in HH:MM when your day time tariff kicks in
      - NIGHTRATESTART=23:30          # Time in HH:MM when your night time tariff kicks in
      - DAYRATE=0.30                # Price in £/$ for your daytime electricity per kWh
      - NIGHTRATE=0.075              # Price in £/$ for your night time electricity per kWh
      - EXPORTRATE=0.075              # Price in £/$ for your export electricity per kWh
      - TZ=Europe/London              # Set to your Timezone
      - WEB_DASH=True                 # Enable the wed dashboard
      - WEB_DASH_PORT=3000            # Port to serve the web dashboard on
      - SMARTTARGET=False             # Enables the PALM capability, requires the three ENV below to be set
      - GEAPI=eyJ0eXAiOiJKV1QiLCJhbGciOiJSUzI1NiJ9.eyJhdWQiOiI5NTc3MDIxOS1jYWE2LTRmOTctOTE3Ni0zNDBlZGMzZDQxNTgiLCJqdGkiOiIyOTk0YzJmZWI4NzVkY2U1ZDViYzlmMDJlY2ZlMmVjZGQ1ZmI3NmU5N2Q1YTI2OTE4MWMyY2ZlNDgxOTg4MTljNWMzY2RhZjE2ZDhkY2NiOSIsImlhdCI6MTcwMjMwMDk2Mi43Mzg5NjgsIm5iZiI6MTcwMjMwMDk2Mi43Mzg5NzIsImV4cCI6MzI1MDM2ODAwMDAuMDE0ODkzLCJzdWIiOiIyMzgzIiwic2NvcGVzIjpbImFwaSJdfQ.kQ90Igamrau71GZgiX0NY2W0M9UhF0xzB_cGHVy-ZBzrOZ9OznSMjEqI3LhcOA9N6SRdTveLf66Qc6Dszq6nmQ                        # API Key for the GivEnergy Cloud
      - SOLCASTAPI=                   # API key for Solcast
      - SOLCASTSITEID=                # SiteID for Solcast site (single array)
      - DATASMOOTHER=High             # Set the data smoothing to most agressive setting (High, medium, low)      
    restart: always
    privileged: true`

Error log (it doesn't seem to be picking up the inverter IP of 192.168.11.7:

> 2023-12-12 19:18:55,628 - startup - [CRITICAL] - SUPERVISOR TOKEN does not exist
2023-12-12 19:18:57,629 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2023-12-12 19:19:02,862 - startup - [CRITICAL] - No inverters found...
2023-12-12 19:19:02,862 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2023-12-12 19:19:08,092 - startup - [CRITICAL] - No inverters found...
2023-12-12 19:19:08,092 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2023-12-12 19:19:13,320 - startup - [CRITICAL] - No inverters found...
2023-12-12 19:19:13,342 - startup - [CRITICAL] - Running Redis
2023-12-12 19:19:13,441 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2023/12/12 19:19:13 [notice] 248#248: using the "epoll" event method
2023/12/12 19:19:13 [notice] 248#248: nginx/1.20.2
2023/12/12 19:19:13 [notice] 248#248: OS: Linux 4.4.302+
2023/12/12 19:19:13 [notice] 248#248: getrlimit(RLIMIT_NOFILE): 1048576:1048576
2023/12/12 19:19:13 [notice] 248#248: start worker processes
2023/12/12 19:19:13 [notice] 248#248: start worker process 252
2023/12/12 19:19:13 [notice] 248#248: start worker process 253
[... 96 lines cut ...]
 HTTP  12/12/2023 7:19:55 PM 172.26.0.1 Returned 304 in 2 ms
 HTTP  12/12/2023 7:19:55 PM 172.26.0.1 GET /app.json
 HTTP  12/12/2023 7:19:55 PM 172.26.0.1 Returned 304 in 1 ms
2023-12-12 19:20:56,532 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
1702408856: New connection from 127.0.0.1:54700 on port 1883.
1702408856: New client connected from 127.0.0.1:54700 as GivEnergy_GivTCP_1_Control (p2, c1, k60).
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
1702408856: Client GivEnergy_GivTCP_1_Control closed its connection.

mjburton, 2023-12-13:
This is trying to communiate with a G1 Hybrid 3.6 with 1 x 8.2 battery

gcoan, 2023-12-13:
I had the "cannot find serial number" error on my gen 1 hybrid in issue #150

It was reported by other people previously and fixed in 2.4.2, but I had it happen in v 2.4.3.  I resolved the issue by downgrading to v 2.4.1

mjburton, 2023-12-13:
I've tried that image too, same result.
I checked /app/GivTCP_1/settings.json and it hasn't picked up the inverter IP from the ENV variables. 

invertorIP=""

It's there in settings.py though.

mjburton, 2023-12-13:
I changed - SELF_RUN=False  to True and this seems to have got it fired up.
Using instructions from https://wiki.terravolt.co.uk/docs/givtcp-installation-version-2-0/

screig, 2023-12-27:
I had a very similar problem. 

I had to additionally set INVERTOR_IP_1 as well as INVERTOR_IP

closed 2024-10-01 by britkat1980