britkat1980/giv_tcp #159: New user on 2.4.3 - Not working - No serial_number found in MQTT queue

defeatedbythecat, 2023-12-18:
Doesn't seem like I can get this to work at all. I have a 9.5kwh battery with a separate Gen1 inverter. 

Logs as follows

`2023-12-18 09:08:38,625 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2023-12-18 09:08:38,628 - startup - [INFO] - Supervisor Timezone: Europe/London
2023-12-18 09:08:40,633 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2023-12-18 09:08:44,176 - startup - [CRITICAL] - Inverter CE2237G100 which is a Gen 1 - AC has been found at: 192.168.1.118
2023-12-18 09:08:44,177 - startup - [INFO] - Searching for Inverters again
2023-12-18 09:08:44,177 - startup - [CRITICAL] - Running Redis
2023-12-18 09:08:44,178 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2023/12/18 09:08:44 [notice] 69#69: using the "epoll" event method
2023/12/18 09:08:44 [notice] 69#69: nginx/1.20.2
2023/12/18 09:08:44 [notice] 69#69: OS: Linux 6.1.63-haos
2023/12/18 09:08:44 [notice] 69#69: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/12/18 09:08:44 [notice] 69#69: start worker processes
2023/12/18 09:08:44 [notice] 69#69: start worker process 73
2023/12/18 09:08:44 [notice] 69#69: start worker process 74
2023-12-18 09:08:45,249 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2023-12-18 09:08:45,250 - startup - [CRITICAL] - Running Invertor (192.168.1.118) read loop every 15s
2023-12-18 09:08:45,250 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2023-12-18 09:08:45,258 - startup - [CRITICAL] - Starting Gunicorn on port 6345
[2023-12-18 09:08:46 +0000] [78] [INFO] Starting gunicorn 21.2.0
[2023-12-18 09:08:46 +0000] [78] [INFO] Listening at: http://0.0.0.0:6345 (78)
[2023-12-18 09:08:46 +0000] [78] [INFO] Using worker: sync
[2023-12-18 09:08:47 +0000] [79] [INFO] Booting worker with pid: 79
[2023-12-18 09:08:47 +0000] [80] [INFO] Booting worker with pid: 80
[2023-12-18 09:08:47 +0000] [81] [INFO] Booting worker with pid: 81
2023-12-18 09:08:47,488 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2023-12-18 09:09:54,243 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2023-12-18 09:10:27,589 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
2023-12-18 09:14:10,833 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...`

hamiltongross, 2023-12-23:
I have exactly the same issue. Same setup. Did you manage to solve it?

gcoan, 2023-12-23:
> I have exactly the same issue. Same setup. Did you manage to solve it?

I had the same issue on 2.4.3, logged as #143

Only way I solved it was to downgrade to 2.4.1 which is working fine for me.
Hoping for a @brikat fix though as I'd like to be back on the latest version

hamiltongross, 2023-12-24:
Hi Geoffrey,

Thanks for the feedback. Just to let you know that I managed to solve this simply by setting the inverter to old firmware in the GivTCP configuration. No need to downgrade the version.

Thanks,

Hamilton

gcoan, 2023-12-24:
Good luck Hamilton.  I have mine set to old firmware as well and it was working fine on 2.4.3, then one inverter stopped working with this serial number error.  Tried everything to fix it to no avail, only downgrading back to 2.4.1 worked.  The second inverter worked fine throughout.

I'm trying 2.4.3 again to see if I can recreate the issue I had.

All very strange.

Geoffrey

gcoan, 2023-12-24:
My test proved that 2.4.3 still has this problem.

Upgraded from 2.4.1 to 2.4.3 and initial poll showed both inverters ok
[image]

But that was the last successful poll of my inverter 2.  Thereafter GivTCP never obtained any data (the battery errors on inverter 1 can be ignored, there's no battery connected)
[image]

Inverter is set as old firmware 
[image]

But mqtt for this inverter isn't populated and no inverter time updates for inverter 2 in the history log - inverter 1 still working fine throughout 
[image]
[image]

Both inverters are on the same firmware 
[image]

Restored the 2.4.1 backup back and it's working perfectly again 
[image]

[image]

pewtertankard, 2024-01-29:
I've been a user since early December and, initially, GivTCP 2.4.3 worked OK (setting the old firmware flag).

However, it's stopped working, with exactly this error, and I have no backup of GivTCP 2.4.1 to go back to to bypass this issue. I'm using a Gen 3 inverter on latest firmware. I've tried restarting the inverter, the GivTVP add-on and rebooting the Pi on which I'm running HAOS (GivTCP add on).

Is there a fix for this being prepared please? Otherwise, some advice would be very welcome.

gcoan, 2024-01-29:
My bug on this issue is #153

Theres a load of detail and logs on that issue. I upgraded back to 2.4.3 and it worked find for a while then suddenly stopped working again with this error 

I managed to resolve it by doing a reset to defaults in the portal 

Still on 2.4,3 - have a look at my issue for more details

pewtertankard, 2024-01-30:
Many thanks. Yes that seems to have fixed it

I may also be dreaming it (it _is_ early after all) but the GivEnergy app now seems much more responsive than it has in weeks. I wonder if I can use this as an early warning sign if it were to happen again.

Many thanks again

closed 2024-10-01 by britkat1980