britkat1980/giv_tcp #161: Request: AIO - Add battery temperature and voltage

blastehh, 2023-12-22:
I had a look at the REST output and under raw.inverter there is `temp_battery` and `v_battery` available, can these be added so HA can see them?
Would it also be possible to send the inverter temp, battery temp, and battery voltage to influxdb?

blastehh, 2023-12-22:
Attaching my output for reference
[aio-rest.json](https://github.com/britkat1980/giv_tcp/files/13752473/aio-rest.json)

7Van737, 2024-01-12:
And following the same logic here is the code for the battery voltage which appears to work.  

sensor:
  - name: "AIO BMS Temp"
    unique_id: aio-bms-temp
    state_topic: "Givenergy/CHXXXXXX/raw/invertor/temp_battery"
    device_class: Temperature
    unit_of_measurement: "°C"
    state_class: measurement
    icon: "mdi:thermometer"

  - name: "AIO Voltage"
    unique_id: aio-voltage
    state_topic: "Givenergy/CHXXXXXX/raw/invertor/v_battery"
    device_class: Voltage
    unit_of_measurement: "V"
    state_class: measurement

PianSom, 2024-01-16:
Thanks for the hints. For me to get this to work I have to have the capital E in GivEnergy

PianSom, 2024-01-16:
Actually - this is a nice solution: https://github.com/Codegnosis/givtcp-battery-card

PianSom, 2024-01-16:
Deleted the comment, because it doesn't have either voltage or temperature! Oops

blastehh, 2024-01-16:
> Here is the code I found on a GE facebook thread that creates the Battery Temperature as an entity. Works for me.
> 
> sensor:
> 
>     * name: "AIO BMS Temp"
>       unique_id: aio-bms-temp
>       state_topic: "Givenergy/CHXXXXXXXX/raw/invertor/temp_battery"
>       device_class: Temperature
>       unit_of_measurement: "°C"
>       state_class: measurement
>       icon: "mdi:thermometer"
> 
> 
> Created a new file called mqtt.yaml in the config folder with 'mqtt: !include mqtt.yaml' added to the configeration.yaml

this is perfect, thanks.

for those who want to copy:
```
sensor:
  - name: "AIO BMS Temp"
    unique_id: aio-bms-temp
    state_topic: "Givenergy/CHXXXXXX/raw/invertor/temp_battery"
    device_class: Temperature
    unit_of_measurement: "°C"
    state_class: measurement
    icon: "mdi:thermometer"

  - name: "AIO BMS Temp"
    unique_id: aio-bms-temp
    state_topic: "Givenergy/CHXXXXXX/raw/invertor/temp_battery"
    device_class: Temperature
    unit_of_measurement: "°C"
    state_class: measurement
    icon: "mdi:thermometer"

  - name: "AIO Voltage"
    unique_id: aio-voltage
    state_topic: "Givenergy/CHXXXXXX/raw/invertor/v_battery"
    device_class: Voltage
    unit_of_measurement: "V"
    state_class: measurement

```

7Van737, 2024-01-16:
> Thanks for the hints. For me to get this to work I have to have the capital E in GivEnergy

I think your capital E comes from how you will have typed "GivEnergy" as the 'MQTT topic ' in the GivTCP configuration.

blastehh, 2024-01-16:
yep, it refers to this 
[image]
 Mine is set to GEAIO

PhillWalker, 2024-04-17:
this would be great to be incorporated, very useful data to have

britkat1980, 2024-05-14:
New (currently in beta) version includes much more AIO battery data inc BCU data and voltage/temps

closed 2024-05-14 by britkat1980

PianSom, 2024-05-14:
Hey @britkat1980 - thanks for the continuing work. 

I don't suppose you are feeling charitable enough to build a docker container version for those of us who use HA Container and can't use Add-ons? 

Pretty please?

britkat1980, 2024-05-14:
It’s already there, checkout the beta branch on dockerhub

PianSom, 2024-05-14:
Blow me - so it is.

I did check before posting - but looked for a beta tag in the main gif_tcp-ma, rather than looking for a beta branch.

Apologies!