britkat1980/giv_tcp #165: 'GiV_Settings' has no attribute 'serial_number'

Mattntate, 2024-01-12:
I'm getting the same problem as [JRascagneres](https://github.com/JRascagneres) #145 from November last year. There was no fix published that I could see.

I'm an absolute beginner - following the setup from 'Speak to the geek' on youtube. I hope you can help!!

In the log it is seeing both inverters, but then gives the error:

2024-01-12 18:41:26,056 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
2024-01-12 18:41:26,796 - Inv2 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_2/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_2/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'

closed 2024-01-12 by Mattntate

gcoan, 2024-01-12:
@Mattntate how did you fix it, or did you close this issue by accident?

I'm still getting this error on GivTCP v2.4.3 and have had to downgrade to v2.4.1 which is working fine still, but an old version 

#153 is my own issue on this bug

Mattntate, 2024-01-12:
@gcoan In the config of GivTCP (2.4.3) I originally had it set to off in "is it on old firmware" because my firmware is up to date.  I saw in another thread with the same problem  (#159) that someone had set it to on.  That's what I did and it worked! Let me know how you get on!