britkat1980/giv_tcp #171: error during startup

Heywoodstephen, 2024-03-09:
I recently started adding some items needed to install predbat on my home assistant running on a raspberry pi, but after a system reboot I stopped getting any data from giv_tcp bellow is the log from the most recent startup.  can anyone give me a few pointers as to how to fix this, I am new to the whole raspberry pi / home assistant scene.

2024-03-09 09:42:20,983 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-03-09 09:42:20,997 - startup - [INFO] - Supervisor Timezone: Europe/London
2024-03-09 09:42:23,013 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-03-09 09:42:27,155 - startup - [CRITICAL] - Inverter SD2145G142 which is a Gen 1 - Hybrid has been found at: 192.168.1.41
2024-03-09 09:42:29,973 - startup - [CRITICAL] - Inverter SD2145G142 which is a Gen 1 - Hybrid has been found at: 192.168.1.41
2024-03-09 09:42:29,975 - startup - [INFO] - Searching for Inverters again
2024-03-09 09:42:29,990 - startup - [CRITICAL] - Running Redis
2024-03-09 09:42:30,011 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024/03/09 09:42:30 [notice] 109#109: using the "epoll" event method
2024/03/09 09:42:30 [notice] 109#109: nginx/1.20.2
2024/03/09 09:42:30 [notice] 109#109: OS: Linux 6.1.73-haos-raspi
2024/03/09 09:42:30 [notice] 109#109: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/03/09 09:42:30 [notice] 109#109: start worker processes
2024/03/09 09:42:30 [notice] 109#109: start worker process 114
2024/03/09 09:42:30 [notice] 109#109: start worker process 115
2024/03/09 09:42:30 [notice] 109#109: start worker process 116
2024/03/09 09:42:30 [notice] 109#109: start worker process 117
2024-03-09 09:42:30,355 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-03-09 09:42:30,356 - startup - [CRITICAL] - Running Invertor (192.168.1.41) read loop every 2s
2024-03-09 09:42:30,363 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-03-09 09:42:30,371 - startup - [CRITICAL] - Starting Gunicorn on port 6345
2024-03-09 09:42:30,383 - startup - [CRITICAL] - Serving Web Dashboard from port 3005
2024-03-09 09:42:30,466 - startup - [CRITICAL] - Setting daily charge target forecast job to run at: 23:50
[2024-03-09 09:42:31 +0000] [121] [INFO] Starting gunicorn 21.2.0
[2024-03-09 09:42:31 +0000] [121] [INFO] Listening at: http://0.0.0.0:6345 (121)
[2024-03-09 09:42:31 +0000] [121] [INFO] Using worker: sync
[2024-03-09 09:42:31 +0000] [128] [INFO] Booting worker with pid: 128
[2024-03-09 09:42:31 +0000] [130] [INFO] Booting worker with pid: 130
[2024-03-09 09:42:31 +0000] [131] [INFO] Booting worker with pid: 131
2024-03-09 09:42:32,813 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
 INFO  Accepting connections at http://localhost:3005
2024-03-09 09:42:37,303 - Inv1 - read        -  [CRITICAL] - First time running so saving AC Charge status
2024-03-09 09:42:37,309 - Inv1 - read        -  [CRITICAL] - Publishing Home Assistant Discovery messages
Traceback (most recent call last):
  File "/app/GivTCP_1/read.py", line 1188, in <module>
    globals()[sys.argv[1]]()
  File "/app/GivTCP_1/read.py", line 785, in self_run2
    runAll("True")
  File "/app/GivTCP_1/read.py", line 750, in runAll
    multi_output = pubFromPickle()
  File "/app/GivTCP_1/read.py", line 771, in pubFromPickle
    publishOutput(multi_output, SN)
  File "/app/GivTCP_1/read.py", line 814, in publishOutput
    from HA_Discovery import HAMQTT
ValueError: source code string cannot contain null bytes

gcoan, 2024-03-09:
I'd suggest you reduce the run loop value down considerably from the 2 second frequency you have it set to.  This will tie the inverter up in responding to GivTCP enquiries all the time and means you'll find it difficult to control or view it in the GivEnergy app/portal.

I have mine set to 20 second run loop frequency but a bit more or a bit less is also fine.

I'm not sure if this will fix your issue but its worth changing first

Heywoodstephen, 2024-03-09:
changed it to 20s and its still not working

gcoan, 2024-03-09:
OK great, what type of of inverter and battery do you have?

If its a newer inverter (eg Gen 3 or AIO), try turning 'use old firmware' to off

Heywoodstephen, 2024-03-09:
Is a gen1 inverter and battery, it was all working until a recent ha core update.

gcoan, 2024-03-09:
Ok good to know it was working fine before.  If it's a gen 1 you shouldnt need the old firmware switch turned off.

I've had a similar issue with newer versions of GivTCP, mine being the "no serial number" error.
I found I could fix that by doing a reset to defaults on the inverter in the portal which restored whatever GivTCP didn't like.

Suggest you try that.

Otherwise can you restore back to an earlier backup?

Heywoodstephen, 2024-03-09:
I have resolved this issue by uninstalling both Matt and givtcp, and reinstalling everything.

gcoan, 2024-03-09:
Well done.  Glad you got it resolved