britkat1980/giv_tcp #172: Which service to set a timeslot?

yavinmaster, 2024-03-14:
How do you set a timeslot to a time in a HomeAssistant automation?  Is it a service call, if so which one?

Sorry for the fundamental question but for some reason I just can't find a solution.

gcoan, 2024-03-14:
> How do you set a timeslot to a time in a HomeAssistant automation?  Is it a service call, if so which one?
> 
> 
> 
> Sorry for the fundamental question but for some reason I just can't find a solution.

If I have understood you correctly, you are trying to set a time such as charge/discharge start/end time.

This is how I do it in my scripts:

```
  - service: select.select_option
    data:
      option: "00:00:00"
    target:
      entity_id:
        - select.givtcp_SERIALNO_discharge_start_time_slot_1
  - service: select.select_option
    data:
      option: "23:59:00"
    target:
      entity_id: select.givtcp_SERIALNO_discharge_end_time_slot_1
```

yavinmaster, 2024-03-14:
Thanks for the reply.  That is what I'm trying to do.  If I however try this in an automation and select the select.select_option service, then the givtcp timeslots aren't available to select as entities.  I could try it in yaml but why are they not available?

gcoan, 2024-03-14:
I suspect you are almost certainly using the input_select.select service rather than the select.select service.

I made this mistake myself early on in my scripts.

Here's the difference:
[image]

[image]

yavinmaster, 2024-03-14:
Thank you so much.  You were absolutely correct.  I've not tested the automation yet, but based on you information I'm confident I have it now.

closed 2024-03-14 by yavinmaster