britkat1980/giv_tcp #176: type object 'GiV_Settings' has no attribute 'serial_number'

ian-hey, 2024-04-04:
HI Wonder if you can help im new to HA and im running 2.4.3

its the correct invertor IP and have only have the 1 so set all the other batts to zero like other posts have said

2024-04-04 15:45:10,310 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-04-04 15:45:10,332 - startup - [INFO] - Supervisor Timezone: Europe/London
2024-04-04 15:45:12,359 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-04-04 15:45:16,543 - startup - [CRITICAL] - Inverter CE2238G042 which is a Gen 1 - AC has been found at: 10.0.0.14
2024-04-04 15:45:16,545 - startup - [INFO] - Searching for Inverters again
2024-04-04 15:45:16,562 - startup - [CRITICAL] - Running Redis
2024-04-04 15:45:16,580 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024/04/04 15:45:16 [notice] 69#69: using the "epoll" event method
2024/04/04 15:45:16 [notice] 69#69: nginx/1.20.2
2024/04/04 15:45:16 [notice] 69#69: OS: Linux 6.1.73-haos-raspi
2024/04/04 15:45:16 [notice] 69#69: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/04/04 15:45:16 [notice] 69#69: start worker processes
2024/04/04 15:45:16 [notice] 69#69: start worker process 74
2024/04/04 15:45:16 [notice] 69#69: start worker process 75
2024/04/04 15:45:16 [notice] 69#69: start worker process 76
2024/04/04 15:45:16 [notice] 69#69: start worker process 77
2024-04-04 15:45:17,018 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-04-04 15:45:17,019 - startup - [CRITICAL] - Running Invertor (10.0.0.14) read loop every 15s
2024-04-04 15:45:17,032 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-04-04 15:45:17,047 - startup - [CRITICAL] - Starting Gunicorn on port 6345
[2024-04-04 15:45:18 +0100] [81] [INFO] Starting gunicorn 21.2.0
[2024-04-04 15:45:18 +0100] [81] [INFO] Listening at: http://0.0.0.0:6345 (81)
[2024-04-04 15:45:18 +0100] [81] [INFO] Using worker: sync
[2024-04-04 15:45:18 +0100] [82] [INFO] Booting worker with pid: 82
[2024-04-04 15:45:18 +0100] [83] [INFO] Booting worker with pid: 83
[2024-04-04 15:45:18 +0100] [84] [INFO] Booting worker with pid: 84
2024-04-04 15:45:22,502 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2024-04-04 15:47:02,649 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
2024-04-04 15:47:46,814 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2024-04-04 15:51:44,043 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2024-04-04 15:55:39,886 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2024-04-04 15:58:52,878 - Inv1 - sync        -  [ERROR   ] - Connection to (10.0.0.14, 8899) failed: timed out
2024-04-04 15:59:41,898 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2024-04-04 16:03:40,937 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force

gcoan, 2024-04-04:
I have occasionally had the 'no serial number' error on my gen 1 hybrid. I found doing a reset to defaults of the inverter in the givenergy portal cured the issue.  Not sure why but GivTCP didn't like the inverter config for some reason

ian-hey, 2024-04-04:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

Oh thanks I will try that

[University of Central Lancashire] <https://www.uclan.ac.uk/>
Please consider the environment before printing.

closed 2024-05-29 by britkat1980