britkat1980/giv_tcp #177: Connection Errors

charlie-chester, 2024-04-08:
I've seen this a little on there Issues but I thought I would start a new thread as they are slightly different to what I've seen. 

I'm using Predbat to look after my Gen 3 5Kw GivEnery Inverter and 9.5 Kwh Battery and I keep getting errors in the log. I think they are generally about losing contact with it or not getting back the answer it expects. out of all the errors though this is the one I'm getting time and time again.  I've posted this with the success ones to show that it's not all the time. 

```
2024-04-08 15:30:33,700 - Inv1 - write       -  [INFO    ] - Setting Charge Enable was a success
2024-04-08 15:35:16,020 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 3600 (50)
2024-04-08 15:35:19,057 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 50 was a success
2024-04-08 16:00:15,778 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 21:30 - 22:00
2024-04-08 16:00:19,912 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 was a success
2024-04-08 16:02:43,409 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2024-04-08 16:02:53,471 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 failed: (<class 'AssertionError'>, AssertionError('Unexpected response from remote end: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 44 bytes (0 received)'), <traceback object at 0xffff9ef986c0>)
2024-04-08 16:12:27,933 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 3600 (50)
2024-04-08 16:12:30,704 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 50 was a success
2024-04-08 16:12:36,993 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 16:00 - 16:30
2024-04-08 16:12:44,560 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 was a success
2024-04-08 16:17:28,687 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 21:30 - 22:00
2024-04-08 16:17:32,860 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 was a success
2024-04-08 16:22:26,832 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 16:00 - 16:30
2024-04-08 16:22:31,127 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 was a success
```

```
2024-04-08 18:20:40,948 - Inv1 - write       -  [INFO    ] - Enabling Discharge was a success
2024-04-08 18:20:47,295 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 63
2024-04-08 18:20:53,525 - Inv1 - write       -  [INFO    ] - Setting shallow charge 63 was a success
2024-04-08 18:20:59,839 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2024-04-08 18:21:02,728 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2024-04-08 18:30:17,717 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Eco
2024-04-08 18:30:21,440 - Inv1 - write       -  [INFO    ] - Setting dynamic mode failed: (<class 'AssertionError'>, AssertionError('Unexpected response from remote end: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 44 bytes (0 received)'), <traceback object at 0xffff9ef7fbc0>)
2024-04-08 18:30:22,982 - Inv1 - write       -  [INFO    ] - Setting shallow charge 63 failed: (<class 'AssertionError'>, AssertionError('Unexpected response from remote end: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response'), <traceback object at 0xffff9ef7fc80>)
2024-04-08 18:30:30,977 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Eco
2024-04-08 18:30:36,823 - Inv1 - write       -  [INFO    ] - Setting dynamic mode was a success
2024-04-08 18:30:39,754 - Inv1 - write       -  [INFO    ] - Setting shallow charge 63 was a success
2024-04-08 18:30:56,305 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 00:00 - 00:00
2024-04-08 18:31:00,697 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-04-08 18:31:07,090 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 3600 (50)
2024-04-08 18:31:10,108 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 50 was a success
2024-04-08 18:31:16,488 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 4
2024-04-08 18:31:22,727 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success
```

Any ideas?

charlie-chester, 2024-04-08:
Just had some more pop in. 
```
2024-04-08 19:45:29,521 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.50.176, 8899) failed: timed out
2024-04-08 19:45:32,027 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.50.176, 8899) failed: timed out
2024-04-08 19:45:34,531 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.50.176, 8899) failed: timed out
2024-04-08 19:45:37,035 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.50.176, 8899) failed: timed out
2024-04-08 19:45:39,539 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.50.176, 8899) failed: timed out
2024-04-08 19:45:59,746 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.50.176, 8899) failed: timed out
2024-04-08 19:46:00,891 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.50.176, 8899) failed: [Errno 113] Host is unreachable
```

gcoan, 2024-04-08:
It does look like comms failures which as you said on the other issue you've done all you can to reduce. 

Just one thought, is 192.168.50.176 the address of your inverter or your home assistant server ?

charlie-chester, 2024-04-08:
> Just one thought, is 192.168.50.176 the address of your inverter or your home assistant server ?

The Inverter

charlie-chester, 2024-04-08:
More bad news!!

```
2024-04-08 18:20:59,839 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2024-04-08 18:21:02,728 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2024-04-08 18:30:17,717 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Eco
2024-04-08 18:30:21,440 - Inv1 - write       -  [INFO    ] - Setting dynamic mode failed: (<class 'AssertionError'>, AssertionError('Unexpected response from remote end: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 44 bytes (0 received)'), <traceback object at 0xffff9ef7fbc0>)
2024-04-08 18:30:22,982 - Inv1 - write       -  [INFO    ] - Setting shallow charge 63 failed: (<class 'AssertionError'>, AssertionError('Unexpected response from remote end: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response'), <traceback object at 0xffff9ef7fc80>)
2024-04-08 18:30:30,977 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Eco
2024-04-08 18:30:36,823 - Inv1 - write       -  [INFO    ] - Setting dynamic mode was a success
2024-04-08 18:30:39,754 - Inv1 - write       -  [INFO    ] - Setting shallow charge 63 was a success
2024-04-08 18:30:56,305 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 00:00 - 00:00
2024-04-08 18:31:00,697 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-04-08 18:31:07,090 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 3600 (50)
2024-04-08 18:31:10,108 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 50 was a success
2024-04-08 18:31:16,488 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 4
2024-04-08 18:31:22,727 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success
2024-04-08 19:45:29,521 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.50.176, 8899) failed: timed out
2024-04-08 19:45:32,027 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.50.176, 8899) failed: timed out
2024-04-08 19:45:34,531 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.50.176, 8899) failed: timed out
2024-04-08 19:45:37,035 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.50.176, 8899) failed: timed out
2024-04-08 19:45:39,539 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.50.176, 8899) failed: timed out
2024-04-08 19:45:59,746 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.50.176, 8899) failed: timed out
2024-04-08 19:46:00,891 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.50.176, 8899) failed: [Errno 113] Host is unreachable
2024-04-08 19:58:11,741 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.50.176, 8899) failed: timed out
2024-04-08 20:20:26,802 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 00:00 - 20:31
2024-04-08 20:20:35,906 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2024-04-08 20:20:42,415 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Timed Export
2024-04-08 20:20:44,068 - Inv1 - write       -  [INFO    ] - Setting export mode was a success
2024-04-08 20:20:46,977 - Inv1 - write       -  [INFO    ] - Enabling Discharge was a success
2024-04-08 20:20:53,551 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 46
2024-04-08 20:21:00,148 - Inv1 - write       -  [INFO    ] - Setting shallow charge 46 was a success
2024-04-08 20:21:06,640 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2024-04-08 20:21:09,521 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2024-04-08 20:27:20,715 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 47
2024-04-08 20:27:23,585 - Inv1 - write       -  [INFO    ] - Setting shallow charge 47 was a success
2024-04-08 20:27:26,647 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.50.176, 8899) failed: timed out
2024-04-08 20:27:31,476 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 47
2024-04-08 20:27:34,439 - Inv1 - write       -  [INFO    ] - Setting shallow charge 47 was a success
2024-04-08 20:30:16,445 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 21:00 - 22:00
2024-04-08 20:30:23,695 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 was a success
2024-04-08 20:30:33,840 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 21:00 - 22:00
2024-04-08 20:30:37,570 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 failed: (<class 'AssertionError'>, AssertionError('Unexpected response from remote end: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 44 bytes (0 received)'), <traceback object at 0xffff9ef98f40>)
2024-04-08 20:30:43,793 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 21:00 - 22:00
2024-04-08 20:30:52,746 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 failed: (<class 'AssertionError'>, AssertionError('Unexpected response from remote end: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response'), <traceback object at 0xffff9efa5080>)
2024-04-08 20:31:01,059 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 21:00 - 22:00
2024-04-08 20:31:05,215 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 was a success
[... cut ...]
2024-04-08 20:33:01,926 - Inv1 - write       -  [INFO    ] - Setting Charge Target 47 was a success
2024-04-08 20:33:08,251 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 4
2024-04-08 20:33:11,215 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success
2024-04-08 20:35:31,889 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 01:00 - 02:00
2024-04-08 20:35:37,395 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 was a success
```

gcoan, 2024-04-08:
> 
> > Just one thought, is 192.168.50.176 the address of your inverter or your home assistant server ?
> 
> 
> 
> The Inverter

Thanks for confirming. If it was HA could have been overloading of the HA server. 

What do you have the output registers and self run timer set to?
[image]

charlie-chester, 2024-04-08:
In the above it was 15 & 2. I've had those all over the place from as low as 5 up to 40. 

I read in an other post about someone making it an AOI as it's a Gen 3 and making batteries 2 but this didn't work either. 

Do you think it maybe as simple as a network fault?

charlie-chester, 2024-04-08:
I'm new to GitHub and it's workings but there is another Branch on here that looks like 2.4.1

[https://github.com/britkat1980/giv_tcp/tree/pr/divenal/135](url)

Is this a way to go back and test?

charlie-chester, 2024-04-08:
The firmware box is for old firmware. I've had that both ways. It works either way and gives the same errors. 

I've got the output raw set too. 

The good thing is it does actually work but just throws up all these errors.

gcoan, 2024-04-09:
> The firmware box is for old firmware. I've had that both ways. It works either way and gives the same errors. 
> 
> I've got the output raw set too. 
> 
> The good thing is it does actually work but just throws up all these errors.

Thanks, trying to eliminate possible sources.

For a Gen 3 I suggest you need to tick the 'new firmware' box as this includes error checking on messages which is supposed to improve reliability and reduce corrupting the inverter registers.  I thought the newer inverters just didn't work unless the box is ticked, so weird that it works for you either way.

I can't think of anything else obvious causing the issue, it points to a comms error between GivTCP and the inverter, either wifi issues or the inverter not responding back to GivTCP in time.  Restarting everything, inverter, home assistant etc is always an option but I'm sure you have already tried that

Any inverter firmware updates available?  Might be worth talking to givenergy and reporting the issues you are getting with inverter timeouts responding to commands. They may have an improved inverter firmware but I'm purely guessing now
Do you get errors from the portal or app?

charlie-chester, 2024-04-09:
Cheers for taking the time I appreciate it. 

You keep saying tick the New Firmware box but it says **Inverter 1 - Is it on old firmware?** so to use new firmware surely this is un ticked?

gcoan, 2024-04-09:
Yes correct, un tick for new firmware

charlie-chester, 2024-04-09:
Brilliant that's what I've done. 

I've just been on the phone to GivEnergy support and spoken to a man who loves a lchallenge. He's looking into it for me and I'll report back here if I get anything from them. 

Just to help though I'll post more error messages here for them to see. 

Cheers Geoff.

charlie-chester, 2024-04-09:
For the info of GivEnery Support this is my main error that keeps showing up.

```
2024-04-09 09:13:15,669 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 failed: (<class 'AssertionError'>, AssertionError('Unexpected response from remote end: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 44 bytes (0 received)'), <traceback object at 0xffff8cb20c40>)
```

charlie-chester, 2024-04-10:
The update from GE was that nothing appeared to be wrong with my setup but they couldn't obviously check my network.

I've actually changed the way Predbat works now I'm controlling from the GE Cloud so all good there. I'm still using GivTCP though as I've a few utility meters running and my Energy Dash is tied into it. 

I'm still getting Timed Out Errors but all my Entities are reading OK as far as I can see. 

I will be trying to hard wire when my brother lends me his 30m cable for testing!!

charlie-chester, 2024-04-14:
I've hard wired the inverter now and have Home Assistant hard wired to the same Router. It's not a node it's the main Router in the system. I'm not getting the same errors but have developed a new one!

The failing to find at the top has happened every time to start and then after aavcfew seconds it finds it. 

it's the
```
2024/04/14 10:21:52 [emerg] 189#189: bind() to 0.0.0.0:8099 failed (98: Address in use)
nginx: [emerg] bind() to 0.0.0.0:8099 failed (98: Address in use)
```
thats new!!

```
2024-04-14 10:21:37,893 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-04-14 10:21:37,917 - startup - [INFO] - Supervisor Timezone: Europe/London
2024-04-14 10:21:39,948 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-04-14 10:21:44,034 - startup - [ERROR] - Gathering inverter details for 192.168.50.179 failed.
2024-04-14 10:21:44,035 - startup - [ERROR] - Unable to interrogate inverter to get base details
2024-04-14 10:21:44,173 - startup - [ERROR] - Gathering inverter details for 192.168.50.179 failed.
2024-04-14 10:21:44,173 - startup - [ERROR] - Unable to interrogate inverter to get base details
2024-04-14 10:21:44,175 - startup - [INFO] - Searching for Inverters again
2024-04-14 10:21:44,175 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-04-14 10:21:48,143 - startup - [ERROR] - Gathering inverter details for 192.168.50.179 failed.
2024-04-14 10:21:48,144 - startup - [ERROR] - Unable to interrogate inverter to get base details
2024-04-14 10:21:48,293 - startup - [ERROR] - Gathering inverter details for 192.168.50.179 failed.
2024-04-14 10:21:48,294 - startup - [ERROR] - Unable to interrogate inverter to get base details
2024-04-14 10:21:48,296 - startup - [INFO] - Searching for Inverters again
2024-04-14 10:21:48,297 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-04-14 10:21:52,264 - startup - [ERROR] - Gathering inverter details for 192.168.50.179 failed.
2024-04-14 10:21:52,265 - startup - [ERROR] - Unable to interrogate inverter to get base details
2024-04-14 10:21:52,414 - startup - [ERROR] - Gathering inverter details for 192.168.50.179 failed.
2024-04-14 10:21:52,415 - startup - [ERROR] - Unable to interrogate inverter to get base details
2024-04-14 10:21:52,428 - startup - [CRITICAL] - Running Redis
2024-04-14 10:21:52,464 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024/04/14 10:21:52 [emerg] 189#189: bind() to 0.0.0.0:8099 failed (98: Address in use)
nginx: [emerg] bind() to 0.0.0.0:8099 failed (98: Address in use)
2024/04/14 10:21:52 [notice] 189#189: try again to bind() after 500ms
2024-04-14 10:21:52,862 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-04-14 10:21:52,863 - startup - [CRITICAL] - Running Invertor (192.168.50.179) read loop every 30s
2024-04-14 10:21:52,878 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-04-14 10:21:52,898 - startup - [CRITICAL] - Starting Gunicorn on port 6345
2024/04/14 10:21:52 [emerg] 189#189: bind() to 0.0.0.0:8099 failed (98: Address in use)
nginx: [emerg] bind() to 0.0.0.0:8099 failed (98: Address in use)
2024/04/14 10:21:52 [notice] 189#189: try again to bind() after 500ms
2024/04/14 10:21:52 [emerg] 189#189: bind() to 0.0.0.0:8099 failed (98: Address in use)
nginx: [emerg] bind() to 0.0.0.0:8099 failed (98: Address in use)
2024/04/14 10:21:52 [notice] 189#189: try again to bind() after 500ms
2024/04/14 10:21:52 [emerg] 189#189: bind() to 0.0.0.0:8099 failed (98: Address in use)
nginx: [emerg] bind() to 0.0.0.0:8099 failed (98: Address in use)
2024/04/14 10:21:52 [notice] 189#189: try again to bind() after 500ms
2024/04/14 10:21:52 [emerg] 189#189: bind() to 0.0.0.0:8099 failed (98: Address in use)
nginx: [emerg] bind() to 0.0.0.0:8099 failed (98: Address in use)
2024/04/14 10:21:52 [notice] 189#189: try again to bind() after 500ms
[2024-04-14 10:21:55 +0100] [197] [INFO] Starting gunicorn 21.2.0
[2024-04-14 10:21:55 +0100] [197] [INFO] Listening at: http://0.0.0.0:6345 (197)
[2024-04-14 10:21:55 +0100] [197] [INFO] Using worker: sync
2024/04/14 10:21:52 [emerg] 189#189: still could not bind()
nginx: [emerg] still could not bind()
[2024-04-14 10:21:55 +0100] [199] [INFO] Booting worker with pid: 199
[2024-04-14 10:21:55 +0100] [198] [INFO] Booting worker with pid: 198
[2024-04-14 10:21:55 +0100] [200] [INFO] Booting worker with pid: 200
2024-04-14 10:22:01,122 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2024-04-14 10:22:06,272 - Inv1 - read        -  [CRITICAL] - First time running so saving AC Charge status
2024-04-14 10:22:06,289 - Inv1 - read        -  [CRITICAL] - Publishing Home Assistant Discovery messages
```

gcoan, 2024-04-14:
0.0.0.0 is in effect a loop back to the same server. The error is saying that something else is running on the same server and port address 8099 so givtcp can't attach to that port.

What other addons are running on the ha server ? It may be because everything is now on the same hub not going through a router any more.
I'm not sure how to fix this. Might need to change nginx configuration maybe?

charlie-chester, 2024-04-14:
I've just taken off the last add on I installed and it went away. 

Brilliant. 

It was an add on called Homeway to access Home Assistant from outside the home network. I've got another way to do that so I'll bin that idea. 

Cheers Geoff.

gcoan, 2024-04-14:
> I've just taken off the last add on I installed and it went away. 
> 
> Brilliant. 
> 
> It was an add on called Homeway to access Home Assistant from outside the home network. I've got another way to do that so I'll bin that idea. 
> 
> Cheers Geoff. 

Great news

I use DuckDNS add-on to give me a remote HTTPS connection to my HA instance, which works great except my mobile provider (O2) blocks the duckdns domain so have to use WireGuard to create a VPN into my home network and then duckDNS works OK.

Its not ideal having to use so many add-on's.  Sounds like I won't try Homeway either then!

charlie-chester, 2024-04-14:
My bother uses it and he's OK!!

I use Instant Guard with my ASUS router  works a treat. Just tried the other method!!

😂😂

charlie-chester, 2024-04-22:
@gcoan since talking you you on here I've installed the GivTCP Dev build and it's taken away all my errors. It's constantly being updated though. 

Have you tried it?

gcoan, 2024-04-22:
> @gcoan since talking you you on here I've installed the GivTCP Dev build and it's taken away all my errors. It's constantly being updated though. 
> 
> Have you tried it?

Hi Charlie 

No I've been just on the main released givtcp. For me it's pretty stable, I do get some occasional connectivity errors and inverter timeouts but nothing major. If possible I'd prefer to be on the core release and not be too bleeding edge - I only just upgraded to home assistant 2024.4 yesterday, 3 weeks after it was released to give time for others to find the bugs !

Good to hear givtcp is being actively developed though but I'll still wait for the public release. 

Cheers Geoffrey

closed 2024-05-29 by britkat1980