britkat1980/giv_tcp #184: EVC time out

abbottcam, 2024-04-29:
Hello, Ive been having problems with GivTCP not connecting to my EVC. this is the message I get in the logs:

2024-04-29 18:33:46,472 - Inv1 - mqtt_client_evc -  [ERROR   ] - No serial_number_evc found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client_evc.py", line 106, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client_evc.py", line 90, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number_evc+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number_evc'. Did you mean: 'serial_number'?
2024-04-29 18:35:08,905 - Inv1 - read        -  [CRITICAL] - Publishing Home Assistant Discovery messages
2024-04-29 18:35:13,698 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.0.251, 502) failed: [Errno 104] Connection reset by peer
2024-04-29 18:40:03,988 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.0.251, 502) failed: [Errno 104] Connection reset by peer
2024-04-29 18:40:08,628 - Inv1 - read        -  [CRITICAL] - Publishing Home Assistant Discovery messages
2024-04-29 18:45:07,293 - Inv1 - read        -  [CRITICAL] - Publishing Home Assistant Discovery messages
2024-04-29 18:45:14,309 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.0.251, 502) failed: [Errno 104] Connection reset by peer
2024-04-29 18:50:04,567 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.0.251, 502) failed: [Errno 104] Connection reset by peer
2024-04-29 18:56:00,920 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.0.251, 502) failed: [Errno 104] Connection reset by peer
2024-04-29 18:56:16,681 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 6000 (50)
2024-04-29 18:56:21,090 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 50 was a success
2024-04-29 18:56:27,512 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to: Disabled
2024-04-29 18:56:29,674 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to Disabled was a success
2024-04-29 18:58:59,102 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.0.251, 502) failed: [Errno 104] Connection reset by peer
2024-04-29 19:03:23,371 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.0.251, 502) failed: [Errno 104] Connection reset by peer
2024-04-29 19:16:22,137 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.0.251, 502) failed: [Errno 104] Connection reset by peer

britkat1980, 2024-05-09:
You need to request latest beta firmware from GE, there are modbus server instability issues with current versions

Grey-Sheep, 2024-08-07:
I have the same issue - it has been there since it was installed.

Givenergy say it isn't working yet!

Does this mean the GivEVC CANNOT be controlled by GIVTCP?  I certainly can't get it working after 3 weeks of trying.

peter1rhodes, 2024-08-08:
It also does not work for me. I have not requested any special beta firmware from GivEnergy though, I could try that...

britkat1980, 2024-08-08:
Hi both, it’s certainly working on numerous installations. However there are reports that it does not work on units connected only via Ethernet. I haven’t confirmed, but are you on WiFi or Ethernet? Perhaps connect WiFi as well and see if it will connect on the WiFi ip address

peter1rhodes, 2024-08-08:
Interesting. My EV charger is only connected by Ethernet, since most of the GivEnergy forums seem to be full of people saying to always hard-wire if at all possible and crowded with others having issues with WiFi! I'll see whether I can try to connect the WiFi as well, but it feels a bit mad! I assume this is a bug on the firmware side?

Grey-Sheep, 2024-08-08:
I was hard wired as it is outside!  Changed to Wi-Fi and all the issues disappeared 👍👍👍
Everything is now appearing as expected in both Home Assistant and the Givenergy app (mobile and web as they are separate and don't share useful details like charging schedules!)

Thanks for the hint.

britkat1980, 2024-08-08:
Glad to hear it. Need to get GE to open up modbus on Ethernet…

peter1rhodes, 2024-08-08:
100%. It's actually wild that they have different behaviours at all in this way, let alone that the recommended, more reliable, more secure, easier-to-implement method is the one that doesn't work... I'll try this when I get home @britkat1980, thank you

peter1rhodes, 2024-08-19:
I have tried switching the charger to WiFi only, but I had the same issues with constant connection failures. Yet to find a fix to get the charger working with local control (the cloud control is very slow and not ideal for e.g. solar charging)

joealden, 2024-09-24:
@britkat1980 do you know if the beta firmware is still required today, and if there has been any movement on GE allowing modbus over ethernet? I don't really want to add an AP just for this if possible!

peter1rhodes, 2024-09-25:
> @britkat1980 do you know if the beta firmware is still required today, and if there has been any movement on GE allowing modbus over ethernet? I don't really want to add an AP just for this if possible!

@joealden for my part, the issues with the EV charger working for a while, but then dropping out and becoming completely unresponsive persist with me, over Ethernet and WiFi equally, with v3.0.0 and beta v3.0.1 just as they did with v2. GivTCP works for a while over ethernet, but eventually (hours or days later) it stops responding, and no amount of GivTCP "waiting five minutes for the modbus to reset" seems to help. The same happens over WiFi. I think I've tried every combination of disconnecting the ethernet altogether, having both connected but only one have internet access (managed via my router), both connected but no internet access, both connected with internet access, and each connection method alone with and without internet access, but I can't get a long-term stable connection.

Grey-Sheep, 2024-09-25:
Likewise, I cannot get a reliable connection or response from the EVC so I've given up with trying to use the modbus/GIVTCP connection.
I also have reliability issues with the cloud based connections.  Several times it has continued to charge after to stop time.
There are also differences in behaviour between the phone app, web interface and api calls.
It charges from the battery when set not to.
It has limited the charge rate to the "excess solar" when charging from the grid!

All in all, not a great experience, but if the rumours about Octopus integration are founded, we might see some improvement in the api reliability.

peter1rhodes, 2024-09-30:
@LeeFarm11 This thread may be more suitable for our discussion of EV charger connection issues

LeeFarm11, 2024-10-01:
Hi @peter1rhodes 

Further to my other posting seems to have been working all day today. But as mentioned there were no evc records in the exisitng log prior to restarting so something must have gone wrong. I did change the time interval from 10 to 60.

For benefit of other readers I am on 3 phase 11kW Inverter with 20kWh batteries and Giv EV Charger all installed in February this year. Having found HA and predbat earlier this year, I am desperate to get it running on my system but predbat relies on specific items in GivTCP, and Giv (or their Chinese owners) have implemented different logic which has made life difficult for Mark to provide a standard 'interface' for predbat to use. Therefore I am still unable to control my Inverter as I 'should' be able to do.

With regard the EV Charger, I don't have an EV yet, so it seems I am lucky that I don't also have the problems with the Charger to contend with. If I did have an EV seems I would be livid that I could not control my Inverter or my Charger in a reliable way.

Back to GivTCP and EVC. Throughout today I have been receiving very consistently the below messages in the log.
One minute this set of messages -

```
2024-10-01 17:59:35,542 - evc - [DEBUG] - Socket isn't yet open
2024-10-01 17:59:35,560 - evc - [DEBUG] - Socket is still open
2024-10-01 17:59:35,572 - evc - [DEBUG] - Dealt with Charger
2024-10-01 17:59:35,573 - evc - [DEBUG] - Publish all to MQTT
2024-10-01 17:59:35,573 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
```

And the next minute this set of messages -

```
2024-10-01 17:58:35,539 - evc - [DEBUG] - Socket is already open
2024-10-01 17:58:35,541 - evc - [DEBUG] - Error: ('ConnectionException', 'evc.py', 123)
```

And the next minute back to the original messages etc.
Somehow it seems to fail every other minute and succeed every other minute.

Not sure what this might indicate.

closed 2024-10-01 by britkat1980