britkat1980/giv_tcp #186: 3 phase Gen 3 inverter. When will GivTCP be compatible?

LeeFarm11, 2024-05-11:
Hi. I had an 11kW 3 phase Gen 3 inverter installed at end of February. As far as I can tell GivTCP is not yet able to communicate with the inverter. Is there a timescale of when it will be added to GivTCP please?

britkat1980, 2024-05-14:
There is a WIP version in the Beta branch of the addon. Add the following repo to the addon store and install the Beta version.
Feedback welcome, there's only been limited testing (as I don't have a 3ph).

https://github.com/britkat1980/ha-addons

tonka-truck, 2024-05-20:
Hi @britkat1980,
Thanks for this.  I also have a Gen1 Hybrid 3 phase and having found this thread have installed.  After a school boy error of not rebooting HA, I'm now able to run GivTCP and this populates data within MQTT (mosquito on HA).  

I get the following error message though in GivTCP
Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping

Both Battery_Details and bcu_1 data are being published in MQTT with numbers which appear correct so I'm not sure where this error is.

tonka-truck, 2024-05-22:
I've been looking into this a little more and it could be a problem with how the battery data (although as a rank amateur I could be well off the mark) is being captured.

The MQTT raw inverter data picks up the first battery serial number (not sure if all battery serial numbers should be found here?), however raw batteries bcu_1 identifies the number of modules and cells but the serial number for each battery is not being captured, this means that the first battery v_cell and t_cell data is captured against a blank battery serial number and the remaining batteries are skipped.

Hope this helps.

tonka-truck, 2024-05-29:
I've updated my inverter to DA0.003-DD0.003-A0.003 and battery to 2 and now can no longer connect via GivTCP to my inverter with repeated timeout error messages.  Tried obvious rebooting of both inverter and HomeAssistant but no joy.

LeeFarm11, 2024-05-29:
I think I have same problem. For about a week or more I have not been able to get any sensor or settings readings. I have tried latest dev and latest beta both with correct IP address which is fixed IP. But everything in Overview says 'unknown'. I just get constant time outs in the log.
[image]

britkat1980, 2024-05-29:
Ah ok, looks like it’s dying when checking for meters… it should handle that gracefully, but seems like it’s not!
Let me take a look

britkat1980, 2024-05-29:
Actually, that’s not quite true, it seems to continue after that, but what do you get with debug logging on? I want to see how far it’s getting in the process.

tonka-truck, 2024-05-29:
Please see below:

2024-05-29 10:55:27,055 - startup - [DEBUG] - SUPERVISOR_TOKEN is: 86429bbf67722ebd2cbe564afd459fff6a62c07f330e2fe706e89cd733ff48e73f1ebc5d8e8aba7e993275d2eac57f517ede27c21b423788
2024-05-29 10:55:27,059 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-05-29 10:55:27,062 - startup - [INFO] - Supervisor Timezone: Europe/London
2024-05-29 10:55:29,067 - startup - [INFO] - Searching for Inverters
2024-05-29 10:55:29,067 - startup - [DEBUG] - Networks available for scanning are: {0: '192.168.73.1'}
2024-05-29 10:55:29,068 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-05-29 10:55:29,068 - startup - [DEBUG] - EVC- Scanning network (1):192.168.73.1
2024-05-29 10:55:30,186 - startup - [DEBUG] - EVC- Scanning network (2):192.168.73.1
2024-05-29 10:55:31,303 - startup - [DEBUG] - INV- Scanning network (1):192.168.73.1
2024-05-29 10:55:39,687 - startup - [DEBUG] - 1 Inverters found on 192.168.73.1 - {1: '192.168.73.41'}
2024-05-29 10:55:39,688 - startup - [DEBUG] - Getting inverter stats for: 192.168.73.41
2024-05-29 10:56:16,004 - givenergy_modbus_async.client.client - [CRITICAL] - Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x04 base_register=60) after 11 tries at 3s, giving up
2024-05-29 10:56:16,007 - startup - [ERROR] - Gathering inverter details for 192.168.73.41 failed.
2024-05-29 10:56:52,320 - givenergy_modbus_async.client.client - [CRITICAL] - Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x04 base_register=60) after 11 tries at 3s, giving up
2024-05-29 10:56:52,320 - startup - [ERROR] - Gathering inverter details for 192.168.73.41 failed.
2024-05-29 10:56:52,324 - startup - [DEBUG] - GivTCP isAddon: True
2024-05-29 10:56:52,325 - startup - [DEBUG] - Config directory already exists
2024-05-29 10:56:52,326 - startup - [CRITICAL] - Running Redis
2024-05-29 10:56:52,339 - startup - [CRITICAL] - Running Config Frontend
2024-05-29 10:56:52,341 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024-05-29 10:56:52,384 - startup - [DEBUG] - Copying in a template settings.json
2024-05-29 10:56:52,386 - startup - [DEBUG] - Recreating settings.json for invertor 1
2024-05-29 10:56:52,387 - startup - [DEBUG] - Using found MQTT data to autosetup settings.json
2024-05-29 10:56:52,387 - startup - [DEBUG] - Using found Inverter data to autosetup settings.json
2024-05-29 10:56:52,388 - startup - [DEBUG] - Recreating settings.py for invertor 1
2024-05-29 10:56:52,390 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-05-29 10:56:52,390 - startup - [CRITICAL] - Running Invertor (192.168.73.41) read loop every 30/120s
2024-05-29 10:56:52,393 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-05-29 10:56:52,397 - startup - [CRITICAL] - Starting Gunicorn on port 6345
[2024-05-29 10:56:53 +0100] [78] [INFO] Starting gunicorn 22.0.0
[2024-05-29 10:56:53 +0100] [78] [INFO] Listening at: http://0.0.0.0:6345 (78)
[2024-05-29 10:56:53 +0100] [78] [INFO] Using worker: sync
[2024-05-29 10:56:53 +0100] [85] [INFO] Booting worker with pid: 85
[2024-05-29 10:56:53 +0100] [86] [INFO] Booting worker with pid: 86
[2024-05-29 10:56:53 +0100] [87] [INFO] Booting worker with pid: 87
2024-05-29 10:56:54,418 - Inv1 - selector_events -  [DEBUG   ] - Using selector: EpollSelector
2024-05-29 10:56:54,422 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-05-29 10:56:54,428 - Inv1 - read        -  [CRITICAL] - Detecting inverter charateristics...
2024-05-29 10:56:54,775 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto

> givtcp-vuejs@0.0.0 dev
> vite --host -l silent

2024-05-29 10:57:31,023 - Inv1 - client      -  [CRITICAL] - Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x04 base_register=60) after 11 tries at 3s, giving up
2024-05-29 10:57:31,035 - Inv1 - read        -  [ERROR   ] - Error in inital detect/refresh: (<class 'TimeoutError'>, TimeoutError(), <traceback object at 0x7fe4fd006c80>)
2024-05-29 10:57:31,037 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-05-29 10:57:31,045 - Inv1 - read        -  [CRITICAL] - Detecting inverter charateristics...
2024-05-29 10:57:31,266 - Inv1 - client      -  [CRITICAL] - Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x05 base_register=60) after 11 tries at 3s, giving up
2024-05-29 10:57:31,518 - Inv1 - client      -  [CRITICAL] - Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x06 base_register=60) after 11 tries at 3s, giving up
2024-05-29 10:57:31,768 - Inv1 - client      -  [CRITICAL] - Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x07 base_register=60) after 11 tries at 3s, giving up
2024-05-29 10:57:32,260 - Inv1 - client      -  [CRITICAL] - Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x08 base_register=60) after 11 tries at 3s, giving up
2024-05-29 10:58:07,362 - Inv1 - client      -  [CRITICAL] - Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x04 base_register=60) after 11 tries at 3s, giving up
2024-05-29 10:58:07,372 - Inv1 - read        -  [ERROR   ] - Error in inital detect/refresh: (<class 'TimeoutError'>, TimeoutError(), <traceback object at 0x7fe4fd01e800>)
2024-05-29 10:58:07,374 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
[... cut ...]
2024-05-29 11:01:12,766 - Inv1 - client      -  [CRITICAL] - Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x06 base_register=60) after 11 tries at 3s, giving up
2024-05-29 11:01:13,017 - Inv1 - client      -  [CRITICAL] - Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x07 base_register=60) after 11 tries at 3s, giving up
2024-05-29 11:01:13,268 - Inv1 - client      -  [CRITICAL] - Timeout awaiting 2:4/ReadInputRegistersResponse(slave_address=0x08 base_register=60) after 11 tries at 3s, giving up

britkat1980, 2024-05-29:
Ok, so it is meter data failing… not sure why it’s not handling gracefully, I’ll look at this and push out a new dev build today/tomorrow

britkat1980, 2024-05-29:
Try dev version 2.4.269, should now work...

tonka-truck, 2024-05-29:
2.4.269 starts up again and passes data to HA dashboard - yah. 

Unfortunately enabling the web interface it fails and does not start with the following as the final error message 

2024-05-29 13:45:42,587 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available. Double check logs for connection errors and restart GivTCP or ensure correct AIO/firmware settings

LeeFarm11, 2024-05-29:
I too now have readings.
Except - all the power readings are bak to 10x too high which was fixed in DEV 262 but seems now broken again,
Like this - 
[image]

and all the readings of all batteries are still 0C and 0V.
Like this - 
[image]

LeeFarm11, 2024-07-28:
Hi @tonka-truck Do you have any success yet to control (not just report) your Inverter? I have managed to get ge_cloud controlling my Inverter, but not yet GivTCP. I am now struggling with error messages relating to mqtt, but I just wonder if you have yours working before I carry on with trying to work through these errors. My Inverter Firmware is now DA0.004-DD0.004-A0.004 and Battery Firmware is now 402852865. Thanks.

tonka-truck, 2024-07-29:
Hi there,I’ve unfortunately updated the DEV version and ever since I’m no longer getting any updates. I’ve also updated inverter and battery to the same versions. Keep updating to the latest DEV release but still no success unfortunately.

britkat1980, 2024-07-29:
Hi I both, latest Dev and Beta require you to use the web gui to config. It defaults to self_run off to allow you to check the config before it starts. The logs will tell you where to go to configure.
I’d like to get to the bottom of why control seems not to work with the 3ph inverters.
If you get latest beta to work with reads then we can try and work through the control side

Valex1p, 2024-07-30:
Hi @britkat1980 
I've spent the last few days trying to get GIVTCP talking to my 3ph inverter until the penny dropped (helped by @piansom) that they may be a different beast to the 1Ph devices. Anyway I'm hopefully in the right place now.

I've installed 2.4.588 (and already had MQTT running) but get errors:-

The first seems to be a problem connecting to MQTT. I've tried adding a HA user with the same name and password as in the webUI config but no change.

024-07-30 17:15:11,215 - startup - [INFO] - ========================== STARTING GivTCP================================
2024-07-30 17:15:11,239 - startup - [INFO] - HA MQTT Service has been found at core-mosquitto
2024-07-30 17:15:11,362 - startup - [INFO] - IP Address is: 192.168.68.82
2024-07-30 17:15:13,364 - startup - [INFO] - Searching for Inverters
2024-07-30 17:15:13,365 - startup - [INFO] - Scanning network for GivEnergy Devices...
2024-07-30 17:15:38,932 - startup - [INFO] - Inverter TD2343G090 which is a Gen1 - Hybrid_3ph with 6 batteries has been found at: 192.168.68.79
2024-07-30 17:15:38,945 - startup - [INFO] - Running Redis
2024-07-30 17:15:38,959 - startup - [INFO] - Running nginx
2024-07-30 17:15:38,964 - startup - [INFO] - Inverter TD2343G090 found in settings file
2024-07-30 17:15:38,975 - startup - [INFO] - Setting up invertor: 1 of 1
2024-07-30 17:15:39,040 - startup - [INFO] - Removing firstrun
2024-07-30 17:15:39,045 - startup - [INFO] - ==============================================================
2024-07-30 17:15:39,045 - startup - [INFO] - ====             Web Gui Config is at                     ====
2024-07-30 17:15:39,045 - startup - [INFO] - ====     http://192.168.68.82:8099/config.html          ====
2024-07-30 17:15:39,046 - startup - [INFO] - ==============================================================
2024-07-30 17:15:39,056 - startup - [INFO] - Running RQ worker to queue and process givernergy-modbus calls
2024-07-30 17:15:39,057 - startup - [INFO] - Running Invertor (192.168.68.79) read loop every 30/120s
2024-07-30 17:15:39,068 - startup - [INFO] - Starting Gunicorn on port 6345
[2024-07-30 17:15:39 +0100] [81] [INFO] Starting gunicorn 22.0.0
[2024-07-30 17:15:39 +0100] [81] [INFO] Listening at: http://0.0.0.0:6345 (81)
[2024-07-30 17:15:39 +0100] [81] [INFO] Using worker: sync
[2024-07-30 17:15:39 +0100] [82] [INFO] Booting worker with pid: 82
[2024-07-30 17:15:39 +0100] [83] [INFO] Booting worker with pid: 83
[2024-07-30 17:15:39 +0100] [84] [INFO] Booting worker with pid: 84
2024-07-30 17:15:41,721 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-07-30 17:15:41,727 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...
2024-07-30 17:15:57,521 - Inv1 - read        -  [INFO    ] - No day/Night rate info so reverting to day
2024-07-30 17:15:57,536 - Inv1 - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
2024-07-30 17:15:57,574 - Inv1 - HA_Discovery -  [ERROR   ] - Error connecting to MQTT Broker: ('KeyError', 'HA_Discovery.py', 69)

Then you get in a loop of failed updates:-

2024-07-30 17:16:34,445 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ('KeyError', 'read.py', 1462)
2024-07-30 17:17:06,963 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ('KeyError', 'read.py', 1462)
2024-07-30 17:17:41,288 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ('KeyError', 'read.py', 1462)

Any suggestions?

britkat1980, 2024-07-30:
Ah ok, so it’s connecting and getting data, but I’ve got a name wrong somewhere… let me check the code

>
> Hi  @britkat1980 (https://github.com/britkat1980)
I've spent the last few days trying to get GIVTCP talking to my 3ph inverter until the penny dropped (helped by  @PianSom (https://github.com/PianSom)) that they may be a different beast to the 1Ph devices. Anyway I'm hopefully in the right place now.
> I've installed 2.4.588 (and already had MQTT running) but get errors:-
>
>
> The first seems to be a problem connecting to MQTT. I've tried adding a HA user with the same name and password as in the [...]
2024-07-30 17:15:11,239 - startup - [INFO] - HA MQTT Service has been found at core-mosquitto
2024-07-30 17:15:11,362 - startup - [INFO] - IP Address is: 192.168.68.82
2024-07-30 17:15:13,364 - startup - [INFO] - Searching for Inverters
2024-07-30 17:15:13,365 - startup - [INFO] - Scanning network for GivEnergy Devices...
2024-07-30 17:15:38,932 - startup - [INFO] - Inverter TD2343G090 which is a Gen1 - Hybrid_3ph with 6 batteries has been found at: 192.168.68.79
2024-07-30 17:15:38,945 - startup - [INFO] - Running Redis
2024-07-30 17:15:38,959 - startup - [INFO] - Running nginx
2024-07-30 17:15:38,964 - startup - [INFO] - Inverter TD2343G090 found in settings file
2024-07-30 17:15:38,975 - startup - [INFO] - Setting up invertor: 1 of 1
2024-07-30 17:15:39,040 - startup - [INFO] - Removing firstrun
2024-07-30 17:15:39,045 - startup - [INFO] - ==============================================================
2024-07-30 17:15:39,045 - startup - [INFO] - ==== Web Gui Config is at ====
2024-07-30 17:15:39,045 - startup - [INFO] - ====  http://192.168.68.82:8099/config.html  ====
2024-07-30 17:15:39,046 - startup - [INFO] - ==============================================================
2024-07-30 17:15:39,056 - startup - [INFO] - Running RQ worker to queue and process givernergy-modbus calls
2024-07-30 17:15:39,057 - startup - [INFO] - Running Invertor (192.168.68.79) read loop every 30/120s
2024-07-30 17:15:39,068 - startup - [INFO] - Starting Gunicorn on port 6345
[2024-07-30 17:15:39 +0100] [81] [INFO] Starting gunicorn 22.0.0
[2024-07-30 17:15:39 +0100] [81] [INFO] Listening at:  http://0.0.0.0:6345  (81)
[2024-07-30 17:15:39 +0100] [81] [INFO] Using worker: sync
[2024-07-30 17:15:39 +0100] [82] [INFO] Booting worker with pid: 82
[2024-07-30 17:15:39 +0100] [83] [INFO] Booting worker with pid: 83
[2024-07-30 17:15:39 +0100] [84] [INFO] Booting worker with pid: 84
2024-07-30 17:15:41,721 - Inv1 - read - [INFO ] - Starting watch_plant loop...
2024-07-30 17:15:41,727 - Inv1 - read - [CRITICAL] - Detecting inverter characteristics...
2024-07-30 17:15:57,521 - Inv1 - read - [INFO ] - No day/Night rate info so reverting to day
2024-07-30 17:15:57,536 - Inv1 - read - [INFO ] - Publishing Home Assistant Discovery messages
2024-07-30 17:15:57,574 - Inv1 - HA_Discovery - [ERROR ] - Error connecting to MQTT Broker: ('KeyError', 'HA_Discovery.py', 69)
>
>
> Then you get in a loop of failed updates:-
>
>
> 2024-07-30 17:16:34,445 - Inv1 - read - [ERROR ] - inverter Update failed so using last known good data from cache: ('KeyError', 'read.py', 1462)
2024-07-30 17:17:06,963 - Inv1 - read - [ERROR ] - inverter Update failed so using last known good data from cache: ('KeyError', 'read.py', 1462)
2024-07-30 17:17:41,288 - Inv1 - read - [ERROR ] - inverter Update failed so using last known good data from cache: ('KeyError', 'read.py', 1462)

britkat1980, 2024-07-30:
Can you turn on debug and share logs, I need to find what key it is failing on. (I don't have access to a 3ph, so its tricky to diagnose)

Valex1p, 2024-07-30:
Here you go:-

2024-07-30 17:15:41,721 - read - [INFO] - Starting watch_plant loop...
2024-07-30 17:15:41,727 - read - [CRITICAL] - Detecting inverter characteristics...
2024-07-30 17:15:57,521 - read - [INFO] - No day/Night rate info so reverting to day
2024-07-30 17:15:57,536 - read - [INFO] - Publishing Home Assistant Discovery messages
2024-07-30 17:15:57,574 - HA_Discovery - [ERROR] - Error connecting to MQTT Broker: ('KeyError', 'HA_Discovery.py', 69)
2024-07-30 17:16:34,445 - read - [ERROR] - inverter Update failed so using last known good data from cache: ('KeyError', 'read.py', 1462)
2024-07-30 17:17:06,963 - read - [ERROR] - inverter Update failed so using last known good data from cache: ('KeyError', 'read.py', 1462)
2024-07-30 17:17:41,288 - read - [ERROR] - inverter Update failed so using last known good data from cache: ('KeyError', 'read.py', 1462)
2024-07-30 17:53:24,106 - read - [INFO] - Starting watch_plant loop...
2024-07-30 17:53:24,111 - read - [CRITICAL] - Detecting inverter characteristics...
2024-07-30 17:53:42,654 - read - [INFO] - No day/Night rate info so reverting to day
2024-07-30 17:53:42,670 - read - [INFO] - Publishing Home Assistant Discovery messages
2024-07-30 17:53:42,709 - HA_Discovery - [ERROR] - Error connecting to MQTT Broker: ('KeyError', 'HA_Discovery.py', 69)
2024-07-30 17:54:18,218 - read - [ERROR] - inverter Update failed so using last known good data from cache: ('KeyError', 'read.py', 1462)
2024-07-30 17:54:52,630 - read - [ERROR] - inverter Update failed so using last known good data from cache: ('KeyError', 'read.py', 1462)
2024-07-30 17:54:53,169 - write - [CRITICAL] - Restarting the GivTCP Addon in 2s...
2024-07-30 17:55:22,705 - selector_events - [DEBUG] - Using selector: EpollSelector
2024-07-30 17:55:22,709 - read - [INFO] - Starting watch_plant loop...
2024-07-30 17:55:22,712 - GivLUT - [DEBUG] - Opening Modbus Connection to 192.168.68.79
2024-07-30 17:55:22,717 - read - [CRITICAL] - Detecting inverter characteristics...
2024-07-30 17:55:40,200 - read - [DEBUG] - Running full refresh
2024-07-30 17:55:40,200 - read - [DEBUG] - Running processData
2024-07-30 17:55:40,201 - read - [DEBUG] - Beginning parsing of Inverter data
2024-07-30 17:55:40,218 - read - [DEBUG] - Getting Battery Details
2024-07-30 17:55:40,221 - read - [DEBUG] - Building battery output: 
2024-07-30 17:55:40,222 - read - [DEBUG] - Battery HY2343G426 added
2024-07-30 17:55:40,223 - read - [DEBUG] - Building battery output: 
2024-07-30 17:55:40,224 - read - [DEBUG] - Battery HY2343G439 added
2024-07-30 17:55:40,225 - read - [DEBUG] - Building battery output: 
2024-07-30 17:55:40,226 - read - [DEBUG] - Battery HY2343G359 added
2024-07-30 17:55:40,227 - read - [DEBUG] - Building battery output: 
2024-07-30 17:55:40,228 - read - [DEBUG] - Battery HY2343G487 added
2024-07-30 17:55:40,228 - read - [DEBUG] - Building battery output: 
2024-07-30 17:55:40,230 - read - [DEBUG] - Battery HY2343G366 added
2024-07-30 17:55:40,230 - read - [DEBUG] - Building battery output: 
2024-07-30 17:55:40,232 - read - [DEBUG] - Battery HY2343G470 added
2024-07-30 17:55:40,233 - read - [DEBUG] - Getting mode control figures
2024-07-30 17:55:40,234 - read - [DEBUG] - Calculating Mode...
2024-07-30 17:55:40,235 - read - [DEBUG] - Mode is: Eco (Paused)
2024-07-30 17:55:40,236 - read - [DEBUG] - Force Export is not Running
2024-07-30 17:55:40,238 - read - [DEBUG] - Getting TimeSlot data
2024-07-30 17:55:40,238 - read - [DEBUG] - Getting TimeSlot data
2024-07-30 17:55:40,244 - read - [DEBUG] - Starting ratecalcs...
2024-07-30 17:55:40,274 - read - [DEBUG] - Night Start= Tue Jul 30 23:30:00 2024
2024-07-30 17:55:40,274 - read - [DEBUG] - Day Start= Tue Jul 30 05:30:00 2024
2024-07-30 17:55:40,275 - read - [INFO] - No day/Night rate info so reverting to day
2024-07-30 17:55:40,276 - read - [DEBUG] - Setting Rate to Day
2024-07-30 17:55:40,278 - read - [DEBUG] - First time running so saving AC Charge status
2024-07-30 17:55:40,279 - read - [DEBUG] - Battery rate calcs complete
2024-07-30 17:55:40,283 - read - [DEBUG] - Successfully processed data from: 192.168.68.79
2024-07-30 17:55:40,284 - read - [DEBUG] - processData result: Success processing data
2024-07-30 17:55:40,285 - read - [DEBUG] - Running pubFromPickle
2024-07-30 17:55:40,288 - read - [DEBUG] - Converting Model to publish safe string
2024-07-30 17:55:40,289 - read - [DEBUG] - Converting datetime to publish safe string
2024-07-30 17:55:40,291 - read - [DEBUG] - Dealt with invertor
2024-07-30 17:55:40,292 - read - [DEBUG] - Dealt with HY2343G426
2024-07-30 17:55:40,292 - read - [DEBUG] - Dealt with HY2343G439
2024-07-30 17:55:40,293 - read - [DEBUG] - Dealt with HY2343G359
2024-07-30 17:55:40,294 - read - [DEBUG] - Dealt with HY2343G487
2024-07-30 17:55:40,295 - read - [DEBUG] - Dealt with HY2343G366
2024-07-30 17:55:40,296 - read - [DEBUG] - Dealt with HY2343G470
2024-07-30 17:55:40,296 - read - [DEBUG] - Dealt with Stack_0
2024-07-30 17:55:40,297 - read - [DEBUG] - Dealt with HV_Battery_Stacks
2024-07-30 17:55:40,297 - read - [DEBUG] - Dealt with Meter_ID_1
2024-07-30 17:55:40,298 - read - [DEBUG] - Dealt with meters
2024-07-30 17:55:40,298 - read - [DEBUG] - Dealt with raw
2024-07-30 17:55:40,299 - read - [DEBUG] - Dealt with HY2343G426
2024-07-30 17:55:40,300 - read - [DEBUG] - Dealt with HY2343G439
[... 372 lines cut ...]
2024-07-30 17:57:20,779 - mqtt - [DEBUG] - Prepping HY2343G470 for publishing
[... cut ...]
2024-07-30 17:57:20,889 - mqtt - [DEBUG] - Publishing: GivEnergy/TD2343G090/Energy
2024-07-30 17:57:20,889 - mqtt - [DEBUG] - Prepping Today for publishing
2024-07-30 17:57:20,890 - mqtt - [DEBUG] - Prepping Total for publishing
2024-07-30 17:57:20,891 - mqtt - [DEBUG] - Prepping Rates for publishing
2024-07-30 17:57:20,897 - mqtt - [DEBUG] - Publishing: GivEnergy/TD2343G090/Timeslots
2024-07-30 17:57:20,902 - mqtt - [DEBUG] - Publishing: GivEnergy/TD2343G090/Control
2024-07-30 17:57:20,915 - mqtt - [DEBUG] - Publishing: GivEnergy/TD2343G090/Stats

britkat1980, 2024-07-30:
Thanks, can you install beta 590 and repost same log (should have more debug info)

Valex1p, 2024-07-30:
which version from the repository?

2.4.588  or 2.4.591

Valex1p, 2024-07-30:
@britkat1980 
saw you had updated to 592 so here are the debugs from that run:-

2024-07-30 22:19:29,434 - selector_events - [DEBUG] - Using selector: EpollSelector
2024-07-30 22:19:29,439 - read - [INFO] - Starting watch_plant loop...
2024-07-30 22:19:29,442 - GivLUT - [DEBUG] - Opening Modbus Connection to 192.168.68.79
2024-07-30 22:19:29,447 - read - [CRITICAL] - Detecting inverter characteristics...
2024-07-30 22:19:57,190 - read - [DEBUG] - Running full refresh
2024-07-30 22:19:57,190 - read - [DEBUG] - Running processData
2024-07-30 22:19:57,191 - read - [DEBUG] - Beginning parsing of Inverter data
2024-07-30 22:19:57,209 - read - [DEBUG] - Getting Battery Details
2024-07-30 22:19:57,212 - read - [DEBUG] - Building battery output: 
2024-07-30 22:19:57,213 - read - [DEBUG] - Battery HY2343G426 added
2024-07-30 22:19:57,214 - read - [DEBUG] - Building battery output: 
2024-07-30 22:19:57,215 - read - [DEBUG] - Battery HY2343G439 added
2024-07-30 22:19:57,216 - read - [DEBUG] - Building battery output: 
2024-07-30 22:19:57,217 - read - [DEBUG] - Battery HY2343G359 added
2024-07-30 22:19:57,218 - read - [DEBUG] - Building battery output: 
2024-07-30 22:19:57,219 - read - [DEBUG] - Battery HY2343G487 added
2024-07-30 22:19:57,220 - read - [DEBUG] - Building battery output: 
2024-07-30 22:19:57,221 - read - [DEBUG] - Battery HY2343G366 added
2024-07-30 22:19:57,222 - read - [DEBUG] - Building battery output: 
2024-07-30 22:19:57,223 - read - [DEBUG] - Battery HY2343G470 added
2024-07-30 22:19:57,225 - read - [DEBUG] - Getting mode control figures
2024-07-30 22:19:57,226 - read - [DEBUG] - Calculating Mode...
2024-07-30 22:19:57,227 - read - [DEBUG] - Mode is: Eco (Paused)
2024-07-30 22:19:57,228 - read - [DEBUG] - Force Export is not Running
2024-07-30 22:19:57,229 - read - [DEBUG] - Getting TimeSlot data
2024-07-30 22:19:57,230 - read - [DEBUG] - Getting TimeSlot data
2024-07-30 22:19:57,236 - read - [DEBUG] - Starting ratecalcs...
2024-07-30 22:19:57,266 - read - [DEBUG] - Night Start= Tue Jul 30 23:30:00 2024
2024-07-30 22:19:57,267 - read - [DEBUG] - Day Start= Tue Jul 30 05:30:00 2024
2024-07-30 22:19:57,268 - read - [INFO] - No day/Night rate info so reverting to day
2024-07-30 22:19:57,269 - read - [DEBUG] - Setting Rate to Day
2024-07-30 22:19:57,270 - read - [DEBUG] - First time running so saving AC Charge status
2024-07-30 22:19:57,271 - read - [DEBUG] - Battery rate calcs complete
2024-07-30 22:19:57,275 - read - [DEBUG] - Successfully processed data from: 192.168.68.79
2024-07-30 22:19:57,277 - read - [DEBUG] - processData result: Success processing data
2024-07-30 22:19:57,277 - read - [DEBUG] - Running pubFromPickle
2024-07-30 22:19:57,281 - read - [DEBUG] - Converting Model to publish safe string
2024-07-30 22:19:57,282 - read - [DEBUG] - Converting datetime to publish safe string
2024-07-30 22:19:57,284 - read - [DEBUG] - Dealt with invertor
2024-07-30 22:19:57,285 - read - [DEBUG] - Dealt with HY2343G426
2024-07-30 22:19:57,285 - read - [DEBUG] - Dealt with HY2343G439
2024-07-30 22:19:57,286 - read - [DEBUG] - Dealt with HY2343G359
2024-07-30 22:19:57,287 - read - [DEBUG] - Dealt with HY2343G487
2024-07-30 22:19:57,288 - read - [DEBUG] - Dealt with HY2343G366
2024-07-30 22:19:57,289 - read - [DEBUG] - Dealt with HY2343G470
2024-07-30 22:19:57,289 - read - [DEBUG] - Dealt with Stack_0
2024-07-30 22:19:57,289 - read - [DEBUG] - Dealt with HV_Battery_Stacks
2024-07-30 22:19:57,290 - read - [DEBUG] - Dealt with Meter_ID_1
2024-07-30 22:19:57,291 - read - [DEBUG] - Dealt with meters
2024-07-30 22:19:57,291 - read - [DEBUG] - Dealt with raw
2024-07-30 22:19:57,292 - read - [DEBUG] - Dealt with HY2343G426
2024-07-30 22:19:57,293 - read - [DEBUG] - Dealt with HY2343G439
2024-07-30 22:19:57,293 - read - [DEBUG] - Dealt with HY2343G359
2024-07-30 22:19:57,294 - read - [DEBUG] - Dealt with HY2343G487
2024-07-30 22:19:57,295 - read - [DEBUG] - Dealt with HY2343G366
2024-07-30 22:19:57,295 - read - [DEBUG] - Dealt with HY2343G470
2024-07-30 22:19:57,296 - read - [DEBUG] - Dealt with Battery_Stack_1
2024-07-30 22:19:57,296 - read - [DEBUG] - Dealt with Battery_Details
2024-07-30 22:19:57,297 - read - [DEBUG] - Dealt with Power
2024-07-30 22:19:57,298 - read - [DEBUG] - Dealt with Power
2024-07-30 22:19:57,298 - read - [DEBUG] - Dealt with TD2343G090
2024-07-30 22:19:57,299 - read - [DEBUG] - Dealt with Meter_ID1
2024-07-30 22:19:57,299 - read - [DEBUG] - Dealt with Meter_Details
2024-07-30 22:19:57,300 - read - [DEBUG] - Dealt with Today
2024-07-30 22:19:57,300 - read - [DEBUG] - Dealt with Total
2024-07-30 22:19:57,301 - read - [DEBUG] - Dealt with Rates
2024-07-30 22:19:57,302 - read - [DEBUG] - Dealt with Energy
[... 255 lines cut ...]
2024-07-30 22:21:04,433 - mqtt - [DEBUG] - Prepping HY2343G470 for publishing
2024-07-30 22:21:04,434 - mqtt - [DEBUG] - Prepping Stack_0 for publishing
2024-07-30 22:21:04,435 - mqtt - [DEBUG] - Prepping HV_Battery_Stacks for publishing
2024-07-30 22:21:04,435 - mqtt - [DEBUG] - Prepping Meter_ID_1 for publishing
2024-07-30 22:21:04,436 - mqtt - [DEBUG] - Prepping meters for publishing
2024-07-30 22:21:04,510 - mqtt - [DEBUG] - Publishing: GivEnergy/TD2343G090/Battery_Details
2024-07-30 22:21:04,511 - mqtt - [DEBUG] - Prepping HY2343G426 for publishing
2024-07-30 22:21:04,512 - mqtt - [DEBUG] - Prepping HY2343G439 for publishing
2024-07-30 22:21:04,513 - mqtt - [DEBUG] - Prepping HY2343G359 for publishing
[... cut ...]
2024-07-30 22:21:04,568 - mqtt - [DEBUG] - Publishing: GivEnergy/TD2343G090/Energy
2024-07-30 22:21:04,568 - mqtt - [DEBUG] - Prepping Today for publishing
2024-07-30 22:21:04,570 - mqtt - [DEBUG] - Prepping Total for publishing
2024-07-30 22:21:04,571 - mqtt - [DEBUG] - Prepping Rates for publishing
2024-07-30 22:21:04,576 - mqtt - [DEBUG] - Publishing: GivEnergy/TD2343G090/Timeslots
2024-07-30 22:21:04,583 - mqtt - [DEBUG] - Publishing: GivEnergy/TD2343G090/Control
2024-07-30 22:21:04,590 - mqtt - [DEBUG] - Publishing: GivEnergy/TD2343G090/Stats

Valex1p, 2024-07-31:
Hi @britkat1980 
I see 601 arrived last night. debugs attached.
[log_inv_1 (1).log](https://github.com/user-attachments/files/16440186/log_inv_1.1.log)

QQ - should I be deleting .pkl files before installing each new version?

Valex1p, 2024-07-31:
logs from 602 attached
[log_inv_1 (2).log](https://github.com/user-attachments/files/16446745/log_inv_1.2.log)

britkat1980, 2024-08-01:
I've finally got to looking at this. Try DEV version 610 which should get you back up and running on 3PH inverters. I'd appreciate feedback on any odd looking values and specifically if any control works (it should in theory, but I haven't had access to test)

Valex1p, 2024-08-01:
@britkat1980 
Installed 610. I still get an MQTT connect failure:-

2024-08-01 16:05:13,529 - HA_Discovery - [DEBUG] - Publishing: GivEnergy/TD2343G090/Power
2024-08-01 16:05:13,530 - mqtt - [DEBUG] - Prepping Power for publishing
2024-08-01 16:05:13,535 - HA_Discovery - [DEBUG] - Publishing: GivEnergy/TD2343G090/TD2343G090
2024-08-01 16:05:13,537 - HA_Discovery - [ERROR] - Error connecting to MQTT Broker: ('KeyError', 'HA_Discovery.py', 69)
2024-08-01 16:05:13,538 - read - [DEBUG] - Publish all to MQTT
2024-08-01 16:05:13,539 - mqtt - [DEBUG] - MQTT Connection appears closed, re-opening
2024-08-01 16:05:13,539 - mqtt - [DEBUG] - Opening MQTT Connection to core-mosquitto
2024-08-01 16:05:13,545 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-08-01 16:05:13,745 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-08-01 16:05:13,946 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-08-01 16:05:14,147 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)

Do I have something not set up correctly? I added a user 'addons' with password as per the GivTCP config file

Full log attached
[log_inv_1 (4).log](https://github.com/user-attachments/files/16458451/log_inv_1.4.log)

britkat1980, 2024-08-01:
Try 611, you shouldn’t have to touch MQTT settings unless you’re running in docker and not as an HA addon

Valex1p, 2024-08-01:
Now we are cooking !! 
I'll take a look at all the entities in a little while and see if they seem sensible and update you.

Thank you for your time and effort!

Valex1p, 2024-08-05:
hi @britkat1980 
first set of measurement checks attached. Some values pulled by HA are not exposed in the GE dashboard so they are a little difficult to quantify and also where they look 'wrong' it could well be because the inverter is not populating the data I guess.

[givtcp measurement check 1.pdf](https://github.com/user-attachments/files/16493544/givtcp.measurement.check.1.pdf)

I'll try and look at controls next.

Valex1p, 2024-08-05:
so first look at controls. I went with a change to something fairly innocuous - Charge Target SOC 1.

Current value in both HA and GE dashboard = 80%. I changed HA to 70% but no change at the inverter end. Change the value in GE dashboard and you see that reflected correctly in HA.

Is there a 'submit' button in HA or should the control changes be written automatically on the next refresh?

Valex1p, 2024-08-05:
> so first look at controls. I went with a change to something fairly innocuous - Charge Target SOC 1.
>
> Current value in both HA and GE dashboard = 80%. I changed HA to 70% but no change at the [...]

from log:-

2024-08-05 10:16:38,099 - write - [ERROR] - Setting Charge Target 1 failed: ('KeyError', 'write.py', 916)
2024-08-05 10:22:23,489 - write - [ERROR] - Setting Charge Target 2 failed: ('KeyError', 'write.py', 916)
2024-08-05 10:25:51,355 - write - [ERROR] - Setting Charge Target 1 failed: ('KeyError', 'write.py', 916)

Valex1p, 2024-08-06:
> > > so first look at controls. I went with a change to something fairly innocuous - Charge Target SOC 1.
> > > Current value in both HA and GE dashboard = 80%. I changed HA to 70% but no change at the inverter end. Change the value in GE dashboard and you see that reflected correctly in HA.
> > > Is there a 'submit' button in HA or should the control changes be written automatically on the next refresh?
> > 
> > 
> > from log:-
> > 2024-08-05 10:16:38,099 - write - [ERROR] - Setting Charge Target 1 failed: ('KeyError', 'write.py', 916) 2024-08-05 10:22:23,489 - write - [ERROR] - Setting Charge Target 2 failed: ('KeyError', 'write.py', 916) 2024-08-05 10:25:51,355 - write - [ERROR] - Setting Charge Target 1 failed: ('KeyError', 'write.py', 916)
> 
> and with debugs turned on:-
> 
> 2024-08-06 13:06:49,195 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G090/setChargeTarget1= 70 2024-08-06 13:06:49,339 - read - [DEBUG] - Write Request recieved 2024-08-06 13:06:49,340 - read - [DEBUG] - Command: setChargeTarget2 was sent: {'chargeToPercent': '70', 'slot': 1} 2024-08-06 13:06:49,341 - write - [ERROR] - Setting Charge Target 1 failed: ('KeyError', 'write.py', 916) 2024-08-06 13:06:49,342 - read - [DEBUG] - responses is: [{'id': 'setChargeTarget2', 'result': '{"result": "Setting Charge Target 1 failed: ('KeyError', 'write.py', 916)"}'}]

should have added this is using Dev 620

gcoan, 2024-08-06:
> Current value in both HA and GE dashboard = 80%. I changed HA to 70% but no change at the inverter end. Change the value in GE dashboard and you see that reflected correctly in HA.

This sounds like normal behaviour.

The GE app and the GE portal both show what they last sent to the inverter, whereas HA is directly reading and updating the inverter controls - an update that the portal knows nothing about.

To retrieve the latest value of an inverter setting in the portal, go into the settings and press the little circle next to the control and this will force a retrieve of the current inverter setting into the portal

Valex1p, 2024-08-12:
Hi @britkat1980 

Seem to have a new issue (2.4.640). Values are read and populated OK on startup and for a while after but then stop.

Log seems to suggest a problem with the self run loop?:-

2024-08-12 17:06:03,889 - startup - [INFO] - Restarting Invertor read loop every 60s
2024-08-12 17:06:05,668 - Inv1 - selector_events -  [DEBUG   ] - Using selector: EpollSelector
2024-08-12 17:06:05,670 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-08-12 17:06:05,671 - Inv1 - GivLUT      -  [DEBUG   ] - Opening Modbus Connection to 192.168.68.79
2024-08-12 17:06:05,675 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...
2024-08-12 17:06:18,918 - startup - [ERROR] - Self Run loop process stuck. Killing and restarting...
2024-08-12 17:06:18,919 - startup - [INFO] - Restarting Invertor read loop every 60s
2024-08-12 17:06:20,780 - Inv1 - selector_events -  [DEBUG   ] - Using selector: EpollSelector
2024-08-12 17:06:20,783 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-08-12 17:06:20,783 - Inv1 - GivLUT      -  [DEBUG   ] - Opening Modbus Connection to 192.168.68.79
2024-08-12 17:06:20,787 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...
2024-08-12 17:06:33,945 - startup - [ERROR] - Self Run loop process stuck. Killing and restarting...
2024-08-12 17:06:33,946 - startup - [INFO] - Restarting Invertor read loop every 60s
2024-08-12 17:06:35,762 - Inv1 - selector_events -  [DEBUG   ] - Using selector: EpollSelector
2024-08-12 17:06:35,765 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-08-12 17:06:35,765 - Inv1 - GivLUT      -  [DEBUG   ] - Opening Modbus Connection to 192.168.68.79
2024-08-12 17:06:35,769 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...
2024-08-12 17:06:48,971 - startup - [ERROR] - Self Run loop process stuck. Killing and restarting...
2024-08-12 17:06:48,972 - startup - [INFO] - Restarting Invertor read loop every 60s
2024-08-12 17:06:50,811 - Inv1 - selector_events -  [DEBUG   ] - Using selector: EpollSelector
2024-08-12 17:06:50,813 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-08-12 17:06:50,814 - Inv1 - GivLUT      -  [DEBUG   ] - Opening Modbus Connection to 192.168.68.79
2024-08-12 17:06:50,817 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...

LeeFarm11, 2024-08-14:
Hi @britkat1980 . I'm now having another go to get GivTCP set up with my 3 phase inverter. I am testing with GivTCP-DEV 2.4.642

I have not yet been able to test if any of the controls work yet. Based on feedback from @Valex1p it seems the controls do not yet work but I will try and give feedback on that soon.

Now I see Battery Voltages and Temperatures for the first time. Great.

In the meantime the first problem I have is when I try to use the OPEN WEB UI nothing happens, just a blank screen.
But I can get to the settings via http://192.168.1.204:8099/config.html which is suggested in the log, so I managed to get past that hurdle.

Second problem is that the GivEnergy Dash (localhost:3000) does not show any values -
[image]

I just tried changing a couple of settings. Discharge Target SOC 1 and Discharge Target SOC 10 via the Overview. I left them for a couple of minutes but settings were not successfully changed. I then checked the log and I did get one error message in the Log - Setting Discharge Target 10 failed: ('KeyError', 'write.py', 948)

britkat1980, 2024-08-15:
> should have added this is using Dev 620

Could you try this but setting charge target for slot 2? There's some oddities around slot 1 and which registers it uses

Valex1p, 2024-08-15:
@britkat1980 

boom - right on the money!

So using Dev 643 write to SOC2:-

2024-08-15 13:26:49,741 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G090/setChargeTarget2= 50
2024-08-15 13:26:49,954 - read - [DEBUG] - Write Request recieved
2024-08-15 13:26:49,955 - read - [DEBUG] - Command: setChargeTarget2 was sent: {'chargeToPercent': '50', 'slot': 2}
2024-08-15 13:26:49,956 - write - [DEBUG] - Setting Charge Target 2 to: 50
2024-08-15 13:26:50,103 - write - [DEBUG] - Pushing control update to mqtt: GivEnergy/TD2343G090/Control/Charge_Target_SOC_2 - 50
2024-08-15 13:26:50,105 - GivLUT - [DEBUG] - Opening regcache at: /config/GivTCP/regCache_1.pkl
2024-08-15 13:26:50,117 - write - [DEBUG] - Pushing control update to pkl cache: Charge_Target_SOC_2 - 50
2024-08-15 13:26:50,118 - write - [DEBUG] - Setting Charge Target 2 was a success
2024-08-15 13:26:50,119 - write - [INFO] - Setting Charge Target 2 was a success
2024-08-15 13:26:50,120 - read - [DEBUG] - responses is: [{'id': 'setChargeTarget2', 'result': '{"result": "Setting Charge Target 2 was a success"}'}]

and then reflected in the GE portal remote controls pretty much immediately:-
 
[image]

Is the fix generic for all controls? I've not dared set a control for anything else yet just in case I can't get it back again!

Valex1p, 2024-08-15:
also - does anybody have a description of the functions of each of the controls? Some are obvious (SOC%) others are less obvious (Battery Power Mode; the values in Force Charge etc)

britkat1980, 2024-08-15:
as of 640 calling : http://IPADDRESS:8099/REST1/api provides a json object describing the functions and their usage, that is the best place to start. Let me know if it can be improved with more info/clarity

Valex1p, 2024-08-15:
One of the controls I'm particularly interested in is pausing battery discharge (to stop my non GE EV charger emptying the battery)

There is a givtcp control called Battery Pause Mode with options of 'disabled/pause charge/pause discharge/pause both'

I tried setting to 'pause discharge' and the log suggest the write was successful:-

2024-08-15 14:39:23,722 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G090/setBatteryPauseMode= PauseDischarge
2024-08-15 14:39:24,023 - read - [DEBUG] - Write Request recieved
2024-08-15 14:39:24,024 - read - [DEBUG] - Command: setBatteryPauseMode was sent: {'state': 'PauseDischarge'}
2024-08-15 14:39:24,025 - write - [DEBUG] - Setting Battery Pause Mode to: PauseDischarge
2024-08-15 14:39:24,159 - write - [DEBUG] - Pushing control update to mqtt: GivEnergy/TD2343G090/Control/Battery_pause_mode - PauseDischarge
2024-08-15 14:39:24,160 - GivLUT - [DEBUG] - Opening regcache at: /config/GivTCP/regCache_1.pkl
2024-08-15 14:39:24,183 - write - [DEBUG] - Pushing control update to pkl cache: Battery_pause_mode - PauseDischarge
2024-08-15 14:39:24,186 - write - [DEBUG] - Setting Battery Pause Mode to PauseDischarge was a success
2024-08-15 14:39:24,186 - write - [INFO] - Setting Battery Pause Mode to PauseDischarge was a success
2024-08-15 14:39:24,188 - read - [DEBUG] - responses is: [{'id': 'setChargeTarget2', 'result': '{"result": "Setting Charge Target 2 was a success"}'}, {'id': 'setChargeTarget2', 'result': '{"result": "Setting Charge Target 2 was a success"}'}]

but the inverter didn't actually pause discharge and on the next refresh givtcp reset the value to 'disabled'.

Similar for the Temp pause discharge control except the log reported an error but also success?:-

-08-15 14:51:13,459 - read - [DEBUG] - Command: tempPauseDischarge was sent: 10.0
2024-08-15 14:51:13,460 - write - [DEBUG] - Pausing Discharge for 10.0 minutes
2024-08-15 14:51:13,470 - write - [ERROR] - Pausing Discharge failed: ('TypeError', 'write.py', 1550)
2024-08-15 14:51:13,473 - read - [DEBUG] - responses is: [{'id': 'setChargeTarget2', 'result': '{"result": "Setting Charge Target 2 was a success"}'}, {'id': 'setChargeTarget2', 'result': '{"result": "Setting Charge Target 2 was a success"}'}, {'id': 'setBatteryPauseMode', 'result': '{"result": "Setting Battery Pause Mode to PauseDischarge was a success"}'}]

GE support have previously said the control 'was coming' however didn't specify whether it was a dev for the inverter or for the portal.

britkat1980, 2024-08-15:
Just rechecking the docs, Battery Pause Mode only applies to Single Phase inverters. Will await future firmware to enable it for 3PH

britkat1980, 2024-08-15:
I would use the Battery Discharge Rate and set it to 0W to "pause" discharge

Valex1p, 2024-08-15:
hmm - both Battery charge and discharge rates are already read as 0W by givtcp. Its not a parameter available in GE remote control.

[image]

GE remote control does have charge and discharge %, but not in givtcp controls?

LeeFarm11, 2024-08-15:
I updated to Dev 646. Now all I get is this in the Log repeated over and over again. But nothing in the log file in file editor.... - 

SyntaxError: invalid decimal literal
Traceback (most recent call last):
  File "/app/startup.py", line 15, in <module>
    from GivTCP.givenergy_modbus_async.client.client import Client
  File "/app/GivTCP/givenergy_modbus_async/client/client.py", line 7, in <module>
    from . import commands
  File "/app/GivTCP/givenergy_modbus_async/client/commands.py", line 338
    reg = (getattr(RegisterMap, f'{"TPH_" if '3ph' in inv_type else ""}{"EMS_" if 'ems'  in inv_type else ""}{"DIS" if discharge else ""}CHARGE_TARGET_SOC_{idx}'))
                                              ^

britkat1980, 2024-08-16:
Try Beta649. Tested timeslots and charge target on Three phase and seems to work. Take a look and let me know

LeeFarm11, 2024-08-16:
As mentioned on Facebook ( https://www.facebook.com/groups/1517262158723613/posts/1946135725836252/ ) I have been using Timed Discharge today. This morning (via the Giv Portal) I set Discharge Slots 1 and 2 with Start Time of 0700 and End Time of 1600. And 'Enable Force Discharge' to Enabled. Those settings are still the same. 

GivTCP was not running during today. Tonight I have switched to Beta 649 and Started. In GivTCP Timeslots I do not see any slots set to anything other than 00:00:00. 

I then set Charge end time slot 7 to 11:00:00 via GivTCP Timeslots and it did change it on the Inverter. 

I then set Dicharge end time slot 1 (from 00:00:00) to 17:30:00 via GivTCP Timeslots and it did change it on the Inverter from 16:00:00 to 17:30:00
I then changed (via Giv Portal) the time slot back to 16:00:00.
Then I did a refresh of Overview page in HA and the Dicharge end time slot 1 switched to 00:00:00 even though Inverter is really set to 16:00:00, but Charge end time slot 7 is still showing 11:00:00

So it seems that GivTCP is not reading the Time Slots correctly. It seems to be assuming that they are set to 00:00:00 unless GivTCP sets them AND the read from the Inverter is still agreeing with what GivTCP set it to.

Valex1p, 2024-08-17:
I tried:-

setting discharge 10 start and stop in givtcp. both were written correctly and read by GE
setting discharge 10 start and stop in GE. both were read correctly by givtcp

The rub is that I forgot the upgrade instruction so this was actually using beta 649!

charge and discharge slot 1 settings are still not being read by givtcp - just 0 when GE has values in there, I think @britkat1980 has already said that slot 1 has some odd registers so may not work.

britkat1980, 2024-08-17:
Could someone try again on Dev657 build. I've done a like for like match with the GE portal, so hoping it should all line up for Timeslots and Charge Targets!

LeeFarm11, 2024-08-17:
I've just loaded DEV 657. Seems the problem still exists. Discharge Start and End for Slots 1 and 2 do not seem to be read. They also look different layout in Giv Portal. 1 and 2 look different to 3 to 10.

[image]

Also now unable to set Slot 1 or Slot 2. Log says (for example) Setting Discharge Slot End 1 was a success. But it was not.

britkat1980, 2024-08-17:
Just to confirm, for Discharge slots 1&2, you set them in Portal and that change doesn't appear in GivTCP?

What firmware are you running on your threephase?

FYI
(GivTCP control actions don't automatically show in portal, so you need to force it to update in the portal, before GivTCP changes appear in portal)

LeeFarm11, 2024-08-17:
Very odd. I enabled debug so I could send you log file. But now it works!

LeeFarm11, 2024-08-17:
Just to confirm, for Discharge slots 1&2, you set them in Portal and that change doesn't appear in GivTCP?
Yes that was the case. Now I'm not so sure.

What firmware are you running on your threephase?
Inverter Firmware 4, Battery Firmware 4

FYI
(GivTCP control actions don't automatically show in portal, so you need to force it to update in the portal, before GivTCP changes appear in portal)
Yes I know this. Refreshing Portal is a bit slow most of the time. But yes I definitely refresh Portal before taking any notice of what it is telling me.

LeeFarm11, 2024-08-17:
It seems I can change all slots from GivTCP including Slots 1 and 2. But Slots 1 and 2 are not read by GivTCP. I just changed end time slot 1 on Portal, then end time slot 2, then 3. Only the change to slot 3 now shows in HA. HA still show the old times for slot 1 and slot 2.

LeeFarm11, 2024-08-17:
Same story on start time slots. I just changed start time slot 1 on Portal, then start time slot 2, then 3. Only the change to slot 3 now shows in HA. HA still show the old times for slot 1 and slot 2.

LeeFarm11, 2024-08-17:
And just to confirm the same problem does NOT exist with Charge slots 1 and 2. Only Discharge Slots 1 and 2

britkat1980, 2024-08-17:
Try Dev659, I think I've fixed the read issue...

LeeFarm11, 2024-08-17:
Tried Dev660. That seems to have solved it.
Now to work out what all the other buttons do and how they relate to the commands in the Portal. Is there an explanation somewhere please?
And should this all now work with Predbat?
Thanks

britkat1980, 2024-08-17:
Predbat should work, not 100% sure as I don’t use it myself.
For other commands, if you can see what you have on portal and then let me know if there’s any that don’t appear in GivTCP

LeeFarm11, 2024-08-17:
These are what I have. Main one I use is Enable Force Discharge, but if I set it in Portal, I don't see anything change in GivTCP.

[image]

LeeFarm11, 2024-08-17:
I tried changing Set Discharge Rate (to 5000 whatever that means), and now GivTCP seems to be some kind of loop. Log file attached.
[log_inv_1.log](https://github.com/user-attachments/files/16646683/log_inv_1.log)

[86 more replies]

closed 2024-08-29 by LeeFarm11