britkat1980/giv_tcp #187: Unable to set battery reserve - Hybrid Gen3

Sc0th, 2024-05-17:
Hi,

Using docker version britkat/giv_tcp-ma:latest

I am unable to set the battery reserve, is this a bug or am I doing it wrong?

[INF] infra01:~# mosquitto_sub -h mqtt -t solar/<serial>/Control/# -v | ts | grep -i reserve
May 17 13:00:30 solar/<serial>/Control/Battery_Power_Reserve 50
May 17 13:00:33 solar/<serial>/Control/setBatteryReserve {reservePercent:15}
May 17 13:00:50 solar/<serial>/Control/Battery_Power_Reserve 50

I am sending the following

mosquitto_pub -h mqtt -t solar/<serial>/Control/setBatteryReserve -m {"reservePercent":"15"}

I see nothing related in the debug logs which suggests to me I maybe sending garbage, any advice would be appreciated.

GIV-HY5.0 - D0.308-A0.308

Cheers!

britkat1980, 2024-05-29:
when sending a command its on the following topic:
GivEnergy/control/<myserial>/<command> (assuming you've changed the GivEnergy Topic to Solar)

control comes first, rather than myserial

Sc0th, 2024-06-02:
Thanks for getting back to me, just tried this:

`mosquitto_pub -h mqtt -t solar/control/myserial/setBatteryReserve -m {"reservePercent":"15"}`

It's still not playing :(

Sc0th, 2024-06-02:
ok, this works

`mosquitto_pub -h mqtt -t solar/control/myserial/setBatteryReserve -m 15`

Do the docs need a tweak or am I just interpreting them incorrectly?

closed 2024-10-01 by britkat1980