britkat1980/giv_tcp #189: AIO unit no serial number found in MQTT in HA any news...

TheJ00ker, 2024-06-12:
2024-06-12 19:40:00,862 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-06-12 19:40:00,864 - startup - [INFO] - Supervisor Timezone: Europe/London
2024-06-12 19:40:02,869 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-06-12 19:40:06,955 - startup - [CRITICAL] - Inverter CH2344G120 which is a Gen 1 - All in One has been found at: 192.168.0.65
2024-06-12 19:40:06,955 - startup - [INFO] - Searching for Inverters again
2024-06-12 19:40:06,956 - startup - [CRITICAL] - Running Redis
2024-06-12 19:40:06,958 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024/06/12 19:40:06 [notice] 69#69: using the "epoll" event method
2024/06/12 19:40:06 [notice] 69#69: nginx/1.20.2
2024/06/12 19:40:06 [notice] 69#69: OS: Linux 6.6.29-haos
2024/06/12 19:40:06 [notice] 69#69: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/06/12 19:40:06 [notice] 69#69: start worker processes
2024/06/12 19:40:06 [notice] 69#69: start worker process 73
2024/06/12 19:40:06 [notice] 69#69: start worker process 74
2024/06/12 19:40:06 [notice] 69#69: start worker process 75
2024/06/12 19:40:06 [notice] 69#69: start worker process 76
2024/06/12 19:40:06 [notice] 69#69: start worker process 77
2024/06/12 19:40:06 [notice] 69#69: start worker process 78
2024/06/12 19:40:06 [notice] 69#69: start worker process 79
2024/06/12 19:40:06 [notice] 69#69: start worker process 80
2024-06-12 19:40:06,968 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-06-12 19:40:06,968 - startup - [CRITICAL] - Running Invertor (192.168.0.65) read loop every 30s
2024-06-12 19:40:06,969 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-06-12 19:40:06,969 - startup - [CRITICAL] - Starting Gunicorn on port 6345
2024-06-12 19:40:06,970 - startup - [CRITICAL] - Serving Web Dashboard from port 3042
[2024-06-12 19:40:07 +0100] [84] [INFO] Starting gunicorn 21.2.0
[2024-06-12 19:40:07 +0100] [84] [INFO] Listening at: http://0.0.0.0:6345 (84)
[2024-06-12 19:40:07 +0100] [84] [INFO] Using worker: sync
[2024-06-12 19:40:07 +0100] [96] [INFO] Booting worker with pid: 96
[2024-06-12 19:40:07 +0100] [99] [INFO] Booting worker with pid: 99
[2024-06-12 19:40:07 +0100] [101] [INFO] Booting worker with pid: 101
2024-06-12 19:40:07,226 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
 UPDATE  The latest version of `serve` is 14.2.3
 INFO  Accepting connections at http://localhost:3042
2024-06-12 19:40:15,471 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2024-06-12 19:41:47,348 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
2024-06-12 19:46:19,453 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2024-06-12 19:52:23,562 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2024-06-12 19:58:28,313 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force

norfolkmustard, 2024-06-14:
In case it helps.

I had an AIO & gateway, and a Giv EV Charger installed today - the AIO and gateway are giving the same error as above, but the EVCharger is working fine, first attempt.

logs for that:-

2024-06-14 19:39:21,457 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-06-14 19:39:21,468 - startup - [INFO] - Supervisor Timezone: Europe/London
2024-06-14 19:39:23,484 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-06-14 19:39:25,977 - startup - [CRITICAL] - GivEVC found at: 10.10.20.72
2024-06-14 19:39:27,353 - startup - [CRITICAL] - Inverter GW######## which is a Gen 1 - Gateway has been found at: 10.10.20.165
2024-06-14 19:39:27,489 - startup - [CRITICAL] - Inverter CH######## which is a Gen 1 - All in One has been found at: 10.10.20.183
2024-06-14 19:39:27,490 - startup - [INFO] - Searching for Inverters again
2024-06-14 19:39:27,501 - startup - [CRITICAL] - Running Redis
2024-06-14 19:39:27,510 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024/06/14 19:39:27 [notice] 68#68: using the "epoll" event method
2024/06/14 19:39:27 [notice] 68#68: nginx/1.20.2
2024/06/14 19:39:27 [notice] 68#68: OS: Linux 6.6.28-haos-raspi
2024/06/14 19:39:27 [notice] 68#68: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/06/14 19:39:27 [notice] 68#68: start worker processes
2024/06/14 19:39:27 [notice] 68#68: start worker process 73
2024/06/14 19:39:27 [notice] 68#68: start worker process 74
2024/06/14 19:39:27 [notice] 68#68: start worker process 75
2024/06/14 19:39:27 [notice] 68#68: start worker process 76
2024-06-14 19:39:27,580 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-06-14 19:39:27,581 - startup - [CRITICAL] - Running Invertor (10.10.20.183) read loop every 15s
2024-06-14 19:39:27,588 - startup - [CRITICAL] - Running EVC read loop every 5s
2024-06-14 19:39:27,597 - startup - [CRITICAL] - Subscribing MQTT Broker for EVC control
2024-06-14 19:39:27,627 - startup - [CRITICAL] - Setting chargeMode loop to manage different charge modes every 60s
2024-06-14 19:39:27,628 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-06-14 19:39:27,650 - startup - [CRITICAL] - Starting Gunicorn on port 6345
[2024-06-14 19:39:28 +0100] [83] [INFO] Starting gunicorn 21.2.0
[2024-06-14 19:39:28 +0100] [83] [INFO] Listening at: http://0.0.0.0:6345 (83)
[2024-06-14 19:39:28 +0100] [83] [INFO] Using worker: sync
[2024-06-14 19:39:28 +0100] [84] [INFO] Booting worker with pid: 84
[2024-06-14 19:39:28 +0100] [86] [INFO] Booting worker with pid: 86
[2024-06-14 19:39:29 +0100] [88] [INFO] Booting worker with pid: 88
2024-06-14 19:39:30,390 - Inv1 - mqtt_client_evc -  [CRITICAL] - Connecting to MQTT broker for EVC control- core-mosquitto
2024-06-14 19:39:30,605 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2024-06-14 19:39:30,657 - Inv1 - evc         -  [CRITICAL] - Publishing Home Assistant Discovery messages
2024-06-14 19:39:39,270 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2024-06-14 19:41:10,729 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'. Did you mean: 'serial_number_evc'?
2024-06-14 19:43:34,135 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...

gcoan, 2024-06-14:
Have you tried the beta version of the GivTCP add-on?  I'm not using it but I understand it has a number of improvements such as battery data for the AIO.  Worth trying to see if it fixes this issue

TheJ00ker, 2024-06-15:
thanks @gcoan i managed to get intouch with someone who diagnosed the battery and updated the beat to support the All in one. beta installed with some tweeks to it by a kind gentleman and i'm working. I belive there is no way it would have worked without a few tweeks..

gcoan, 2024-06-15:
Great, glad you are sorted

gazm2k5, 2024-07-04:
> thanks @gcoan i managed to get intouch with someone who diagnosed the battery and updated the beat to support the All in one. beta installed with some tweeks to it by a kind gentleman and i'm working. I belive there is no way it would have worked without a few tweeks..

Can you explain how you got it working as I'm facing the same issues with 2.4.3 and a GivEnergy AIO. I've been hitting my head against a wall for ages trying to get this to work.

gcoan, 2024-07-04:
To install the Beta branch of the addon. Add the following repo to the addon store and install the Beta version.

https://github.com/britkat1980/ha-addons

Or pull the docker file from the appropriate sub directory

norfolkmustard, 2024-07-05:
> To install the Beta branch of the addon. Add the following repo to the addon store and install the Beta version.
> 
> https://github.com/britkat1980/ha-addons
> 
> Or pull the docker file from the appropriate sub directory

That's working a treat for me, thanks. Now have data from the AIO and EV Charger in HA

gazm2k5, 2024-07-05:
> That's working a treat for me, thanks. Now have data from the AIO and EV Charger in HA

This is promising to hear! I will give it a try.

I'm not using HA OS so I don't have access to the addon store and will be running the container manually in docker. How do I get data integrated into Home Assistant this way?

gazm2k5, 2024-07-07:
> To install the Beta branch of the addon. Add the following repo to the addon store and install the Beta version.
> 
> https://github.com/britkat1980/ha-addons
> 
> Or pull the docker file from the appropriate sub directory

Unfortunately I tried building the docker image for the beta in this link but it failed to build with this error:

`ERROR: failed to solve: failed to compute cache key: failed to calculate checksu                     m of ref 878ca2e9-5652-49a2-a5d4-bdccb76801e5::y8wm91hil8u8tdw48rxcbf6az: "/inde                     x.html": not found`

I don't see an index.html in the repo. I guess it only works on home assistant OS's addon store.

davebuk, 2024-07-07:
@gazm2k5 
I trying to do something similar. I'm trying to run GivTCP in a Docker Desktop (windows) container and use the MQTT with openHAB.

After the container starts I get the following even though I have set the MQTT topic myself in the `docker-compose.yml` file.
```
2024-07-07 19:49:20 2024-07-07 19:49:20,705 - startup - [CRITICAL] - SUPERVISOR TOKEN does not exist
2024-07-07 19:49:22 2024-07-07 19:49:22,708 - startup - [CRITICAL] - Scanning network for inverters...
2024-07-07 19:49:25 2024-07-07 19:49:25,364 - startup - [CRITICAL] - No inverters found...
2024-07-07 19:49:25 2024-07-07 19:49:25,365 - startup - [CRITICAL] - Running Redis
2024-07-07 19:49:25 2024-07-07 19:49:25,367 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024-07-07 19:49:25 2024/07/07 19:49:25 [notice] 48#48: using the "epoll" event method
2024-07-07 19:49:25 2024/07/07 19:49:25 [notice] 48#48: nginx/1.20.2
2024-07-07 19:49:25 2024/07/07 19:49:25 [notice] 48#48: OS: Linux 5.15.153.1-microsoft-standard-WSL2
2024-07-07 19:49:25 2024/07/07 19:49:25 [notice] 48#48: getrlimit(RLIMIT_NOFILE): 1048576:1048576
2024-07-07 19:49:25 2024/07/07 19:49:25 [notice] 48#48: start worker processes
2024-07-07 19:49:25 2024/07/07 19:49:25 [notice] 48#48: start worker process 49
2024-07-07 19:49:25 2024/07/07 19:49:25 [notice] 48#48: start worker process 50
2024-07-07 19:49:25 2024/07/07 19:49:25 [notice] 48#48: start worker process 51
2024-07-07 19:49:25 2024/07/07 19:49:25 [notice] 48#48: start worker process 52
2024-07-07 19:49:26  UPDATE  The latest version of `serve` is 14.2.3
2024-07-07 19:49:27  INFO  Accepting connections at http://localhost:3001
2024-07-07 19:49:25 2024-07-07 19:49:25,395 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-07-07 19:49:25 2024-07-07 19:49:25,395 - startup - [CRITICAL] - Running Invertor read loop every 15s
2024-07-07 19:49:25 2024-07-07 19:49:25,396 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-07-07 19:49:25 2024-07-07 19:49:25,397 - startup - [CRITICAL] - Starting Gunicorn on port 6345
2024-07-07 19:49:25 2024-07-07 19:49:25,404 - startup - [CRITICAL] - Serving Web Dashboard from port 3001
2024-07-07 19:49:25 [2024-07-07 19:49:25 +0100] [59] [INFO] Starting gunicorn 21.2.0
2024-07-07 19:49:25 [2024-07-07 19:49:25 +0100] [59] [INFO] Listening at: http://0.0.0.0:6345 (59)
2024-07-07 19:49:25 [2024-07-07 19:49:25 +0100] [59] [INFO] Using worker: sync
2024-07-07 19:49:25 [2024-07-07 19:49:25 +0100] [71] [INFO] Booting worker with pid: 71
2024-07-07 19:49:25 [2024-07-07 19:49:25 +0100] [72] [INFO] Booting worker with pid: 72
2024-07-07 19:49:25 [2024-07-07 19:49:25 +0100] [73] [INFO] Booting worker with pid: 73
2024-07-07 19:49:26 2024-07-07 19:49:26,601 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- 192.168.1.100
2024-07-07 19:51:06 2024-07-07 19:51:06,704 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
2024-07-07 19:51:06 Traceback (most recent call last):
2024-07-07 19:51:06   File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
2024-07-07 19:51:06     client.loop_forever()
2024-07-07 19:51:06   File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
2024-07-07 19:51:06     rc = self._loop(timeout)
2024-07-07 19:51:06   File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
2024-07-07 19:51:06     rc = self.loop_read()
2024-07-07 19:51:06   File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
2024-07-07 19:51:06     rc = self._packet_read()
2024-07-07 19:51:06   File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
2024-07-07 19:51:06     rc = self._packet_handle()
2024-07-07 19:51:06   File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
2024-07-07 19:51:06     return self._handle_connack()
2024-07-07 19:51:06   File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
2024-07-07 19:51:06     on_connect(
2024-07-07 19:51:06   File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
2024-07-07 19:51:06     client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
2024-07-07 19:51:06 AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
2024-07-07 19:52:28 2024-07-07 19:52:28,811 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2024-07-07 19:56:01 2024-07-07 19:56:01,471 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
```

gazm2k5, 2024-07-09:
> @gazm2k5 I trying to do something similar. I'm trying to run GivTCP in a Docker Desktop (windows) container and use the MQTT with openHAB.
>
> After the container starts I get the following even though [...]

I managed to get mine working. Do you have an AIO? I was missing an environmental variable from my docker compose, `INVERTOR_AIO_1=True` which I only found because my friend sent me his env vars from his HA OS implementation, where he simply gets a toggle on the UI for this option. The env var doesn't seem to be documented anywhere.

davebuk, 2024-07-09:
Hi. Yes I have the AIO and the `INVERTOR_AIO_1=True` set in the `.yml` file.

gazm2k5, 2024-07-09:
> Hi. Yes I have the AIO and the `INVERTOR_AIO_1=True` set in the `.yml` file.
> […](#)

And number of batteries set to 0? Here is my config:

```
  givtcp:
    container_name: givtcp
    image: britkat/giv_tcp-ma:latest
    restart: unless-stopped
    ports:
      - "6345:6345"                   # REST (add increments for each inverter)
      - "3005:3005"                   # This should match the WEB_DASH_PORT ENV below
    environment:
      - TZ=Europe/London
      - NUMINVERTORS=1
      - INVERTOR_IP_1=192.168.4.3
      - INVERTOR_AIO_1=True
      - INVERTOR_AC_1=False
      - NUMBATTERIES_1=0              # Number of battery modules installed and connected to the above inverter
      - MQTT_OUTPUT=True              # "True" if you want to publish your data to MQTT, "False" otherwise
      - MQTT_ADDRESS=mosquitto
      - MQTT_USERNAME=[redacted]
      - MQTT_PASSWORD=[redacted]
      - MQTT_TOPIC=GivEnergy/Data     # Root topic to publish data to. If left blank it will default to GivEnergy/<invertor_serial_number>
      - MQTT_PORT=1883                # Port of your existing broker, leave as "1883" for internal broker
      - HA_AUTO_D=True
      - LOG_LEVEL=Error
      - HOSTIP=192.168.2.1
      - WEB_DASH=True                 # Enable the wed dashboard
      - WEB_DASH_PORT=3005            # Port to serve the web dashboard on
```
Also check INVERTOR_IP_1 is correct as I believe the name of this env var changed at one point.

davebuk, 2024-07-09:
My config is shown in #194. I'll play with some other settings and see what I can achieve.

gcoan, 2024-07-09:
Do put in a PR to enhance the documentation if there are gaps in explaining how to get your inverter working with givtcp

davebuk, 2024-07-09:
If I can find a way of getting it working WRT post #194 I'll add some details and suggest a PR.

davebuk, 2024-07-15:
I've managed to get something working in post #194. I'll add some configs to openHAB over the weekend and see if I can integrate GivTCP/Docker into openHAB using MQTT.

closed 2024-10-01 by britkat1980