britkat1980/giv_tcp #193: GivTCP-Beta Charge Slots & SOC % Limit

rupi08, 2024-07-05:
GivTCP has 10 Charge Slots with individual SOC % Limits,  could this be integrated into GivTCP-Beta ?

britkat1980, 2024-07-05:
What model and firmware is your inverter. 10 slots do exist in GivTCP, but only for Gen3 models. Perhaps it’s not correctly identifying your model correctly.

rupi08, 2024-07-05:
I used to use GivTCP with 1 AIO which displayed the 10 slots, I recently upgraded to the Beta Parallel setup which has 2 AIOs with GW.  I believe now the GW controls everything.  I am now using GivTCP-Beta.
I am only seeing 1 Charge Slot even though the GW has 10 available.
GW firmware is A0.010, the AIOs are on D0.612-A0.612 but I don’t think the AIO is relevant anymore.

britkat1980, 2024-07-05:
Ah, I hadn't included Gateway for getting the 10 slots. It'll be available in the next Beta release.

closed 2024-10-01 by britkat1980