britkat1980/giv_tcp #194: How to get GivTCP to run on Docker Desktop, Windows and configure AIO.

davebuk, 2024-07-06:


davebuk, 2024-07-07:
I have started to get some where.

I have downloaded the code from the 2.4.0 release and saved it to my hard drive. I have edited the `docker-compose.yml` file to suit my AIO install, details below:
```
version: "3.9"
services:
  GivTCP:
    image: britkat/giv_tcp-ma:2.4.0
    ports:
      - "1883:1883"                   # MQTT
      - "6345:6345"                   # REST (add increments for each inverter)
      - "3001:3001"                   # This should match the WEB_DASH_PORT ENV below
      - "6379:6379"                   # Redis port for job queueing
      - "9181:9181"                   # RQ Dashboard to view job scheduling
    environment:
## Critical to update
      - NUMINVERTORS=1                # Set this to the number of Inverters in your setup, then replicate the next two lines for each inverter (changing the last number of the ENV)
      - INVERTOR_IP_1=192.168.1.****                # Set this to the IP address of your Inverter on your local network
      - INVERTOR_AIO_1=True          # Set this to true if you have an AIO
      - INVERTOR_AC_1=False           # Set this to true if you have an AC on old firmware and don't have the new registers
      - NUMBATTERIES_1=0              # Number of battery modules installed and connected to the above inverter
      - INVERTOR_IP_2=                # Set this to the IP address of your Inverter on your local network
      - INVERTOR_AIO_2=False          # Set this to true if you have an AIO
      - INVERTOR_AC_2=False           # Set this to true if you have an AC on old firmware and don't have the new registers
      - NUMBATTERIES_2=1              # Number of battery modules installed and connected to the above inverter
      - INVERTOR_IP_3=                # Set this to the IP address of your Inverter on your local network
      - INVERTOR_AIO_3=False          # Set this to true if you have an AIO
      - INVERTOR_AC_3=False           # Set this to true if you have an AC on old firmware and don't have the new registers
      - NUMBATTERIES_3=1              # Number of battery modules installed and connected to the above inverter
      - MQTT_OUTPUT=True              # "True" if you want to publish your data to MQTT, "False" otherwise
      - MQTT_ADDRESS=192.168.1.****                 # IP address of an existing MQTT broker, or leave as "127.0.0.1" to use the internal broker
      - MQTT_USERNAME=****                # Username of your existing broker, if needed. Not required for internal broker
      - MQTT_PASSWORD=****                # Password of your existing broker, if needed. Not required for internal broker
      - HOSTIP=                       # External IP address of the docker host (needed for web dashboard)
      - CACHELOCATION=/config/GivTCP  # Location of cache data, this folder can be mapped to a persistence storage outside the container
      - TZ=Europe/London              # Set to your Timezone
```
(I had to change port 3000 to 3001 due to the port apparently already being in use. I also set the 'image' value to 2.4.0).

From PowerShell I can 'cd' to the directory where the 'docker-compose.yml' file is and then ran ' docker compose up -d'.

The stack/container is built and the container seems to run but I am getting inverter time out errors.

**QUESTIONS**

- I already have an MQTT broker running on the main machine so set that as the IP address. Should  the MQTT port at the top of the file be changed or even not set in the first place?
- Does the extract from the 'docker-compose.yml' file look correct for a single AIO battery (no gateway) installation running 'D0.611-A0.611' firmware?

davebuk, 2024-07-10:
@britkat1980 
Can you help at all?
Using the following `docker-compose.yml` I get the `debug` listed below.
```
#version: "3.9"
services:
  GivTCP:
    image: britkat/giv_tcp-ma:2.4.2
    ports:
      #- "1883:1883"                   # MQTT
      - "6345:6345"                   # REST (add increments for each inverter)
      - "3001:3001"                   # This should match the WEB_DASH_PORT ENV below
      - "6379:6379"                   # Redis port for job queueing
      #- "9181:9181"                   # RQ Dashboard to view job scheduling
    environment:
## Critical to update
      - NUMINVERTORS=1                # Set this to the number of Inverters in your setup, then replicate the next two lines for each inverter (changing the last number of the ENV)
      - INVERTOR_IP_1=192.168.1.147   # Set this to the IP address of your Inverter on your local network
      - INVERTOR_AIO_1=True           # Set this to true if you have an AIO
      - INVERTOR_AC_1=False           # Set this to true if you have an AC on old firmware and don't have the new registers
      - NUMBATTERIES_1=0              # Number of battery modules installed and connected to the above inverter
      - MQTT_OUTPUT=True              # "True" if you want to publish your data to MQTT, "False" otherwise
      - MQTT_ADDRESS=192.168.1.100    # IP address of an existing MQTT broker, or leave as "127.0.0.1" to use the internal broker
      - MQTT_USERNAME=****      # Username of your existing broker, if needed. Not required for internal broker
      - MQTT_PASSWORD=****        # Password of your existing broker, if needed. Not required for internal broker
      - HOSTIP=192.168.1.100                       # External IP address of the docker host (needed for web dashboard)
      - CACHELOCATION=/config/GivTCP  # Location of cache data, this folder can be mapped to a persistence storage outside the container
      - TZ=Europe/London              # Set to your Timezone
```

```
2024-07-10 20:19:40 GivTCP-1  | 2024-07-10 20:19:40,763 - Inv1 - read        -  [DEBUG   ] - ----------------------------Starting----------------------------
2024-07-10 20:19:40 GivTCP-1  | 2024-07-10 20:19:40,764 - Inv1 - read        -  [DEBUG   ] - Getting All Registers
2024-07-10 20:19:40 GivTCP-1  | 2024-07-10 20:19:40,764 - Inv1 - read        -  [DEBUG   ] - Connecting to: 192.168.1.147
2024-07-10 20:19:40 GivTCP-1  | 2024-07-10 20:19:40,765 - Inv1 - queue       -  [DEBUG   ] - Pushed job 1bfbe48d-e155-4dfe-9c7b-1875830f9eca into GivTCP_1
2024-07-10 20:19:41 GivTCP-1  | 2024-07-10 20:19:41,253 - Inv1 - sync        -  [DEBUG   ] - Connection to Modbus server established. Socket ('172.19.0.2', 58221)
2024-07-10 20:19:41 GivTCP-1  | 2024-07-10 20:19:41,254 - Inv1 - transaction -  [DEBUG   ] - Current transaction state - IDLE
2024-07-10 20:19:41 GivTCP-1  | 2024-07-10 20:19:41,254 - Inv1 - transaction -  [DEBUG   ] - Running transaction 1
2024-07-10 20:19:41 GivTCP-1  | 2024-07-10 20:19:41,255 - Inv1 - sync        -  [DEBUG   ] - New Transaction state 'SENDING'
2024-07-10 20:19:41 GivTCP-1  | 2024-07-10 20:19:41,422 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2024-07-10 20:19:41 GivTCP-1  | 2024-07-10 20:19:41,423 - Inv1 - payload     -  [DEBUG   ] - [b'\x00\x00', b'\x00\x00', b'\x00\x00', b'\x00\x8a']
2024-07-10 20:19:41 GivTCP-1  | 2024-07-10 20:19:41,423 - Inv1 - transaction -  [DEBUG   ] - Adding transaction 1
2024-07-10 20:19:41 GivTCP-1  | 2024-07-10 20:19:41,424 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2024-07-10 20:19:41 GivTCP-1  | 2024-07-10 20:19:41,925 - Inv1 - transaction -  [DEBUG   ] - Current transaction state - TRANSACTION_COMPLETE
2024-07-10 20:19:41 GivTCP-1  | 2024-07-10 20:19:41,925 - Inv1 - transaction -  [DEBUG   ] - Running transaction 2
2024-07-10 20:19:41 GivTCP-1  | 2024-07-10 20:19:41,926 - Inv1 - sync        -  [DEBUG   ] - New Transaction state 'SENDING'
2024-07-10 20:19:41 GivTCP-1  | 2024-07-10 20:19:41,927 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2024-07-10 20:19:41 GivTCP-1  | 2024-07-10 20:19:41,927 - Inv1 - payload     -  [DEBUG   ] - [b'\x00\x00', b'\x00\x00', b'\x00\x00', b'\x00\x8a']
2024-07-10 20:19:41 GivTCP-1  | 2024-07-10 20:19:41,928 - Inv1 - transaction -  [DEBUG   ] - Adding transaction 2
2024-07-10 20:19:41 GivTCP-1  | 2024-07-10 20:19:41,928 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2024-07-10 20:19:42 GivTCP-1  | 2024-07-10 20:19:42,429 - Inv1 - transaction -  [DEBUG   ] - Current transaction state - TRANSACTION_COMPLETE
2024-07-10 20:19:42 GivTCP-1  | 2024-07-10 20:19:42,430 - Inv1 - transaction -  [DEBUG   ] - Running transaction 3
2024-07-10 20:19:42 GivTCP-1  | 2024-07-10 20:19:42,430 - Inv1 - sync        -  [DEBUG   ] - New Transaction state 'SENDING'
2024-07-10 20:19:42 GivTCP-1  | 2024-07-10 20:19:42,431 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2024-07-10 20:19:42 GivTCP-1  | 2024-07-10 20:19:42,432 - Inv1 - payload     -  [DEBUG   ] - [b'\x00\x00', b'\x00\x00', b'\x00\x00', b'\x00\x8a']
2024-07-10 20:19:42 GivTCP-1  | 2024-07-10 20:19:42,432 - Inv1 - transaction -  [DEBUG   ] - Adding transaction 3
[... cut ...]
2024-07-10 20:20:03 GivTCP-1  | 2024-07-10 20:20:03,520 - Inv1 - payload     -  [DEBUG   ] - [b'\x00\x00', b'\x00\x00', b'\x00\x00', b'\x00\x8a']
2024-07-10 20:20:03 GivTCP-1  | 2024-07-10 20:20:03,521 - Inv1 - transaction -  [DEBUG   ] - Adding transaction 7
2024-07-10 20:20:03 GivTCP-1  | 2024-07-10 20:20:03,522 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2024-07-10 20:20:04 GivTCP-1  | 2024-07-10 20:20:04,044 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
```

davebuk, 2024-07-15:
I'm sort of getting somewhere. The only way I can get it to work is to use the BETA HA addon, pull into Docker, then run it from there putting the relevant `ENV` entries in for the AIO. I can't get any of the builds from github working, only the BETA from Docker Hub. I presume these are different sets of code?

Using `docker pull britkat/giv_tcp-beta` in Windows PowerShell, this will import the image into Docker Desktop. When I run the image I can set the `ENV` as detailed below.

```
NUMINVERTORS	1 
INVERTOR_IP_1	192.168.1.***
INVERTOR_AIO_1	True
NUMBATTERIES_1	0 
MQTT_OUTPUT	True 
MQTT_ADDRESS	192.168.1.***
MQTT_USERNAME	**** 
MQTT_PASSWORD	****
```

This all seems to work and I get MQTT messages out but if I want to make any changes, I'd have to start again and re-input the `ENVs`. I also wanted to run the Web Dash UI, but I can't get access to the files in the image to add the port to the container.

If anyone knows of a way of getting access to the image pulled from `docker pull britkat/giv_tcp-beta` so I can change the configuration after the container is built, I may be able to add the Web Dash UI.

davebuk, 2024-07-24:
I've got GivTCP running with Docker Desktop and MQTT/openHAB. As above, I can pull from Docker Hub using PowerShell. Docker Desktop then shows the image that I can run and create a Container from. I filled in the additional setting before running and it all seems to work.

[image]

It doesn't allow the WebUI to work which might be because I can't get port 3001 to appear in the list of ports for the container. Enabling it and setting the ENVs to those below doesn't show the port in the container.
```
WEB_DASH	True 
WEB_DASH_PORT	3001
```
I've only started testing READ values. Below are the ones I'm using at the moment.
openHAB .things
```
Bridge mqtt:broker:pimqttbroker [host="192.168.1.***", secure=false, qos=1, username="***", password="***"] {

Thing topic Giv_tcp "Giv Energy AIO Data" @ "Systems" 
{ Channels: 
    Type number : givtcp_flow_sol-to-house "Watts - Solar to House" [ stateTopic="AIO01/CH***/Power/Flows/Solar_to_House"]
    Type number : givtcp_flow_sol-to-batt "Watts - Solar to Battery" [ stateTopic="AIO01/CH***/Power/Flows/Solar_to_Battery"]   
    Type number : givtcp_flow_sol-to-grid "Watts - Solar to Grid" [ stateTopic="AIO01/CH***/Power/Flows/Solar_to_Grid"]
    Type number : givtcp_flow_bat-to-house "Watts - Battery to House" [ stateTopic="AIO01/CH***/Power/Flows/Battery_to_House"]
    Type number : givtcp_flow_bat-to-grid "Watts - Battery to Grid" [ stateTopic="AIO01/CH***/Power/Flows/Battery_to_Grid"]
    Type number : givtcp_flow_grid-to-house "Watts - Grid to House" [ stateTopic="AIO01/CH***/Power/Flows/Grid_to_House"]
    Type number : givtcp_flow_grid-to-battery "Watts - Grid to Battery" [ stateTopic="AIO01/CH***/Power/Flows/Grid_to_Battery"]
    Type number : givtcp_battery_SOC "Battery State of Charge %" [ stateTopic="AIO01/CH***/Battery_Details/Battery_Stack_1/Stack_SOC_High"]
    Type datetime : givtcp_Stats_Last_Updated_Time "Time and Date of last update" [ stateTopic="AIO01/CH***/Stats/Last_Updated_Time"]
}
}
```
openHAB .items
```
Number:Power givtcpAIO_flow_soltohouse "Watts - Solar to House [%d %unit%]" <energy> { unit="W", channel="mqtt:topic:pimqttbroker:Giv_tcp:givtcp_flow_sol-to-house" }
Number:Power givtcpAIO_flow_soltobatt "Watts - Solar to Battery [%d %unit%]" <energy> { unit="W", channel="mqtt:topic:pimqttbroker:Giv_tcp:givtcp_flow_sol-to-batt" }
Number:Power givtcpAIO_flow_soltogrid "Watts - Solar to Grid [%d %unit%]" <energy> { unit="W", channel="mqtt:topic:pimqttbroker:Giv_tcp:givtcp_flow_sol-to-grid" }
Number:Power givtcpAIO_flow_battohouse "Watts - Battery to House [%d %unit%]" <energy> { unit="W", channel="mqtt:topic:pimqttbroker:Giv_tcp:givtcp_flow_bat-to-house" }
Number:Power givtcpAIO_flow_battogrid "Watts - Battery to Grid [%d %unit%]" <energy> { unit="W", channel="mqtt:topic:pimqttbroker:Giv_tcp:givtcp_flow_bat-to-grid" }
Number:Power givtcpAIO_flow_gridtohouse "Watts - Grid to House [%d %unit%]" <energy> { unit="W", channel="mqtt:topic:pimqttbroker:Giv_tcp:givtcp_flow_grid-to-house" }
Number:Power givtcpAIO_flow_gridtobattery "Watts - Grid to Battery [%d %unit%]" <energy> { unit="W", channel="mqtt:topic:pimqttbroker:Giv_tcp:givtcp_flow_grid-to-battery" }
Number givtcpAIO_battery_SOC "Battery State of Charge [%d %%]" <energy> { channel="mqtt:topic:pimqttbroker:Giv_tcp:givtcp_battery_SOC" }
DateTime givtcpAIO_Stats_Last_Updated_Time "Time and Date of last update [%1$td.%1$tm.%1$tY %1$tH:%1$tM:%1$tS - UTC]" <time> { channel="mqtt:topic:pimqttbroker:Giv_tcp:givtcp_Stats_Last_Updated_Time" }
```
openHAB .sitemap
```
sitemap giv_tcp label="GivTCP Battery Data"
{
Frame label="Power Info" {
Default item=givtcpAIO_flow_soltohouse
Default item=givtcpAIO_flow_soltobatt
Default item=givtcpAIO_flow_soltogrid
Default item=givtcpAIO_flow_battohouse
Default item=givtcpAIO_flow_battogrid
Default item=givtcpAIO_flow_gridtohouse
Default item=givtcpAIO_flow_gridtobattery
}
Frame label="Battery State" {
Default item=givtcpAIO_battery_SOC
Default item=givtcpAIO_Stats_Last_Updated_Time
}
}
```

davebuk, 2024-07-24:
I have also tried downloading the `Docker` file from https://github.com/britkat1980/ha-addons/tree/main/givtcp-beta and using it from within PowerShell to build an image but I can't get it to work.

It would be nice to be able to create a `Docker` file or `compose.yaml`, pre-configure my settings for AIO etc and build the working BETA image with WebUI working.

britkat1980, 2024-07-24:
Web gui runs on port 8099, which serves both the rest interface and config gui.
Open that port on container, or run on host network to allow auto-discovery of inverters on the network.

davebuk, 2024-07-24:
I get '500 Internal Server Error' for port 8099.

davebuk, 2024-07-24:
Some of your `Docker` files have `COPY WebDashboard ./WebDashboard` while others have `COPY WebDashboard/ ./WebDashboard`. Will this `WebDashboard/` have any direct impact?

davebuk, 2024-09-25:
As of 25-Sep-2024 the version 3 from the main branch works by downloading the whole repo to a local directory, going to that directory in powershell (`cd`'ing). I changed the IHD web port to `3001` in the `docker-compose.yml` file as I have another service running on `3000`. I then ran `docker compose up` in powershell. Once running, although the GivTCP config webUI comes up on port `8099` it doesn't seem to read/write settings to the config file.

Once running in Docker Desktop I edited the `setting.json` file. Added my AIO IP, set up MQTT credentials for my broker, enabled other settings and enabled the IHD web page (after changing the port to `3001`). After a restart of the container I get the MQTT data and the IHD web page is showing the correct data.

closed 2024-09-25 by davebuk