britkat1980/giv_tcp #195: Warnings in log due to negative Values for Self-Consumption

ProphetOfDoom, 2024-07-08:
I'm running GivTCP 2.4.3

Not sure whether this was due to any recent change to the add-on, but I now seem to be getting occasional slightly -ve values ever since 28 June. I have an AIO/gateway but no Solar PV.

[image]

Log details:
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:326
integration: Sensor (documentation, issues)
First occurred: 7 July 2024 at 20:15:26 (2 occurrences)
Last logged: 7 July 2024 at 20:15:26

Entity sensor.givtcp_chxxxxxxxx_self_consumption_energy_today_kwh from integration mqtt has state class total_increasing, but its state is negative. Triggered by state -0.1 with last_updated set to 2024-07-07T19:09:59.999999+00:00. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.givtcp_chxxxxxxxx_self_consumption_energy_total_kwh from integration mqtt has state class total_increasing, but its state is negative. Triggered by state -46.1 with last_updated set to 2024-07-07T19:09:59.999999+00:00. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

gcoan, 2024-07-08:
I get these quite frequently and just put it down to rounding errors in the calculation.  I suspect I get these more frequently because of having two inverters that don't know about each other and so share the load between them

ProphetOfDoom, 2024-07-08:
I also get these warnings fairly frequently too

Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:303
integration: Sensor (documentation, issues)
First occurred: 7 July 2024 at 20:40:10 (3 occurrences)
Last logged: 02:05:10

Entity sensor.givtcp_chxxxxxxxx_soc_kwh from integration mqtt has state class total_increasing, but its state is not strictly increasing. Triggered by state 9.419 (9.578) with last_updated set to 2024-07-07T19:36:25.679523+00:00. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.givtcp_chxxxxxxxx_load_energy_total_kwh from integration mqtt has state class total_increasing, but its state is not strictly increasing. Triggered by state 1801.0 (1801.1) with last_updated set to 2024-07-08T00:04:02.302368+00:00. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.givtcp_chxxxxxxxx_load_energy_today_kwh from integration mqtt has state class total_increasing, but its state is not strictly increasing. Triggered by state 0.9 (1.0) with last_updated set to 2024-07-08T01:02:59.644264+00:00. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

ProphetOfDoom, 2024-07-08:
I do have to wonder whether state class total_increasing is entirely appropriate for this sensor, as it's up and down like a yoyo, as you would expect, but this is ground that's also been gone over in the Solcast integration discussions/issues

[image]

gcoan, 2024-07-08:
I think this is a bug, they are not total increasing sensors because they don't increase, they should really be measurement or something similar

peter1rhodes, 2024-07-23:
I think it is a bug, and I believe I have found it. [Here, in `HA_Discovery.py`](https://github.com/britkat1980/giv_tcp/blob/90f78b32060b7808a2a055cf29993219bf9d8f66/GivTCP/HA_Discovery.py#L142C1-L142C63), I believe the index should be `1`, not `2`, i.e. the line should read `  if "soc" in str(topic.split("/")[-1]).lower():`

ProphetOfDoom, 2024-07-23:
One down. One to go. Thanks @peter1rhodes. If only I could read/write Python!

britkat1980, 2024-08-08:
Yes, this is probably a rounding error, I've tweaked code to only return value greater or equal to 0 (will land in versions of beta after 620 and v3)

britkat1980, 2024-08-17:
available in current Beta builds

closed 2024-08-17 by britkat1980