britkat1980/giv_tcp #196: AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'

hansboelen, 2024-07-09:
Hi,

It looks like i have a simular issue as other people posted earlier but the posted solutions did not work for me.
My situation: I have home assistant installed as OS and i am using the GivTCP Add-on version.
Version GivTCP 2.4.3 is running, and i have already remove and re-installed it a few times.

I saw some discussions about the config parameters:
AIO (yes/no), How many connected batteries (1/0), Old firmware (yes/no)
I have 6 batteries (20 kwh), inverter: GIV-3HY-8, firmware: DA0.004-DD0.004-A0.004

ps. i also did a try in a docker container with exactly the same results.

Any help would be nice .

Here is some logging in the add-on version:

2024-07-09 21:06:34,204 - startup - [CRITICAL] - Inverter TB2336G121 which is a Gen 1 - Hybrid has been found at: 192.168.1.90
2024-07-09 21:06:34,205 - startup - [INFO] - Searching for Inverters again
2024-07-09 21:06:34,205 - startup - [DEBUG] - GivTCP isAddon: True
2024-07-09 21:06:34,205 - startup - [DEBUG] - No config directory exists, so creating it...
2024-07-09 21:06:34,205 - startup - [CRITICAL] - Running Redis
2024-07-09 21:06:34,206 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024/07/09 21:06:34 [notice] 69#69: using the "epoll" event method
2024/07/09 21:06:34 [notice] 69#69: nginx/1.20.2
2024/07/09 21:06:34 [notice] 69#69: OS: Linux 6.6.33-haos
2024/07/09 21:06:34 [notice] 69#69: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/07/09 21:06:34 [notice] 69#69: start worker processes
2024/07/09 21:06:34 [notice] 69#69: start worker process 73
2024/07/09 21:06:34 [notice] 69#69: start worker process 74
2024-07-09 21:06:34,230 - startup - [DEBUG] - Copying in a template settings.json
2024-07-09 21:06:34,230 - startup - [DEBUG] - Recreating settings.py for invertor 1
2024-07-09 21:06:34,231 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-07-09 21:06:34,231 - startup - [CRITICAL] - Running Invertor (192.168.1.90) read loop every 30s
2024-07-09 21:06:34,233 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-07-09 21:06:34,234 - startup - [CRITICAL] - Starting Gunicorn on port 6345
2024-07-09 21:06:34,235 - startup - [DEBUG] - Creating web dashboard config
2024-07-09 21:06:34,236 - startup - [CRITICAL] - Serving Web Dashboard from port 3042
[2024-07-09 21:06:34 +0200] [78] [INFO] Starting gunicorn 21.2.0
[2024-07-09 21:06:34 +0200] [78] [INFO] Listening at: http://0.0.0.0:6345 (78)
[2024-07-09 21:06:34 +0200] [78] [INFO] Using worker: sync
[2024-07-09 21:06:34 +0200] [90] [INFO] Booting worker with pid: 90
[2024-07-09 21:06:34 +0200] [91] [INFO] Booting worker with pid: 91
[2024-07-09 21:06:34 +0200] [92] [INFO] Booting worker with pid: 92
2024-07-09 21:06:35,094 - Inv1 - read        -  [DEBUG   ] - ----------------------------Starting----------------------------
2024-07-09 21:06:35,094 - Inv1 - read        -  [DEBUG   ] - Getting All Registers
2024-07-09 21:06:35,097 - Inv1 - read        -  [DEBUG   ] - Connecting to: 192.168.1.90
2024-07-09 21:06:35,108 - Inv1 - queue       -  [DEBUG   ] - Pushed job 1286b5ec-ce9f-44d0-aa1c-36cc7c4b72e7 into GivTCP_1
2024-07-09 21:06:35,328 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2024-07-09 21:06:35,627 - Inv1 - sync        -  [DEBUG   ] - Connection to Modbus server established. Socket ('192.168.1.16', 60785)
2024-07-09 21:06:35,629 - Inv1 - transaction -  [DEBUG   ] - Current transaction state - IDLE
2024-07-09 21:06:35,629 - Inv1 - transaction -  [DEBUG   ] - Running transaction 1
2024-07-09 21:06:35,629 - Inv1 - sync        -  [DEBUG   ] - New Transaction state 'SENDING'
 UPDATE  The latest version of `serve` is 14.2.3
 INFO  Accepting connections at http://localhost:3042
2024-07-09 21:06:37,634 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2024-07-09 21:06:38,136 - Inv1 - sync        -  [DEBUG   ] - Connection to Modbus server established. Socket ('192.168.1.16', 38603)
2024-07-09 21:06:38,136 - Inv1 - transaction -  [DEBUG   ] - Current transaction state - TRANSACTION_COMPLETE
2024-07-09 21:06:38,136 - Inv1 - transaction -  [DEBUG   ] - Running transaction 2
2024-07-09 21:06:38,136 - Inv1 - sync        -  [DEBUG   ] - New Transaction state 'SENDING'
2024-07-09 21:06:40,139 - Inv1 - transaction -  [DEBUG   ] - No response received, Expected 164 bytes Recieved 0 bytes !!!!
2024-07-09 21:06:40,139 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2024-07-09 21:06:40,139 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2024-07-09 21:06:40,640 - Inv1 - sync        -  [DEBUG   ] - Connection to Modbus server established. Socket ('192.168.1.16', 51773)
2024-07-09 21:06:40,641 - Inv1 - transaction -  [DEBUG   ] - Current transaction state - TRANSACTION_COMPLETE
2024-07-09 21:06:40,641 - Inv1 - transaction -  [DEBUG   ] - Running transaction 3
2024-07-09 21:06:40,641 - Inv1 - sync        -  [DEBUG   ] - New Transaction state 'SENDING'
2024-07-09 21:06:40,804 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2024-07-09 21:06:40,805 - Inv1 - payload     -  [DEBUG   ] - [b'\x00\x00', b'\x00\x00', b'\x00\x00', b'\x00\x8a']
2024-07-09 21:06:40,805 - Inv1 - transaction -  [DEBUG   ] - Adding transaction 3
2024-07-09 21:06:40,805 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
[... 48 lines cut ...]
...
2024-07-09 21:08:15,430 - Inv1 - mqtt_client -  [DEBUG   ] - Connecting to broker(sub): core-mosquitto
2024-07-09 21:08:15,430 - Inv1 - payload     -  [DEBUG   ] - [b'\x00\x00', b'\x00\x00', b'\x00\x00', b'\x00\x8a']
2024-07-09 21:08:15,430 - Inv1 - transaction -  [DEBUG   ] - Adding transaction 5
2024-07-09 21:08:15,430 - Inv1 - transaction -  [DEBUG   ] - Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2024-07-09 21:08:15,531 - Inv1 - mqtt_client -  [DEBUG   ] - connected OK Returned code=0
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
...

gcoan, 2024-07-09:
Looks like you are running the new 3 phase hybrid.  Have you tried the beta version of GivTCP, I think this has support for 3 phase inverters in it (its not in the main release yet)

hansboelen, 2024-07-10:
thanks for your reply. I did already a short try with the beta and this did also not working. 
I will check this beta version later this day with more patient and let you know.

gcoan, 2024-07-10:
I find if I get the no serial number error, it can be cleared by doing a reset to defaults of the inverter in the portal

gcoan, 2024-07-10:
It's on the computer screen icon / settings / last tab if you scroll across 

This is on my phone so not easy to see, sorry 
[image]

hansboelen, 2024-07-10:
Thanks, but this setting is not available for me. see picture,

[image]

I will give it a new try with the beta version within https://github.com/britkat1980/ha-addons/
i will let you know the outcome

gcoan, 2024-07-10:
I think there is something wrong with the givenergy portal, there should be a lot more tabs available to set Eco mode, charge and discharge slots and to reset the inverter.

The inverter sometimes gets confused (mine certainly does) and reset to defaults cures this.

Suggest you query this with GivEnergy support.

But yes, do try the GivTCP Beta as well

hansboelen, 2024-07-10:
Hi, the beta version give me some progress.
The errors are gone. 
So my original problem (about serial_number) looks solved.
The Giv data is visible in my mqtt explorer and it show lots of data of my Giv.
The GivTCP devices are not visible in HA but maybe the will come within a few hours?
Th web dashboard is also not filled with data (everything is zero).

But stil lots of progress with the Beta version.

gcoan, 2024-07-10:
👍

The givtcp sensors should be created automatically

Have you installed MQTT and the MQTT integration?

Have a look at this video that covers step by step the install process, which logically should be the same for your 3 phase inverter https://www.youtube.com/watch?v=ygD9KyciX54

hansboelen, 2024-07-13:
Hi, I have tried a few things but still i cant see my the GivTCP devices in Home Assistant. 
Follows my mosquitto log it looks like there is  valid conection but also i see a unknown connection which i dont understand.
I am running out of inspiration, problably this is still not working correctly in the beta version.

Mosquitto broker logs:
2024-07-13 13:01:33: New connection from 172.30.32.1:49018 on port 1883.
error: received null username or password for unpwd check
2024-07-13 13:01:33: Client auto-008484A2-8109-616C-15A0-4342C4335491 disconnected, not authorised.
2024-07-13 13:02:10: New connection from 172.30.32.2:33134 on port 1883.
2024-07-13 13:02:10: Client <unknown> closed its connection.
2024-07-13 13:02:31: New connection from 172.30.32.1:53585 on port 1883.
2024-07-13 13:02:31: New client connected from 172.30.32.1:53585 as GivEnergy_GivTCP_1_Control (p2, c1, k60, u'addons').
2024-07-13 13:02:31: New connection from 172.30.32.1:36095 on port 1883.
2024-07-13 13:02:31: New client connected from 172.30.32.1:36095 as GivEnergy_GivTCP_1 (p2, c1, k60, u'addons').
2024-07-13 13:02:32: Client GivEnergy_GivTCP_1 disconnected.

In my mqtt explore i can see lots of Inverter/battery data so it must be correctly configured.
But follows the video i should enter my mqtt user accounts in the givTCP configuration but that part is missing on the screen.
When i look at the terminal in the GivTCP/allsettings.json i can see a random mqtt user which i dont understand.
When i change the to my own mqtt user then it brokes.

GivTCP logs:
2024-07-13 13:01:30,732 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-07-13 13:01:30,734 - startup - [INFO] - Supervisor Timezone: Europe/Amsterdam
2024-07-13 13:01:32,737 - startup - [INFO] - Searching for Inverters
2024-07-13 13:01:32,737 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-07-13 13:01:59,765 - startup - [CRITICAL] - Inverter TB2336G121 which is a Gen1 - Hybrid_3ph with 6 batteries has been found at: 192.168.1.90
2024-07-13 13:01:59,767 - startup - [CRITICAL] - Running Redis
2024-07-13 13:01:59,769 - startup - [CRITICAL] - Running Config Frontend
2024-07-13 13:01:59,774 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024-07-13 13:01:59,792 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-07-13 13:01:59,792 - startup - [CRITICAL] - Running Invertor (192.168.1.90) read loop every 30/120s
2024-07-13 13:01:59,794 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-07-13 13:01:59,795 - startup - [CRITICAL] - Starting Gunicorn on port 6345
2024-07-13 13:01:59,799 - startup - [CRITICAL] - Serving Web Dashboard from port 3042
[2024-07-13 13:02:00 +0200] [84] [INFO] Starting gunicorn 22.0.0
[2024-07-13 13:02:00 +0200] [84] [INFO] Listening at: http://0.0.0.0:6345 (84)
[2024-07-13 13:02:00 +0200] [84] [INFO] Using worker: sync
[2024-07-13 13:02:00 +0200] [97] [INFO] Booting worker with pid: 97
[2024-07-13 13:02:00 +0200] [98] [INFO] Booting worker with pid: 98
[2024-07-13 13:02:00 +0200] [99] [INFO] Booting worker with pid: 99
2024-07-13 13:02:01,242 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-07-13 13:02:01,245 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...
 UPDATE  The latest version of `serve` is 14.2.3
2024-07-13 13:02:01,357 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto

> givtcp-vuejs@0.0.0 dev
> vite --host -l silent

 INFO  Accepting connections at http://localhost:3042
2024-07-13 13:02:30,509 - Inv1 - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
2024-07-13 13:02:30,526 - Inv1 - HA_Discovery -  [ERROR   ] - Error connecting to MQTT Broker: ('KeyError', 'HA_Discovery.py', 130)
2024-07-13 13:03:05,798 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ('KeyError', 'read.py', 1420)

gcoan, 2024-07-13:
@britkat1980 has mentioned on another post that if you swap from the published release to the beta givtcp then you need to delete all the pkl files from the givtcp directory.

Its strange that you can't see the configuration details for the givtcp add-on, can you post what you do have?
It looks like its close to working !

hansboelen, 2024-07-13:
Hi, thanks for your patient :)

i know about the delete of alle files which i did (even the hidden files). I have pretty much linux knowledge.
some more details: 
- Home assistant OS 12.4
- Home assistant core 2024.7.2
- Mosquitto broker: version 6.4.1 (default installation, with the creation of an extra dedicted mqtt user as described in the Video).
- GivTCP Beta 2.4.324
- Home Assistant and Inverter are in the same subnet / vlan.
- GivTCP mqtt part is not available in the (gui)config.
- GivTCP configuration (yaml):
NUMINVERTORS: 1
INVERTOR_IP_1: 192.168.1.90
INVERTOR_NAME_1: Inv1
HADEVICEPREFIX: GivTCP
INVERTOR_IP_2: ""
INVERTOR_NAME_2: Inv2
HADEVICEPREFIX_2: GivTCP2
INVERTOR_IP_3: ""
INVERTOR_NAME_3: Inv3
HADEVICEPREFIX_3: GivTCP3
INVERTOR_IP_4: ""
INVERTOR_NAME_4: Inv4
HADEVICEPREFIX_4: GivTCP4
LOG_LEVEL: Info
SELF_RUN_LOOP_TIMER: 30
SELF_RUN_LOOP_TIMER_FULL: 120
DYNAMICTARIFF: false
DAYRATE: "0.155"
NIGHTRATE: "0.055"
EXPORTRATE: "0.04"
DAYRATESTART: "04:30"
NIGHTRATESTART: "00:30"
INFLUX_OUTPUT: false
INFLUX_URL: ""
INFLUX_TOKEN: ""
INFLUX_BUCKET: ""
INFLUX_ORG: ""
WEB_DASH: true
WEB_DASH_PORT: 3042
DATASMOOTHER: medium
SMARTTARGET: false
PALM_WINTER: 01,02,11,12
PALM_SHOULDER: 03,04,09,10
PALM_MIN_SOC_TARGET: 25
PALM_MAX_SOC_TARGET: 45
PALM_BATT_RESERVE: 4
PALM_BATT_UTILISATION: 0.85
PALM_WEIGHT: 35
GEAPI: ""
SOLCASTAPI: ""
SOLCASTSITEID: ""
SOLCASTSITEID2: ""
LOAD_HIST_WEIGHT: "1"
EVC_ENABLE: false
EVC_IP_ADDRESS: ""
EVC_SELF_RUN_TIMER: 5

GivTCP gui config:

[image]

- inverter 2/3/4 details which i let to its default

[image]

- Dynamic tariff is disabled
- InfluxDB Output is disabled

[image]
[... 291 lines cut ...]
2024-07-13 15:08:06,829 - Inv1 - mqtt        -  [DEBUG   ] - Publishing: GivEnergy/TB2336G121/Battery_Details
2024-07-13 15:08:06,830 - Inv1 - mqtt        -  [DEBUG   ] - Prepping HY2336G523 for publishing
2024-07-13 15:08:06,830 - Inv1 - mqtt        -  [DEBUG   ] - Prepping HY2336G567 for publishing
2024-07-13 15:08:06,830 - Inv1 - mqtt        -  [DEBUG   ] - Prepping HY2336G456 for publishing
2024-07-13 15:08:06,831 - Inv1 - mqtt        -  [DEBUG   ] - Prepping HY2336G480 for publishing
2024-07-13 15:08:06,831 - Inv1 - mqtt        -  [DEBUG   ] - Prepping HY2336G569 for publishing
2024-07-13 15:08:06,831 - Inv1 - mqtt        -  [DEBUG   ] - Prepping HY2336G596 for publishing
2024-07-13 15:08:06,831 - Inv1 - mqtt        -  [DEBUG   ] - Prepping Battery_Stack_1 for publishing
2024-07-13 15:08:06,839 - Inv1 - mqtt        -  [DEBUG   ] - Publishing: GivEnergy/TB2336G121/Power
2024-07-13 15:08:06,839 - Inv1 - mqtt        -  [DEBUG   ] - Prepping Power for publishing
2024-07-13 15:08:06,840 - Inv1 - mqtt        -  [DEBUG   ] - Publishing: GivEnergy/TB2336G121/TB2336G121
2024-07-13 15:08:06,841 - Inv1 - mqtt        -  [DEBUG   ] - Publishing: GivEnergy/TB2336G121/Meter_Details
2024-07-13 15:08:06,842 - Inv1 - mqtt        -  [DEBUG   ] - Prepping Meter_ID1 for publishing
2024-07-13 15:08:06,842 - Inv1 - mqtt        -  [DEBUG   ] - Prepping Meter_ID2 for publishing
2024-07-13 15:08:06,842 - Inv1 - mqtt        -  [DEBUG   ] - Publishing: GivEnergy/TB2336G121/Energy
2024-07-13 15:08:06,842 - Inv1 - mqtt        -  [DEBUG   ] - Prepping Today for publishing
2024-07-13 15:08:06,842 - Inv1 - mqtt        -  [DEBUG   ] - Prepping Total for publishing
2024-07-13 15:08:06,842 - Inv1 - mqtt        -  [DEBUG   ] - Prepping Rates for publishing
2024-07-13 15:08:06,844 - Inv1 - mqtt        -  [DEBUG   ] - Publishing: GivEnergy/TB2336G121/Timeslots
2024-07-13 15:08:06,845 - Inv1 - mqtt        -  [DEBUG   ] - Publishing: GivEnergy/TB2336G121/Control
2024-07-13 15:08:06,847 - Inv1 - mqtt        -  [DEBUG   ] - Publishing: GivEnergy/TB2336G121/Stats

In my mqtt exporere i see the following:

[image]

gcoan, 2024-07-13:
i'm sorry I can't help.

I am running the normal GivTCP release and the config screens have several more options including number of batteries, firmware and mqtt. I can't find any corresponding .yaml config file either to share that.

It looks like the battery voltage your 3-phase inverter is reporting is higher than givtcp expects, so its ignoring that, but other than that I don't see any clues

hansboelen, 2024-07-13:
No problem at all.
I want to thank you for your help!
Probably there is sometime a new firmware or new givtcp version which is working out of the box.
for me, it is not a big problem that it is not working now.
it was more a nice to have for me.
I suggest that we end this issue.
Again, thanks. lot.

gcoan, 2024-07-13:
Thanks for your feedback, sorry I couldn't resolve it for you.
There may be value in leaving this open so @britkat1980 can try to resolve it for you, there's not many people with 3-phase inverters so getting it working isn't straightforward !

britkat1980, 2024-08-08:
Latest beta (620) has been reported as working on 3PH inverters (read only atm), so try that.

hansboelen, 2024-08-09:
Hi, i have installed beta 2.4.614 last few days and i am happy to see that (with this version) my inverter is visible.
Several givtcp devices are visible and contains lots of data. 
My first year i only need to monitor the inverter and controlling it is not needed for me.
Maybe, over some months probably i will give it a try to give orders to the inverter.
It looks like my problem with the givtcp is solved now.
Thanks a lot.

closed 2024-08-21 by britkat1980