britkat1980/giv_tcp #198: GivTCP SDXXXXXX Energy GivTCP Export Energy Today kWh

APoller1, 2024-07-22:
The Export Energy Today variable appears to reset on a change to Forced Export, which then means that the load attribute is then set to include the missing export and massively increases the perceived load.
This has happened on three separate occassions. It does not impact the totals, so my work around has been to use the Export Total and Load Total.
I do not know if this is a GivTCP or integration error.
Gen 1 Inverter.

gcoan, 2024-07-22:
That's weird

I have two gen 1 hybrid inverters and I don't see this behaviour at all:

[image]

I'm running the new fast response firmware 191/193 so I looked back to earlier in the year when I was running the stock 450/451 firmware to see if it was the same, and I don't see any unexpected reset to zero's but I do see the sensor not resetting to zero every day, it resets to 0.8kW on quite a few days.
But this is looking at the long term statistics so it may be there was something wrong with the LTS so I wouldn't take it as gospel that there is an issue with that firmware.
[image]

But either way, no signs of the sensor resetting to zero during a forced export

APoller1, 2024-07-23:
Hi,
      I ran a forced discharge on two days, 17th and 18th and it reset on those days. Either side no discharge nor reset.
`            I looked back and it happened on 3rd June too, not sure if I was exporting that day.
               The total export energy today attribute shows no discontinuity, so it is a strange one...
      Giv Portal shows correct total.

      I can try again today as it's sunny and see if it's a coincidence?

      I'm on version 450.451 - not sure when I'll qualify for the faster version...

Thanks for the prompt response, (I have a work around, just thought I'd register it)

Regards,
      Drew

[cid:507120d9-9bf0-41ce-bcc5-c4db0e1cd38f]

CFE5BA7B-47CD-4C65-836F-4E9D617C2595.png (view on web)<https://github.com/user-attachments/assets/9d0a7e14-c9e5-4be9-87e3-93156955eeb0>

CCCC007E-36BC-4EA8-8074-27FAA9BF9B4B.png (view on web)<https://github.com/user-attachments/assets/9f6a9902-e580-4b96-a462-34c10f832643>

APoller1, 2024-07-24:
I tried export again yesterday and no problem occured. 
It's happened 4 times since Jan. No idea what it correlates with. 
Could be firmware.
The graph I attached in the response does not appear above, did you receive it or do I need to re-add?

gcoan, 2024-07-24:
> The graph I attached in the response does not appear above, did you receive it or do I need to re-add?

Seems to have disappeared, can you re-add?

If it reoccurs can you add a givtcp log, and if it keeps happening maybe turn the debug log level up and include that

APoller1, 2024-07-25:
[image]

APoller1, 2024-07-25:
> > The graph I attached in the response does not appear above, did you receive it or do I need to re-add?
> 
> Seems to have disappeared, can you re-add?
> 
> If it reoccurs can you add a givtcp log, and if it keeps happening maybe turn the debug log level up and include that

Will add a log when and if it reoccurs. Thanks

APoller1, 2024-08-15:
Reoccured on 13th August at 11.13 ish - Nothing in the Log?
[image]

Log is here for the day;
2024-08-13 00:00:03,366 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/SDGXXX/Energy/Today/Self_Consumption_Energy_Today_kWh - ['Self_Consumption_Energy_Today_kWh', 7.799999999999999]
2024-08-13 00:00:03,366 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/SDGXXX/Energy/Today/Load_Energy_Today_kWh - ['Load_Energy_Today_kWh', 9.3]
2024-08-13 00:00:03,367 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/SDGXXX/Energy/Today/Battery_Charge_Energy_Today_kWh - ['Battery_Charge_Energy_Today_kWh', 3.7]
2024-08-13 00:00:03,368 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/SDGXXX/Energy/Today/Battery_Discharge_Energy_Today_kWh - ['Battery_Discharge_Energy_Today_kWh', 5.4]
2024-08-13 00:00:03,369 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/SDGXXX/Energy/Today/Battery_Throughput_Today_kWh - ['Battery_Throughput_Today_kWh', 9.100000000000001]
2024-08-13 00:00:03,373 - Inv1 - mqtt        -  [ERROR   ] - MQTT error trying to send a <class 'list'> to the MQTT broker for: GivEnergy/SDGXXX/Energy/Rates/Import_ppkwh_Today - ['Import_ppkwh_Today', 0.312]
2024-08-13 00:00:18,475 - Inv1 - read        -  [CRITICAL] - Midnight, so resetting Day/Night stats...
2024-08-13 00:00:34,803 - Inv1 - read        -  [CRITICAL] - Midnight, so resetting Day/Night stats...
2024-08-13 00:00:51,128 - Inv1 - read        -  [CRITICAL] - Midnight, so resetting Day/Night stats...
2024-08-13 03:11:05,429 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2024-08-13 04:01:27,647 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2024-08-13 04:56:47,636 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2024-08-13 06:17:20,197 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2024-08-13 06:27:23,925 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2024-08-13 07:00:06,303 - Inv1 - read        -  [INFO    ] - Saving current energy stats at start of day rate tariff (Dynamic)
2024-08-13 07:12:40,233 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2024-08-13 09:15:00,153 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 120 (1)
2024-08-13 09:15:03,776 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 1 was a success
2024-08-13 11:30:00,149 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 1250 (13)
2024-08-13 11:30:02,256 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 13 was a success
2024-08-13 11:49:28,568 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2024-08-13 13:40:08,846 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2024-08-13 15:00:00,191 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 2600 (50)
2024-08-13 15:00:02,343 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 50 was a success
2024-08-13 16:36:14,600 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2024-08-13 18:11:50,329 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2024-08-13 19:57:29,090 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2024-08-13 20:32:41,731 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2024-08-13 21:28:01,060 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2024-08-13 22:33:25,529 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2024-08-13 22:53:35,733 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2024-08-13 23:45:00,329 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 20
2024-08-13 23:45:04,573 - Inv1 - write       -  [INFO    ] - Setting Charge Target 20 was a success

APoller1, 2024-08-15:
I have old firmware set to 'on' if that changes anything (Gen 1 )
If I set old firmware to 'off' it doesn't connect and I get the error below;

2024-08-15 17:15:03,733 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3138, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'

gcoan, 2024-08-15:
> I have old firmware set to 'on' if that changes anything (Gen 1 )
> If I set old firmware to 'off' it doesn't connect and I get the error below;

If you are running the standard Gen1 firmware version 450/451 then the 'old firmware' switch has to be on.

If you're using the fast performance 191/193 then you can turn old firmware off and you get some extra config options that are in that firmware.

Sounds like you have 450/451 though

Geoffrey

APoller1, 2024-08-15:
Yes I have 450/1.

Just ruling that out.

closed 2024-10-01 by britkat1980