britkat1980/giv_tcp #200: EVC Issues - Error controlling EVC: (<class 'struct.error'>, error('required argument is not an integer')

Grey-Sheep, 2024-07-23:
Hi

I have been struggling for a week to get the giv_tcp to talk to my newly installed EVC.

It occasionally works (after Givenergy support have done something) but mostly doesn't.

The errors that I can't rack down are:-

2024-07-23 21:32:40,784 - Inv1 - evc         -  [INFO    ] - Setting Charge control to: Start
2024-07-23 21:32:40,786 - Inv1 - evc         -  [INFO    ] - Setting Charge current limit to: 21.684602649006624
2024-07-23 21:32:40,788 - Inv1 - evc         -  [ERROR   ] - Error controlling EVC: (<class 'struct.error'>, error('required argument is not an integer'), <traceback object at 0x7fbaf23b00>)
2024-07-23 21:32:51,588 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.120, 502) failed: [Errno 104] Connection reset by peer
---------------------------------and

2024-07-23 21:37:02,267 - Inv1 - evc         -  [ERROR   ] - Error controlling EVC: (<class 'pymodbus.exceptions.ConnectionException'>, ConnectionException('Failed to connect[ModbusTcpClient(192.168.1.120:502)]'), <traceback object at 0x7fa162ecc0>)
2024-07-23 21:37:02,268 - Inv1 - evc         -  [INFO    ] - Setting Charge current limit to: 22.23045267489712
2024-07-23 21:37:02,271 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.120, 502) failed: [Errno 111] Connection refused
2024-07-23 21:37:02,271 - Inv1 - evc         -  [ERROR   ] - Error controlling EVC: (<class 'pymodbus.exceptions.ConnectionException'>, ConnectionException('Failed to connect[ModbusTcpClient(192.168.1.120:502)]'), <traceback object at 0x7fa162fdc0>)

Has anyone else seen these errors?

Have found the installers manual, and the only area that may be a clue is “RS485 EM115 meter comms (for solar/hybrid charging)”.  Does this mean that it needs this cable (connecting to the gateway??) for solar charging to work?  As I don't have this connection, could this be the source of the errors?

Thanks

Jan

peter1rhodes, 2024-08-02:
Interesting, I am getting similar messages. I have been plagued by the `Connection reset by peer` error, and I am also getting `Error controlling EVC: (<class 'struct.error'>, error('required argument is not an integer'), <traceback object at 0x7fb5051ec0>)`. It's hard to say whether they're related

peter1rhodes, 2024-08-02:
@britkat1980 I think I may know why this error is happening, so I humbly offer my suggestion for a fix. I'm happy to do this as a PR, but I'm not confident in actually running custom code on my HA instance as an add-on, so I can't easily test it at the moment. I also see you've made edits to `evc.py` in your `dev3` branch, but I don't think they cover this issue yet.

[In `evc.py`, where the current limit is set](https://github.com/britkat1980/giv_tcp/blob/90f78b32060b7808a2a055cf29993219bf9d8f66/GivTCP/evc.py#L372), it's converted from a float in Amps to deciamps (i.e. it's multiplied by 10). I'm fairly sure the struct needs to pack an integer, not a float, but this cast is never done. So I think a fix would simply be to change:

`client.write_registers(91,(val*10))`

 to

`client.write_registers(91, int(round(val*10, 0)))`

Grey-Sheep, 2024-08-07:
This appears to have been fixed in the latest release.

However, I still can't get the inverter to talk to the EVC.  Givenergy say it doesn't work and is is "coming later"

closed 2024-08-07 by Grey-Sheep