britkat1980/giv_tcp #202: All entities "Unknown"

CaptSB, 2024-07-24:
Until 10:40:25 this morning (24 July 2024) everything was working. At that point HA seems to have stopped communicating with my GivEnergy system (Hybrid Inverter V1). Everything looks good in the GivEnergy app. 

I have rebooted HA and restarted the GivTCP service but nothing seems to work. 

Running on a Raspberry Pi 4 with all the latest HA updates installed.

CaptSB, 2024-07-24:
I tried lots of things and eventually, uninstalling and reinstalling GivTCP fixed the issue. Seemed like the IP address had changed or become corrupted.

britkat1980, 2024-08-08:
Did the IP address change? The upcoming v3 should auto detect a change in IP address, and fix for you.

toxicneo, 2024-08-29:
I have a similar issue but no change in IP address, here's what's in the log when it started up today after an update to HA. I'm going to try uninstall/reinstall as per @CaptSB see if that resolves it

`2024-08-29 06:43:13,074 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-08-29 06:43:15,113 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-08-29 06:43:19,662 - startup - [CRITICAL] - Inverter SA2202G167 which is a Gen 1 - Hybrid has been found at: 192.168.50.242
2024-08-29 06:43:19,672 - startup - [CRITICAL] - Running Redis
2024-08-29 06:43:19,678 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024/08/29 06:43:19 [notice] 68#68: using the "epoll" event method
2024/08/29 06:43:19 [notice] 68#68: nginx/1.20.2
2024/08/29 06:43:19 [notice] 68#68: OS: Linux 6.6.46-haos
2024/08/29 06:43:19 [notice] 68#68: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/08/29 06:43:19 [notice] 68#68: start worker processes
2024/08/29 06:43:19 [notice] 68#68: start worker process 72
2024/08/29 06:43:19 [notice] 68#68: start worker process 73
2024-08-29 06:43:19,831 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-08-29 06:43:19,832 - startup - [CRITICAL] - Running Invertor (192.168.50.242) read loop every 5s
2024-08-29 06:43:19,836 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-08-29 06:43:19,840 - startup - [CRITICAL] - Starting Gunicorn on port 6345
2024-08-29 06:43:19,856 - startup - [CRITICAL] - Serving Web Dashboard from port 3042
[2024-08-29 06:43:21 +0100] [77] [INFO] Starting gunicorn 23.0.0
[2024-08-29 06:43:21 +0100] [77] [INFO] Listening at: http://0.0.0.0:6345 (77)
[2024-08-29 06:43:21 +0100] [77] [INFO] Using worker: sync
[2024-08-29 06:43:21 +0100] [89] [INFO] Booting worker with pid: 89
[2024-08-29 06:43:21 +0100] [90] [INFO] Booting worker with pid: 90
[2024-08-29 06:43:21 +0100] [91] [INFO] Booting worker with pid: 91
2024-08-29 06:43:26,488 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
 UPDATE  The latest version of `serve` is 14.2.3
 INFO  Accepting connections at http://localhost:3042
Traceback (most recent call last):
  File "/app/GivTCP_1/read.py", line 62, in getData
    raise Exception ("Garbage or failed inverter Response: "+ str(plant.result))
Exception: Garbage or failed inverter Response: ERROR:-(<class 'KeyError'>, KeyError(HR:013), <traceback object at 0x7f61056e5c80>)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/GivTCP_1/read.py", line 1188, in <module>
    globals()[sys.argv[1]]()
  File "/app/GivTCP_1/read.py", line 785, in self_run2
    runAll("True")
  File "/app/GivTCP_1/read.py", line 748, in runAll
    result=getData(full_refresh)
  File "/app/GivTCP_1/read.py", line 76, in getData
    consecFails(e)
  File "/app/GivTCP_1/read.py", line 724, in consecFails
    oldDataCount= pickle.load(inp)
EOFError: Ran out of input
2024-08-29 06:45:06,629 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2297, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1686, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2100, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3142, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3814, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3934, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'`

britkat1980, 2024-08-29:
Ok, has the inverter recently updated firmware? If so Perhaps the old firmware toggle needs changing?

toxicneo, 2024-08-30:
> Ok, has the inverter recently updated firmware? If so Perhaps the old firmware toggle needs changing?

No changes to the firmware recently for either the battery or the inverter, I'm running D0.450-A0.451 and 3005 on Gen1 gear. I toggled and restarted to double check but still have the same issue.

I can connect on the GE app & Rubikcube Inverter app and GivEnergy Local works on HA. Not sure if that helps?

britkat1980, 2024-08-30:
Can you try removing all discovery messages from the MQTT broker then restarting HA. It could be that the discovery messages are incorrect or missing.

toxicneo, 2024-09-10:
So I've tried that, both by publishing a blank retained message in Actions through HA, and by deleting them in MQTT explorer and still had the same issue. Took the extreme of uninstalling MQTT and the broker, restarting and setting up from scratch and still get it.

> 2024-09-10 10:28:30,927 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
>  INFO  Accepting connections at http://localhost:3000
> 2024-09-10 10:28:36,715 - Inv1 - read        -  [CRITICAL] - First time running so saving AC Charge status
> Traceback (most recent call last):
>   File "/app/GivTCP_1/read.py", line 691, in getData
>     previousUpdate = pickle.load(inp)
> EOFError: Ran out of input
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/app/GivTCP_1/read.py", line 1188, in <module>
>     globals()[sys.argv[1]]()
>   File "/app/GivTCP_1/read.py", line 785, in self_run2
>     runAll("True")
>   File "/app/GivTCP_1/read.py", line 748, in runAll
>     result=getData(full_refresh)
>   File "/app/GivTCP_1/read.py", line 713, in getData
>     consecFails(e)
>   File "/app/GivTCP_1/read.py", line 724, in consecFails
>     oldDataCount= pickle.load(inp)
> EOFError: Ran out of input
> 2024-09-10 10:30:11,069 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
> Traceback (most recent call last):
>   File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
>     client.loop_forever()
>   File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2297, in loop_forever
>     rc = self._loop(timeout)
>   File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1686, in _loop
>     rc = self.loop_read()
>   File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2100, in loop_read
>     rc = self._packet_read()
>   File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3142, in _packet_read
>     rc = self._packet_handle()
>   File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3814, in _packet_handle
>     return self._handle_connack()
>   File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3934, in _handle_connack
>     on_connect(
>   File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
>     client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
> AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
> 2024-09-10 10:30:12,614 - Inv1 - app         -  [ERROR   ] - Exception on /runAll [GET]
> Traceback (most recent call last):
>   File "/app/GivTCP_1/read.py", line 691, in getData
>     previousUpdate = pickle.load(inp)
> EOFError: Ran out of input
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.10/site-packages/flask/app.py", line 2529, in wsgi_app
>     response = self.full_dispatch_request()
>   File "/usr/local/lib/python3.10/site-packages/flask/app.py", line 1825, in full_dispatch_request
>     rv = self.handle_user_exception(e)
>   File "/usr/local/lib/python3.10/site-packages/flask_cors/extension.py", line 194, in wrapped_function
>     return cors_after_request(app.make_response(f(*args, **kwargs)))
>   File "/usr/local/lib/python3.10/site-packages/flask/app.py", line 1823, in full_dispatch_request
>     rv = self.dispatch_request()
>   File "/usr/local/lib/python3.10/site-packages/flask/app.py", line 1799, in dispatch_request
>     return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
>   File "/app/GivTCP_1/REST.py", line 38, in getAll
>     return rd.runAll(True)
>   File "/app/GivTCP_1/read.py", line 748, in runAll
>     result=getData(full_refresh)
>   File "/app/GivTCP_1/read.py", line 713, in getData
>     consecFails(e)
>   File "/app/GivTCP_1/read.py", line 724, in consecFails
>     oldDataCount= pickle.load(inp)
> EOFError: Ran out of input
> 2024-09-10 10:30:17,819 - Inv1 - app         -  [ERROR   ] - Exception on /runAll [GET]
> Traceback (most recent call last):
>   File "/app/GivTCP_1/read.py", line 62, in getData
>     raise Exception ("Garbage or failed inverter Response: "+ str(plant.result))
> Exception: Garbage or failed inverter Response: ERROR:-(<class 'KeyError'>, KeyError(IR:110), <traceback object at 0x7f3e75d1f100>)
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.10/site-packages/flask/app.py", line 2529, in wsgi_app
>     response = self.full_dispatch_request()
>   File "/usr/local/lib/python3.10/site-packages/flask/app.py", line 1825, in full_dispatch_request
>     rv = self.handle_user_exception(e)
>   File "/usr/local/lib/python3.10/site-packages/flask_cors/extension.py", line 194, in wrapped_function
>     return cors_after_request(app.make_response(f(*args, **kwargs)))
>   File "/usr/local/lib/python3.10/site-packages/flask/app.py", line 1823, in full_dispatch_request
>     rv = self.dispatch_request()
>   File "/usr/local/lib/python3.10/site-packages/flask/app.py", line 1799, in dispatch_request
>     return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
>   File "/app/GivTCP_1/REST.py", line 38, in getAll
>     return rd.runAll(True)
>   File "/app/GivTCP_1/read.py", line 748, in runAll
>     result=getData(full_refresh)
>   File "/app/GivTCP_1/read.py", line 76, in getData
>     consecFails(e)
>   File "/app/GivTCP_1/read.py", line 724, in consecFails
>     oldDataCount= pickle.load(inp)
> EOFError: Ran out of input

gcoan, 2024-09-10:
> EOFError: Ran out of input
> 2024-09-10 10:30:11,069 - Inv1 - mqtt_client - [ERROR ] - No serial_number found in MQTT queue. MQTT Control not available.
> Traceback (most recent call last):

I occasionally get unknown serial number error on my gen 1 inverter. It's likely a problem with the inverter not givtcp 

Resetting the inverter to system defaults through the GivEnergy portal and restarting the inverter can fix this issue for me. Sometimes with powering off the inverter. 

Sometimes deleting the pkl files in the givtcp folder and restarting givtcp is enough though.

toxicneo, 2024-09-10:
@gcoan deleting the pkl files and a restart seems to have solved it! Also got rid of all the _2 entities that appeared as well! Thank you!

gcoan, 2024-09-10:
> @gcoan deleting the pkl files and a restart seems to have solved it! Also got rid of all the _2 entities that appeared as well! Thank you!

Great news

closed 2024-10-01 by britkat1980