britkat1980/giv_tcp #204: GivTCP Beta - Parallel AIO

jasonwfrancis1, 2024-07-29:
Hi, At the weekend I had a second AIO installed.

I installed the beta version of GivTCP to home assistant and it pretty much went without a hitch.

However, I cannot seem to find an entity for the combined SOC of the two AIO's.

In the config I have added the gateway and both AIO's, and I get the separate SOC's for each AIO sent from the gateway, just not the combined SOC.

Is this expected?

jasonwfrancis1, 2024-07-29:
I also do not see a combined battery capacity and the gateway is reporting a 6000 export limit, when it should be 12000 based on 2xAIO's

jasonwfrancis1, 2024-07-31:
Id really be interested in having a chat about parallel AIO and getting the right entities so that predbat can work, its not a normal setup and i don't know what data is available in the gateway etc.

britkat1980, 2024-08-08:
For a combined SOC, what do you want to see? An average of the two SOCs? Its not given by the Gateway directly, but it could be calculated in GivTCP.
For the Export limit, thats reported by the gateway itself, so If it comes up as 6000, that's what's coming from the Gateway, perhaps get someone at GE or your installer to confirm if its right?

thephoton, 2024-08-13:
My vote would be for the (mean) average of all attached AIOs.

Predbat is looking in the REST data for 
Power/Power/SOC (average)
Power/Power/SOC_kWh (a sum)

Which at the moment don't exist, so it errors out, but adding to both MQTT and REST would be awesome.

jasonwfrancis1, 2024-08-16:
Yeah the average of the attached AIO’s would be ideal, I think that’s what the gateway shows in the portal anyway.

at the moment I have had to disable rest to get pedbat to work, and I had to create a bunch of combines sensors to get it to work without rest.

for the custom power flow card also I had to create some combined sensors to make things work.

I’ll put together a full overview of what’s missing in general and rest and maybe you can see if you can do anything to help? There’s a bunch of people heading down the multiple AIO route now.

britkat1980, 2024-08-16:
This is now available in beta649 if you can check it matches what you expect

jasonwfrancis1, 2024-08-16:
Thanks I’ll take a look later tonight, there will be 2 new entities in the gateway? (I will need to disable my manually created ones or they will clash)

britkat1980, 2024-08-16:
Just an SOC one at the moment. I need to find a way of getting the battery capacity from the gateway before I can provide SOC_kWh

jasonwfrancis1, 2024-08-16:
In my givtcp config I have the gateway and both aios configured. So you have the data you need, it’s just against the individual aios.

So if in the config the aios are configured and you’re reading them, then provide the extra entities? (I believe that gateway reports the AIO serial numbers so you would know what to look for?) But if only the gateway is configured or not all aios are configured then don’t provide them?

This is basically how I have created my manual sensors for all the stuff that’s currently missing from the gateway.

jasonwfrancis1, 2024-08-16:
> This is now available in beta649 if you can check it matches what you expect

My HA is only showing 642?

britkat1980, 2024-08-16:
The architecture of GivTCP is that each device is run as a discrete process so GW process doesn’t know about AIO process… that makes it tricky to grab AIO data simply to get this. It’s on my long list to bring in AIO battery data into GW somehow, but I need to get v3 out to prod first

jasonwfrancis1, 2024-08-16:
Ah ok I get it.

Will be the same principle for a lot of the other missing stuff.

I’ll still make a list and longer term you can see what’s possible.

I kind of have everything working at the moment.

jasonwfrancis1, 2024-08-17:
I updated to 651 and 652 but both give me this error and givtcp does not start

[image0.jpeg]

jasonwfrancis1, 2024-08-17:
> I updated to 651 and 652 but both give me this error and givtcp does not start
> […](#)

In error 1 is the gateway.

[image]

britkat1980, 2024-08-17:
update to Beta656 and this should now be solved

jasonwfrancis1, 2024-08-17:
New SOC entity for the gateway shows as 0%

AIO1 provided by the gateway is 50% and the other is 49%

britkat1980, 2024-08-17:
Well that’s annoying! Looked like it was working on another Gateway… let me take a look

thephoton, 2024-08-17:
0% for me too unfortunately.

Is it a computed value or from a GW register?

The correct individual figures are in the mqtt tree
[image]

jasonwfrancis1, 2024-08-17:
Can you take the individual values and average them?

The soc% does show on the portal against the gateway but I don’t know how they do it - wether it’s a register or they just do a calculation…

Screenshot_20240817_184451_Microsoft.Remote.Desktop.jpg (view on web)<https://github.com/user-attachments/assets/c67d27ff-1cba-47eb-9bff-d9ddb561c21a>

jasonwfrancis1, 2024-08-17:
Screenshot from the portal:

[image]

britkat1980, 2024-08-17:
try Dev660 and see if it gives you want you need

jasonwfrancis1, 2024-08-18:
Updating to 665 I get this error from the gateway.

[image]

britkat1980, 2024-08-18:
667 has a catch, but I can't reproduce the bug, let me know if it works for you

jasonwfrancis1, 2024-08-19:
Seems to be good now, and the doc is working too .

closed 2024-08-21 by britkat1980